MichaIng/DietPi #3823: Add Paper Minecraft Server

ravenclaw900, 2020-10-11:
### Creating a software request

#### Formal software information
- Software title | 
Paper
- Short description | 
Highly optimized Minecraft server for Java Edition
- Official URL (if available) | 
https://papermc.io
- GitHub URL (if available) | 
https://github.com/PaperMC/Paper
- (Official) install documentation | 
https://paper.readthedocs.io/en/latest/server/index.html

#### Are there similar/alternative software titles available with DietPi-Software?

- Nukkit

#### What makes your requested software better than the above solutions, if available?

- Highly Optimized
- Runs on Java Edition

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Lots of configuration, could be automated

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. `sudo apt update`
2. `sudo apt install openjdk-8-headless`
3. `curl -sSf https://papermc.io/ci/job/Paper-1.16/lastSuccessfulBuild/artifact/paperclip.jar -o paperclip.jar`
4. `sudo mv paperclip.jar /usr/local/bin/paper/`
5. `java -jar /usr/local/bin/paper/paperclip.jar`
6. Ask to agree to the EULA
7. `echo eula=true > /usr/local/bin/eula.txt`
8. Configure

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

ravenclaw900, 2020-10-27:
PR merged.

MichaIng, 2020-10-27:
Let's keep the issue open to show that it's openen for fine tuning and testing 🙂.

ravenclaw900, 2020-12-13:
This looks interesting: https://github.com/magmafoundation/Magma/.  PaperMC, but with support for both Forge mods and plugins.  Just a little outdated though.

MichaIng, 2020-12-13:
@omavoss
Let's discuss here.

If `enable-rcon=true` was not added to `/mnt/dietpi_userdata/papermc/server.properties` it makes sense that mcrcon fails to access as.

omavoss, 2020-12-13:
Thanks for moving my thread, I stopped the papermc prozess successfully and I wait now for instructions.

ravenclaw900, 2020-12-13:
What's in the log (located at /var/log/papermc/latest.log)?

omavoss, 2020-12-13:
@ravenclaw900 👍 

[code]
[21:48:15] [main/INFO]: Environment: authHost='https://authserver.mojang.com', accountsHost='https://api.mojang.com', sessionHost='https://sessionserver.mojang.com', servicesHost='https://api.minecraftservices.com', name='PROD'
[21:48:16] [main/INFO]: Found new data pack file/bukkit, loading it automatically
[21:48:18] [main/INFO]: Reloading ResourceManager: Default, bukkit
[21:48:25] [Worker-Main-2/INFO]: Loaded 7 recipes
[21:48:36] [Server thread/INFO]: Starting minecraft server version 1.16.4
[21:48:36] [Server thread/WARN]: To start the server with more ram, launch it as "java -Xmx1024M -Xms1024M -jar minecraft_server.jar"
[21:48:36] [Server thread/INFO]: Loading properties
[21:48:37] [Server thread/INFO]: This server is running Paper version git-Paper-324 (MC: 1.16.4) (Implementing API version 1.16.4-R0.1-SNAPSHOT)
[21:48:37] [Server thread/INFO]: Console input is disabled due to --noconsole command argument
[21:48:37] [Server thread/INFO]: Server Ping Player Sample Count: 12
[21:48:37] [Server thread/INFO]: Using 4 threads for Netty based IO
[21:48:37] [Server thread/INFO]: Debug logging is disabled
[21:48:37] [Server thread/INFO]: Default game type: SURVIVAL
[21:48:37] [Server thread/INFO]: Generating keypair
[21:48:39] [Server thread/INFO]: Starting Minecraft server on *:25565
[21:48:39] [Server thread/INFO]: Using default channel type
[21:48:42] [Server thread/INFO]: [Geyser-Spigot] Loading Geyser-Spigot v1.2.0-SNAPSHOT
[21:48:42] [Server thread/INFO]: [floodgate-bukkit] Loading floodgate-bukkit v1.0-SNAPSHOT
[21:48:50] [Server thread/INFO]: [floodgate-bukkit] Loading Floodgate linked player database...
[21:48:51] [Server thread/INFO]: Server permissions file permissions.yml is empty, ignoring it
[21:48:51] [Server thread/INFO]: Preparing level "world"
[21:49:45] [Server thread/WARN]: Unable to find spawn biome
[21:49:56] [Server thread/WARN]: Unable to find spawn biome
[21:49:59] [Server thread/INFO]: Preparing start region for dimension minecraft:overworld
[21:50:02] [Server thread/INFO]: Loaded 0 spawn chunks for world world
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:02] [Worker-Main-2/INFO]: Preparing spawn area: 0%
[21:50:02] [Server thread/INFO]: Time elapsed: 3643 ms
[21:50:02] [Server thread/INFO]: Preparing start region for dimension minecraft:the_nether
[21:50:07] [Server thread/INFO]: Loaded 0 spawn chunks for world world_nether
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:07] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:08] [Worker-Main-1/INFO]: Preparing spawn area: 0%
[21:50:10] [Server thread/INFO]: Time elapsed: 7709 ms
[21:50:10] [Server thread/INFO]: Preparing start region for dimension minecraft:the_end
[21:50:15] [Server thread/INFO]: Loaded 0 spawn chunks for world world_the_end
[21:50:15] [Server thread/INFO]: Time elapsed: 4799 ms
[21:50:15] [Server thread/INFO]: [Geyser-Spigot] Enabling Geyser-Spigot v1.2.0-SNAPSHOT
[21:50:22] [Server thread/INFO]: [Geyser-Spigot] Auto-loaded Floodgate key
[21:50:22] [Server thread/INFO]: [Geyser-Spigot] ******************************************
[21:50:22] [Server thread/INFO]: [Geyser-Spigot] 
[21:50:22] [Server thread/INFO]: [Geyser-Spigot] Loading Geyser version 1.2.0-SNAPSHOT (git-master-41cb593)
[21:50:22] [Server thread/INFO]: [Geyser-Spigot] 
[21:50:22] [Server thread/INFO]: [Geyser-Spigot] ******************************************
[21:53:29] [Thread-1/INFO]: Stopping main thread (Ignore any thread death message you see! - DO NOT REPORT THREAD DEATH TO PAPER)
[21:53:29] [Server thread/ERROR]: Main thread terminated by WatchDog due to hard crash
java.lang.ThreadDeath: null
	at java.lang.Thread.stop(Thread.java:942) [?:?]
	at net.minecraft.server.v1_16_R3.MinecraftServer.stop(MinecraftServer.java:758) ~[patched_1.16.4.jar:git-Paper-324]
	at net.minecraft.server.v1_16_R3.DedicatedServer.stop(DedicatedServer.java:704) ~[patched_1.16.4.jar:git-Paper-324]
	at net.minecraft.server.v1_16_R3.MinecraftServer.close(MinecraftServer.java:732) ~[patched_1.16.4.jar:git-Paper-324]
	at org.bukkit.craftbukkit.v1_16_R3.util.ServerShutdownThread.run(ServerShutdownThread.java:28) ~[patched_1.16.4.jar:git-Paper-324]
[21:53:29] [Thread-1/INFO]: Stopping server
[21:53:29] [Thread-1/INFO]: Saving players
[21:53:30] [Thread-1/INFO]: Saving worlds
[21:53:30] [Thread-1/INFO]: Saving chunks for level 'ServerLevel[world]'/minecraft:overworld
[/code]

ravenclaw900, 2020-12-13:
No problem.  When I ran it, I got an out of memory error (and the exact same behavior as you), but I'm not seeing that here.  How about journalctl?  `journalctl | grep 'papermc'`

omavoss, 2020-12-13:
`root@DietPi:~# journalctl | grep 'papermc'
Dez 13 21:47:23 DietPi systemd[1]: Started papermc (DietPi).
Dez 13 21:47:34 DietPi papermc[1187]: System Info: Java 11 (OpenJDK Server VM 11.0.9.1+1-post-Raspbian-1deb10u2) Host: Linux 5.4.79-v7+ (arm)
Dez 13 21:47:34 DietPi papermc[1187]: Loading libraries, please wait...
Dez 13 21:48:15 DietPi papermc[1187]: [21:48:15 INFO]: Environment: authHost='https://authserver.mojang.com', accountsHost='https://api.mojang.com', sessionHost='https://sessionserver.mojang.com', servicesHost='https://api.minecraftservices.com', name='PROD'
Dez 13 21:48:16 DietPi papermc[1187]: [21:48:16 INFO]: Found new data pack file/bukkit, loading it automatically
Dez 13 21:48:18 DietPi papermc[1187]: [21:48:18 INFO]: Reloading ResourceManager: Default, bukkit
Dez 13 21:48:25 DietPi papermc[1187]: [21:48:25 INFO]: Loaded 7 recipes
Dez 13 21:48:36 DietPi papermc[1187]: [21:48:36 INFO]: Starting minecraft server version 1.16.4
Dez 13 21:48:36 DietPi papermc[1187]: [21:48:36 WARN]: To start the server with more ram, launch it as "java -Xmx1024M -Xms1024M -jar minecraft_server.jar"
Dez 13 21:48:36 DietPi papermc[1187]: [21:48:36 INFO]: Loading properties
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: This server is running Paper version git-Paper-324 (MC: 1.16.4) (Implementing API version 1.16.4-R0.1-SNAPSHOT)
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: Console input is disabled due to --noconsole command argument
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: Server Ping Player Sample Count: 12
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: Using 4 threads for Netty based IO
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: Debug logging is disabled
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: Default game type: SURVIVAL
Dez 13 21:48:37 DietPi papermc[1187]: [21:48:37 INFO]: Generating keypair
Dez 13 21:48:39 DietPi papermc[1187]: [21:48:39 INFO]: Starting Minecraft server on *:25565
Dez 13 21:48:39 DietPi papermc[1187]: [21:48:39 INFO]: Using default channel type
Dez 13 21:48:42 DietPi papermc[1187]: [21:48:42 INFO]: [Geyser-Spigot] Loading Geyser-Spigot v1.2.0-SNAPSHOT
Dez 13 21:48:42 DietPi papermc[1187]: [21:48:42 INFO]: [floodgate-bukkit] Loading floodgate-bukkit v1.0-SNAPSHOT
Dez 13 21:48:50 DietPi papermc[1187]: [21:48:50 INFO]: [floodgate-bukkit] Loading Floodgate linked player database...
Dez 13 21:48:51 DietPi papermc[1187]: [21:48:51 INFO]: Server permissions file permissions.yml is empty, ignoring it
Dez 13 21:48:51 DietPi papermc[1187]: [21:48:51 INFO]: Preparing level "world"
Dez 13 21:49:45 DietPi papermc[1187]: [21:49:45 WARN]: Unable to find spawn biome
Dez 13 21:49:56 DietPi papermc[1187]: [21:49:56 WARN]: Unable to find spawn biome
Dez 13 21:49:59 DietPi papermc[1187]: [21:49:59 INFO]: Preparing start region for dimension minecraft:overworld
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Loaded 0 spawn chunks for world world
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing spawn area: 0%
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Time elapsed: 3643 ms
Dez 13 21:50:02 DietPi papermc[1187]: [21:50:02 INFO]: Preparing start region for dimension minecraft:the_nether
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Loaded 0 spawn chunks for world world_nether
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:07 DietPi papermc[1187]: [21:50:07 INFO]: Preparing spawn area: 0%
Dez 13 21:50:08 DietPi papermc[1187]: [21:50:08 INFO]: Preparing spawn area: 0%
Dez 13 21:50:10 DietPi papermc[1187]: [21:50:10 INFO]: Time elapsed: 7709 ms
Dez 13 21:50:10 DietPi papermc[1187]: [21:50:10 INFO]: Preparing start region for dimension minecraft:the_end
Dez 13 21:50:15 DietPi papermc[1187]: [21:50:15 INFO]: Loaded 0 spawn chunks for world world_the_end
Dez 13 21:50:15 DietPi papermc[1187]: [21:50:15 INFO]: Time elapsed: 4799 ms
Dez 13 21:50:15 DietPi papermc[1187]: [21:50:15 INFO]: [Geyser-Spigot] Enabling Geyser-Spigot v1.2.0-SNAPSHOT
Dez 13 21:50:22 DietPi papermc[1187]: [21:50:22 INFO]: [Geyser-Spigot] Auto-loaded Floodgate key
[... cut ...]
Dez 13 22:21:05 DietPi papermc[2765]: [22:21:05 INFO]: [Geyser-Spigot] ******************************************
Dez 13 22:21:05 DietPi papermc[2765]: [22:21:05 INFO]: [Geyser-Spigot]
Dez 13 22:21:05 DietPi papermc[2765]: [22:21:05 INFO]: [Geyser-Spigot] Loading Geyser version 1.2.0-SNAPSHOT (git-master-41cb593)
Dez 13 22:21:05 DietPi papermc[2765]: [22:21:05 INFO]: [Geyser-Spigot]
Dez 13 22:21:05 DietPi papermc[2765]: [22:21:05 INFO]: [Geyser-Spigot] ******************************************
root@DietPi:~#
`

omavoss, 2020-12-13:
@Joulinar 👍 
Drag and drop from putty to github, I would like to try it. Thank you.

ravenclaw900, 2020-12-13:
Well, we might as well get to solving the problem with the config files.  Run `cd /mnt/dietpi_userdata/papermc ; rm -rf ./[!plugins]* ; systemctl start papermc`, then wait for it to finish starting up.  After that, could you post the `latest.log` file again?

omavoss, 2020-12-13:
root@DietPi:~# cd /mnt/dietpi_userdata/papermc
root@DietPi:/mnt/dietpi_userdata/papermc# rm -rf ./[!plugins]*
root@DietPi:/mnt/dietpi_userdata/papermc# systemctl start papermc
root@DietPi:/mnt/dietpi_userdata/papermc#

`[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//it.unimi.dsi.fastutil.ints.Int2ObjectOpenHashMap$1.iterator(Int2ObjectOpenHashMap.java:959)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.processTrackQueue(PlayerChunkMap.java:2040)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.g(PlayerChunkMap.java:2063)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:865)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.DedicatedServer.b(DedicatedServer.java:377)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.a(MinecraftServer.java:1209)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:997)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:01:09] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:09] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH ---
[23:01:09] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:20] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH  - git-Paper-324 (MC: 1.16.4) ---
[23:01:20] [Paper Watchdog Thread/ERROR]: The server has not responded for 11 seconds! Creating thread dump
[23:01:20] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:20] [Paper Watchdog Thread/ERROR]: Server thread dump (Look for plugins here before reporting to Paper!):
[23:01:22] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:22] [Paper Watchdog Thread/ERROR]: Current Thread: Server thread
[23:01:22] [Paper Watchdog Thread/ERROR]: 	PID: 16 | Suspended: false | Native: false | State: RUNNABLE
[23:01:22] [Paper Watchdog Thread/ERROR]: 	Stack:
[23:01:22] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer$$Lambda$3470/0x5d0e3228.accept(Unknown Source)
[23:01:24] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.forEachVisibleChunk(PlayerChunkMap.java:628)
[23:01:24] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:825)
[23:01:27] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:01:27] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.DedicatedServer.b(DedicatedServer.java:377)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.a(MinecraftServer.java:1209)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:997)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:01:31] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:31] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH ---
[23:01:31] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH  - git-Paper-324 (MC: 1.16.4) ---
[23:01:33] [Paper Watchdog Thread/ERROR]: The server has not responded for 24 seconds! Creating thread dump
[23:01:33] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: Server thread dump (Look for plugins here before reporting to Paper!):
[23:01:33] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: Current Thread: Server thread
[23:01:33] [Paper Watchdog Thread/ERROR]: 	PID: 16 | Suspended: false | Native: false | State: RUNNABLE
[23:01:33] [Paper Watchdog Thread/ERROR]: 	Stack:
[... cut ...]
[23:02:24] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:815)
[23:02:29] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:02:29] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:02:29] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
`
@Joulinar: pls. help with attached one file with winscp . . . into code-tags.

omavoss, 2020-12-13:
@Joulinar:
What am I doing wrong? I would like to be able to insert the files corrrectly by myself please. Thank you.

ravenclaw900, 2020-12-13:
Are you sure that this is the full file?  It looks incomplete (starting with an error and ending with an incomplete thread dump).

Joulinar, 2020-12-13:
@omavoss 
it you like to mark a single word or sentence, you can use single quote before and after:

``` ` ```

if it is a full text block, it's 3 of them

` ``` `

omavoss, 2020-12-13:
```
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//it.unimi.dsi.fastutil.ints.Int2ObjectOpenHashMap$1.iterator(Int2ObjectOpenHashMap.java:959)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.processTrackQueue(PlayerChunkMap.java:2040)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.g(PlayerChunkMap.java:2063)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:865)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.DedicatedServer.b(DedicatedServer.java:377)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.a(MinecraftServer.java:1209)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:997)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:01:09] [Paper Watchdog Thread/ERROR]: 		java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:01:09] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:09] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH ---
[23:01:09] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:20] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH  - git-Paper-324 (MC: 1.16.4) ---
[23:01:20] [Paper Watchdog Thread/ERROR]: The server has not responded for 11 seconds! Creating thread dump
[23:01:20] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:20] [Paper Watchdog Thread/ERROR]: Server thread dump (Look for plugins here before reporting to Paper!):
[23:01:22] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:22] [Paper Watchdog Thread/ERROR]: Current Thread: Server thread
[23:01:22] [Paper Watchdog Thread/ERROR]: 	PID: 16 | Suspended: false | Native: false | State: RUNNABLE
[23:01:22] [Paper Watchdog Thread/ERROR]: 	Stack:
[23:01:22] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer$$Lambda$3470/0x5d0e3228.accept(Unknown Source)
[23:01:24] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.forEachVisibleChunk(PlayerChunkMap.java:628)
[23:01:24] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:825)
[23:01:27] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:01:27] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.DedicatedServer.b(DedicatedServer.java:377)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.a(MinecraftServer.java:1209)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:997)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:01:31] [Paper Watchdog Thread/ERROR]: 		java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:01:31] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:31] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH ---
[23:01:31] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH  - git-Paper-324 (MC: 1.16.4) ---
[23:01:33] [Paper Watchdog Thread/ERROR]: The server has not responded for 24 seconds! Creating thread dump
[23:01:33] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: Server thread dump (Look for plugins here before reporting to Paper!):
[23:01:33] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: Current Thread: Server thread
[23:01:33] [Paper Watchdog Thread/ERROR]: 	PID: 16 | Suspended: false | Native: false | State: RUNNABLE
[23:01:33] [Paper Watchdog Thread/ERROR]: 	Stack:
[23:01:33] [Paper Watchdog Thread/ERROR]: 		app//it.unimi.dsi.fastutil.ints.Int2ObjectOpenHashMap$1.iterator(Int2ObjectOpenHashMap.java:959)
[23:01:40] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.PlayerChunkMap.processTrackQueue(PlayerChunkMap.java:2040)
[... cut ...]
[23:11:09] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:11:14] [Paper Watchdog Thread/ERROR]: 		app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:11:36] [Paper Watchdog Thread/ERROR]: 		java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:11:36] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:11:36] [Paper Watchdog Thread/ERROR]: Entire Thread Dump:
```

@Joulinar: Many thanks.

omavoss, 2020-12-13:
```
root@DietPi:~# cat /var/log/papermc/latest.log
[23:01:06] [Paper Watchdog Thread/ERROR]:       Stack:
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//it.unimi.dsi.fastutil.ints.Int2ObjectOpenHashMap$1.iterator(Int2ObjectOpenHashMap.java:959)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.PlayerChunkMap.processTrackQueue(PlayerChunkMap.java:2040)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.PlayerChunkMap.g(PlayerChunkMap.java:2063)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:865)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.DedicatedServer.b(DedicatedServer.java:377)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.a(MinecraftServer.java:1209)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:997)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:01:09] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:01:09] [Paper Watchdog Thread/ERROR]:               java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:01:09] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:09] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH ---
[23:01:09] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:20] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH  - git-Paper-324 (MC: 1.16.4) ---
[23:01:20] [Paper Watchdog Thread/ERROR]: The server has not responded for 11 seconds! Creating thread dump
[23:01:20] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:20] [Paper Watchdog Thread/ERROR]: Server thread dump (Look for plugins here before reporting to Paper!):
[23:01:22] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:22] [Paper Watchdog Thread/ERROR]: Current Thread: Server thread
[23:01:22] [Paper Watchdog Thread/ERROR]:       PID: 16 | Suspended: false | Native: false | State: RUNNABLE
[23:01:22] [Paper Watchdog Thread/ERROR]:       Stack:
[23:01:22] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.ChunkProviderServer$$Lambda$3470/0x5d0e3228.accept(Unknown Source)
[23:01:24] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.PlayerChunkMap.forEachVisibleChunk(PlayerChunkMap.java:628)
[23:01:24] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tickChunks(ChunkProviderServer.java:825)
[23:01:27] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.ChunkProviderServer.tick(ChunkProviderServer.java:743)
[23:01:27] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.WorldServer.doTick(WorldServer.java:460)
[23:01:31] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.b(MinecraftServer.java:1356)
[23:01:31] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.DedicatedServer.b(DedicatedServer.java:377)
[23:01:31] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.a(MinecraftServer.java:1209)
[23:01:31] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:997)
[23:01:31] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:178)
[23:01:31] [Paper Watchdog Thread/ERROR]:               app//net.minecraft.server.v1_16_R3.MinecraftServer$$Lambda$3115/0x5eaa1428.run(Unknown Source)
[23:01:31] [Paper Watchdog Thread/ERROR]:               java.base@11.0.9.1/java.lang.Thread.run(Thread.java:834)
[23:01:31] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:31] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH ---
[23:01:31] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: --- DO NOT REPORT THIS TO PAPER - THIS IS NOT A BUG OR A CRASH  - git-Paper-324 (MC: 1.16.4) ---
[23:01:33] [Paper Watchdog Thread/ERROR]: The server has not responded for 24 seconds! Creating thread dump
[23:01:33] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: Server thread dump (Look for plugins here before reporting to Paper!):
[23:01:33] [Paper Watchdog Thread/ERROR]: ------------------------------
[23:01:33] [Paper Watchdog Thread/ERROR]: Current Thread: Server thread
[... cut ...]
        at java.lang.Thread.run(Thread.java:834) [?:?]
Caused by: java.lang.OutOfMemoryError: Java heap space
[23:30:34] [Worker-Main-2/WARN]: java.util.concurrent.CompletionException: java.lang.OutOfMemoryError: Java heap space
[23:30:34] [Worker-Main-2/WARN]:        at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:314)
[23:30:34] [Worker-Main-2/WARN]:        at java.base/java.util.concurrent.CompletableFuture.completeThrowable(CompletableFuture.java:319)
root@DietPi:~#
```

Joulinar, 2020-12-13:
well you could download the entire file to your computer and attach it to GitHub using drag & drop

ravenclaw900, 2020-12-13:
Yep, there's the out of memory error.  Remember to quit the service.  The only way to fix this would be to increase the swapfile size, I think.  @MichaIng, do you think that we should disable (or warn about) PaperMC for SBCs with less than 1 GiB of RAM?

omavoss, 2020-12-13:
I will now going 8 -10 hours zu bed, I hope see you in the morning; thank you.

Joulinar, 2020-12-13:
I guess a RPi 3 Model B is used. Usually using DietPi default setup, you should have 1GB ram + 1GB swap

ravenclaw900, 2020-12-13:
Same where I was testing on my VM.  If there are other things running, though, it could take up too much RAM anyway.

Joulinar, 2020-12-13:
independent from the issue during installation, PaperMC is starting on my RPi3B+ without issues. It's using around 400MB of Ram

```
root@DietPi3:# ps aux | head -1; ps aux | grep ^paper| sort -rnk 4
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
papermc    422  145 38.9 446012 388316 ?       Ssl  00:36  16:39 /usr/bin/java -jar /opt/papermc/paperclip.jar --nogui --noconsole
root@DietPi3:#

```

MichaIng, 2020-12-14:
We do already increase the swap size and/or reduce the concurrent build runners based on available memory, so that at very least 2 GiB are available and should be enough to finish the build.

Has the issue with mcrcon been sorted, respectively why the config file has not been created during install as intended? As well due to service crash with OOM?

omavoss, 2020-12-14:
@Joulinar 
the same here:
```
root@DietPi:~# ps aux | head -1; ps aux | grep ^paper| sort -rnk 4
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
papermc   1189 66.9 38.6 433020 370964 ?       Ssl  Dez13 504:28 /usr/bin/java -jar /opt/papermc/paperclip.jar --nogui --noconsole
root@DietPi:~#
```

closed 2020-12-20 by MichaIng

ravenclaw900, 2021-08-02:
Hey @MichaIng, is there a reason for PaperMC requiring Node.js?

MichaIng, 2021-08-02:
It was wrong translated here 😅: https://github.com/MichaIng/DietPi/pull/4314/files#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5L1058-R1021

MichaIng, 2021-08-02:
It did depend on build-essentail, but I'm not sure if this is required. AFAIK currently it fails because too low Java version on Debian Stretch and Buster.