MichaIng/DietPi #3825: Default over_voltage on RPi Zero?

MichaIng, 2020-10-14:
Can someone who runs an RPi Zero do a quick verification that the default `over_voltage` there is `6` instead of `0`? Would make sense considering SoC and frequencies compared to RPi 1, but probably someone can verify:
```
vcgencmd get_config over_voltage
```
grep '^[[:blank:]]*over_voltage=' /boot/config.txt
```

Since we need to change what is shown in `dietpi-config` > `Performance Options` when the setting is unset + the commented value on first boot in `config.txt`.

Joulinar, 2020-10-14:
Sorry @MichaIng I'm not owning such device. :weary:

maartenlangeveld, 2020-10-14:
`vcgencmd get_config over_voltage`
`over_voltage=0`

`vcgencmd measure_volts`
`volt=1.2000V`

MichaIng, 2020-10-14:
Okay, then the official docs are wrong in this point, not only that default is `0` but also since it switches between those two boarders just like on every other RPi model. From the docs it seems like voltage is static on RPi 1 and Zero: https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/overclocking.md

Joulinar, 2020-10-14:
I have an RPi B, do you need some testing on it as well?

MichaIng, 2020-10-14:
That would be great, just to assure that I am currently editing the docs correctly.

Joulinar, 2020-10-14:
ok will do tomorrow as I need to switch devices. Still need to get a switch with POE to be able to power on devices remotely 😆

MichaIng, 2020-10-15:
I have another measure: `vcgencmd measure_volts sdram_p`
While this should be 1.2V, on my RPi 2 it is 1.2250V, and other than I thought first, not affected by `over_voltage` but by `over_voltage_sdram_p` only. Now interesting if `over_voltage_sdram_p=1` it the default which leads to this or if that one leads to 1.25V 😄.

Joulinar, 2020-10-15:
@MichaIng 
there you go
```
 ─────────────────────────────────────────────────────
 DietPi v6.33.1 (beta) : 13:18 - Thu 10/15/20
 ─────────────────────────────────────────────────────
 - Device model : RPi B (armv6l)
 - CPU temp : 53'C : 127'F (Running warm, but safe)
 - LAN IP : 192.168.0.15 (eth0)
 - MOTD : RPi users, please do not lower your idle CPU frequency yet:
          https://github.com/MichaIng/DietPi/issues/3690
 ─────────────────────────────────────────────────────

root@DietPi1:~# vcgencmd get_config over_voltage
over_voltage=2
root@DietPi1:~# vcgencmd measure_volts
volt=1.2000V
root@DietPi1:~# vcgencmd measure_volts
volt=1.2500V
root@DietPi1:~# vcgencmd measure_volts
volt=1.2500V
root@DietPi1:~# vcgencmd measure_volts
volt=1.2000V
root@DietPi1:~#

```

MichaIng, 2020-10-15:
Many thanks, if the Pi is still running, what does `vcgencmd measure_volts sdram_p` report?
___________
So then the documentation is true for RPi 1 (only), since indeed 1.2V is the default (min+max) while raising `over_voltage` raises the max by 0.025V each integer step. On all other RPi models the upper limit is 1.35V by default instead. I found a few other statements in the docs which simply show that those parts have simply not been updated since RPi 1 (like no frequency scaling without applying overclocking settings). I'm currently rewriting/updating that.

Joulinar, 2020-10-15:
```
root@DietPi1:~# vcgencmd measure_volts sdram_p
volt=1.2250V
root@DietPi1:~#
```

MichaIng, 2020-10-15:
Awesome, thanks. I also found out that changing `over_voltage_sdram_p` setting has no effect on the voltage, currently, although I only switched between `-1` `0` and `1`.

maartenlangeveld, 2020-10-15:
> vcgencmd measure_volts sdram_p

```
 ─────────────────────────────────────────────────────
 DietPi v6.32.2 : 13:36 - Thu 10/15/20
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - Uptime : up 5 days, 22 hours, 41 minutes
 - CPU temp : 28'C : 82'F (Who put me in the freezer!)
 - FQDN/hostname : M-STREAMER
 - LAN IP : 192.xxx.xxx.xxx (wlan0)
 - Freespace (RootFS) : 6.5G
 - Freespace (music)  : 215G
 - MOTD : RPi users, please do not lower your idle CPU frequency yet:
          https://github.com/MichaIng/DietPi/issues/3690
 ─────────────────────────────────────────────────────

root@M-STREAMER:~# vcgencmd measure_volts sdram_p
volt=1.2250V
root@M-STREAMER:~# vcgencmd measure_volts sdram_p
volt=1.2250V
root@M-STREAMER:~# vcgencmd measure_volts sdram_p
volt=1.2250V
root@M-STREAMER:~# vcgencmd measure_volts sdram_p
volt=1.2250V
root@M-STREAMER:~# vcgencmd measure_volts sdram_p
volt=1.2250V
root@M-STREAMER:~# vcgencmd measure_volts sdram_p
volt=1.2250V
```

MichaIng, 2020-10-15:
Okay, I think I got it now right: https://github.com/raspberrypi/documentation/pull/1698

closed 2020-10-17 by MichaIng