MichaIng/DietPi #3833: DietPi-Software | double click on a shortcut on a desktop environment will ask what to do

Joulinar, 2020-10-16:
@MichaIng 
We have 2 reports on our forum asking the same question

https://dietpi.com/phpbb/viewtopic.php?f=9&t=7875
https://dietpi.com/phpbb/viewtopic.php?f=9&t=8068

It's about using our predefined shortcuts on desktop environments. Every time you double klick on such a shortcut, you will be ask if you like to execute or abort. User expectation is to have the application open without any question to answer.

As far as I can see, this is coming from the usage if sym links to define the shortcuts on the desktop environment

https://github.com/MichaIng/DietPi/blob/7dd9d5f40de1cae2db5566f91a96d1fc094ae816/dietpi/dietpi-software#L2757-L2761

```
root@DietPi3:~# ls -la /root/Desktop/
total 12
drwxr-xr-x  2 root root 4096 Oct 16 12:21 .
drwx------ 13 root root 4096 Oct 16 13:17 ..
lrwxrwxrwx  1 root root   45 Oct 16 12:17 dietpi-config.desktop -> /usr/share/applications/dietpi-config.desktop
lrwxrwxrwx  1 root root   47 Oct 16 12:17 dietpi-launcher.desktop -> /usr/share/applications/dietpi-launcher.desktop
lrwxrwxrwx  1 root root   47 Oct 16 12:17 dietpi-software.desktop -> /usr/share/applications/dietpi-software.desktop
lrwxrwxrwx  1 root root   36 Oct 16 12:17 htop.desktop -> /usr/share/applications/htop.desktop
lrwxrwxrwx  1 root root   39 Oct 16 12:17 pcmanfm.desktop -> /usr/share/applications/pcmanfm.desktop
root@DietPi3:~#
```

A possible solution/workaround could be replacing the sym links with config files. This file need to contain the information about the app and where to find it. For testing, I created a 2nd dietpi-software shortcut

`nano /root/Desktop/dietpi-software.desktop2`

Add following

```
[Desktop Entry]
Type=Link
Name=DietPi-Software
Icon=/var/lib/dietpi/dietpi-software/installed/desktop/icons/dietpi-icon.png
URL=/usr/share/applications/dietpi-software.desktop
```
Save and exit the file. You now have the 2nd dietpi-software shortcut, that can be used with being ask anything 😄 

Config files could be stored on Github same way as the other config files for our desktop systems `https://raw.githubusercontent.com/$G_GITOWNER/DietPi/$G_GITBRANCH/.conf/desktop/`

MichaIng, 2020-10-16:
Ah yes, I recognised this being normal now on LXDE/Openbox. The question can actually be disabled and I already wanted to find out where this flag is stored so we can apply it by default. Let me see if I find it again.

Interesting that a `Type=Link` entry does not include the question prompt. However an actual link "file" to a regular desktop/application item sounds like an overhead to me and that it bypasses the question makes me wonder if other, probably more important parts of regular desktop application executions are missing then as well 🤔.
Also: When opening the application menu, one ist still asked, so I'd prefer disabling this question all together 😉.

MichaIng, 2020-10-16:
There it is: `PCManFM` > `Edit` > `Preferences` > `General` > `Don't ask options to launch executable file`
[image]

Joulinar, 2020-10-16:
found it in `/root/.config/libfm/libfm.conf`. As soon as `quick_exec=1`, you will not be ask to confirm execution 😄

MichaIng, 2020-10-16:
Bingo: `quick_exec=1`
G_CONFIG_INJECT 'quick_exec=' 'quick_exec=1'

Joulinar, 2020-10-16:
LOL you found it as well.

closed 2020-10-17 by MichaIng