MichaIng/DietPi #3835: NanoPi M2 not installing properly from DietPi SD Card

mrwesleycrusher, 2020-10-16:
#### Required Information
- DietPi version | 6.32.2
- Distro version | 10.5
- Kernel version | Linux DietPi 3.4.39-s5p4418 #4 SMP PREEMPT Mon Oct 16 11:24:24 HKT 2017 armv7l GNU/Linux
- SBC model | NanoPi M2
- Power supply used | 5V 4A cyberpower power strip USB output
- SDcard used | Sandisk Ultra 64GB

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? Fresh install 
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Flash the USB Drive with a fresh DietPi image
2. Insert the SD Card into the NanoPi M2 and power it on
3. Log in with default credentials
4. Follow prompts to change passwords for user accounts and software
5. Press "OK" on the installer to install DietPi

#### Expected behaviour
- DietPi installs to the SD Card

#### Actual behaviour
- DietPi install fails with an error

Joulinar, 2020-10-16:
Hi,

Many thanks for your report. Pls can you share some error messages you are facing!

mrwesleycrusher, 2020-10-16:
I can get something more descriptive here in a little bit, I am in the middle of something else and can't access the NanoPi at the moment. The apt update failed with an error regarding low free space if I remember correctly...

mrwesleycrusher, 2020-10-17:
[logs.zip](https://github.com/MichaIng/DietPi/files/5394744/logs.zip)

There is a zip file containing all the logs from my boot taken from /var/tmp/dietpi/

apt update is failing and that seems to be causing a LOT of other issues

MichaIng, 2020-10-17:
Here is the issue in `fs_partition_resize.log`:
```
Error: Error informing the kernel about modifications to partition /dev/mmcblk0p2 -- Invalid argument.  This means Linux won't know about any changes you made to /dev/mmcblk0p2 until you reboot -- so you shouldn't mount it or use it in any way before rebooting.
Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
```
The partition has been expended successfully but `partprobe` failed for some reason so the file system could not be expended.

Please do the following:
```
partprobe /dev/mmcblk0
partx -u /dev/mmcblk0
resize2fs /dev/mmcblk0p2
```
Hopefully this succeeds then in expending the file system. The redo the firstrun setup:
```
systemctl enable dietpi-firstboot
echo -1 > /boot/dietpi/.install_stage
```

I'll add `partx -u` to our script as well as it tells the kernel about updated partition table in a different way that, as I observed, succeeds in some cases where `partprobe` (alone) fails, e.g. I observed this especially in cases when using the same loop device like `/dev/loop0` for two different image files one after another. At least it does not hurt.

mrwesleycrusher, 2020-10-17:
Thanks for the help! Unfortunately none of those 3 strategies suceeded:
`# partprobe /dev/mmcblk0
Error: Error informing the kernel about modifications to partition /dev/mmcblk0p2 -- Invalid argument.  This means Linux won't know about any changes you made to /dev/mmcblk0p2 until you reboot -- so you shouldn't mount it or use it in any way before rebooting.
Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.`

`#partx -u /dev/mmcblk0
partx: /dev/mmcblk0: error updating partitions 1-2`

`#resize2fs /dev/mmcblk0
resize2fs 1.44.5 (15-Dec-2018)
resize2fs: Device or resource busy while trying to open /dev/mmcblk0
Couldn't find valid filesystem superblock.`

MichaIng, 2020-10-17:
Okay then something with the partition tyble seems to be broken. Can you show:
```
fdisk -l
parted /dev/mmcblk0 print
lsblk /dev/mmcblk
```

mrwesleycrusher, 2020-10-18:
`root@DietPi:~# fdisk -l
Disk /dev/mmcblk0: 59.5 GiB, 63864569856 bytes, 124735488 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5714d005

Device         Boot  Start       End   Sectors  Size Id Type
/dev/mmcblk0p1        2048    133119    131072   64M 83 Linux
/dev/mmcblk0p2      133120 124735487 124602368 59.4G 83 Linux
`
`root@DietPi:~# parted /dev/mmcblk0 print
Model: SD SC64G (sd/mmc)
Disk /dev/mmcblk0: 63.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  68.2MB  67.1MB  primary  ext4
 2      68.2MB  63.9GB  63.8GB  primary  ext4`

`root@DietPi:~# lsblk /dev/mmcblk
lsblk: /dev/mmcblk: not a block device`

`root@DietPi:~# df -T
Filesystem     Type     1K-blocks   Used Available Use% Mounted on
udev           devtmpfs    412924      0    412924   0% /dev
tmpfs          tmpfs        83020   2576     80444   4% /run
/dev/mmcblk0p2 ext4        414996 412824      2172 100% /
tmpfs          tmpfs       415080      0    415080   0% /dev/shm
tmpfs          tmpfs         5120      0      5120   0% /run/lock
tmpfs          tmpfs       415080      0    415080   0% /sys/fs/cgroup
tmpfs          tmpfs        51200      8     51192   1% /var/log
tmpfs          tmpfs       414720      0    414720   0% /tmp
/dev/mmcblk0p1 ext4         64496  19012     45484  30% /boot`

MichaIng, 2020-10-18:
Ah a typo, the one command missed a `0` should have been:
```
lsblk /dev/mmcblk0
```
But everything else shows the pretty fine partition table, partitions with correctly expended size and correct file system so I am puzzled why informing the kernel about this fails.
You did try a reboot, didn't you?

mrwesleycrusher, 2020-10-18:
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  59.5G  0 disk 
─mmcblk0p1 179:1    0    64M  0 part /boot
─mmcblk0p2 179:2    0 413.1M  0 part /
```
There that should be better.

I have indeed tried a reboot, although I am glad you asked just in case I hadn't

MichaIng, 2020-10-18:
I'll run a test install and analyse as currently I have no idea how this can happen. One last thing you could do:
```
> /forcefsck
reboot
```
=> Force a file system check on reboot, probably it finds something that is not apparent even when running into the partprobe error.

mrwesleycrusher, 2020-11-02:
Hey sorry I disappeared for the last few weeks. I tried forcing fsck like you recommended, but no dice. If I inspect when the filesystem was last checked on the next boot, it shows that it was inspected on September 22nd, but the forcefsck file is no longer present in the `/` directory. So I am not certain if fsck is actually running or not.

MichaIng, 2020-11-03:
fsck may be been executed within the initramfs and I think then there is no timestamp updated. If you're still in the same session you could check if this file exists: `/run/initramfs/fsck-root`
That is created when the initramfs runs fsck, if it does not exist, systemd-fsck will instead and the following should show the output:
```
journalctl -t systemd-fsck
```
or
```
journalctl -u systemd-fsck-root
```

mrwesleycrusher, 2020-11-08:
`/run/initramfs/fsck-root` does exist, it is an empty file however...

MichaIng, 2020-11-08:
That is expected. It is a flag file which only mainly tells `systemd-fsck` to not run fsck a second time. There could be a log file at `/run/initramfs/fsck.log`, could you check it's content:
```
cat /run/initramfs/fsck.log
```

mrwesleycrusher, 2020-11-09:
```
root@DietPi:~# cat /run/initramfs/fsck.log
Log of fsck -a -t ext4 /dev/mmcblk0p2
Fri Jan  1 08:09:28 2016

fsck from util-linux 2.26.2
rootfs: clean, 14792/32000 files, 105223/105763 blocks

Fri Jan  1 08:09:28 2016
----------------
root@DietPi:~# date
Mon Nov  9 04:39:00 GMT 2020
```

MichaIng, 2020-11-09:
The timestamp is expectedly wrong, as it's done before anything could have updated/synced system time. That the files exist is basically a prove that it ran at this boot session.

Okay, so file system is clean. Still it does not want to expend? `resize2fs /dev/mmcblk0`

mrwesleycrusher, 2020-11-10:
```
root@DietPi:~# resize2fs /dev/mmcblk0
resize2fs 1.44.5 (15-Dec-2018)
resize2fs: Device or resource busy while trying to open /dev/mmcblk0
Couldn't find valid filesystem superblock.
```

I haven't rebooted the system since the fsck FYI

mrwesleycrusher, 2020-11-10:
ah so that failed because I wasn't actually on a partition:
```
root@DietPi:~# resize2fs /dev/mmcblk0
resize2fs 1.44.5 (15-Dec-2018)
resize2fs: Device or resource busy while trying to open /dev/mmcblk0
Couldn't find valid filesystem superblock.
```

MichaIng, 2020-11-10:
> ah so that failed because I wasn't actually on a partition:

And I have overseen it at your first attempt and then forgot to add the `p2` partition suffix myself in last post (just fixed it).

> The filesystem on /dev/mmcblk0p2 is now 15575296 (4k) blocks long.

15575296 4k blocks =~ 59.4 GiB so the file system has correct size now. I guess there was in fact an issue with the partition or file system first but has been solved with subsequent fsck/reboot.

> Should I reboot and try installing again?

Does not hurt, but the following should do it as well: `/boot/dietpi/dietpi-login`
It should start with an update and then prompt `dietpi-software` for firstrun setup/installs.

mrwesleycrusher, 2020-11-13:
Yeah so that fixed it! Woohoo! Do you think this is something that should be put in the documentation as a remedy? Or do you think there is a bug fix that can be done here?

MichaIng, 2020-11-14:
The initial issue was a file system error, so a general troubleshooting guide with often occurring issues, debugging and solution attempts would indeed be nice in the docs.

mrwesleycrusher, 2020-11-14:
Okay! I will go make an issue over in the docs to track this and make a fork!

closed 2020-11-14 by mrwesleycrusher