MichaIng/DietPi #3837: Light weight desktop is not starting with usb ssd boot with SD card everything is fine.

PIOTA89, 2020-10-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-10-18:
Hi,

many thanks for your message. Do you have any error message? Is your system booting normal into login screen and just desktop is not starting? Do you flashed the image directly to SSD? What kind of device you are using? Was it already booting using SSD or did this never worked, even without having a desktop installed?

PIOTA89, 2020-10-19:
Good Morning,

today I will send you all details I have to that bug, because it’s very important for my project that the bug gets fixed soon as possible. 

Thank you in advance.

PIOTA89, 2020-10-19:
Hello, 

I don’t know if you did something. But now it works very well. Thank you for your great work. That’s really awesome. 
Best regards 
Paul 

Von meinem iPad gesendet

> Am 19.10.2020 um 07:25 schrieb PIOTA <piota@mail.de>:
>

Joulinar, 2020-10-19:
well we didn't do anything. 😃 

Ok I will close the topic. Feel free to reopen if needed.

closed 2020-10-19 by Joulinar