MichaIng/DietPi #3839: Allo Sparky USBridge: 6.33.3 update fails

plsvn, 2020-10-19:
#### Details:
- Date           | Sun Oct 11 22:07:20 CEST 2020
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | DietPi
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux AlloUSBridge 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | wget --spider -t 1 -T 10 ssh.dietpi.com
- Exit code      | 5
- Software title | DietPi-Bugreport
#### Steps to reproduce:

1. ssh into dietpi (6.32.2)
2. run "dietpi-update
#### Expected behaviour:

- dietpi updates to latest version
#### Actual behaviour:

- displays "FAILED"
#### Extra details:

- ...
#### Additional logs:
```
URL transformed to HTTPS due to an HSTS policy
Spider mode enabled. Check if remote file exists.
--2020-10-11 22:07:20--  https://ssh.dietpi.com/
Resolving ssh.dietpi.com (ssh.dietpi.com)... 185.101.93.93
Connecting to ssh.dietpi.com (ssh.dietpi.com)|185.101.93.93|:443... connected.
ERROR: The certificate of ‘ssh.dietpi.com’ is not trusted.
ERROR: The certificate of ‘ssh.dietpi.com’ doesn't have a known issuer.
The certificate's owner does not match hostname ‘ssh.dietpi.com’

Joulinar, 2020-10-19:
Hi,

Many thanks for your report. Pls can you have a look to your system time? Looks like it's not up-to-date.

plsvn, 2020-10-19:
oooops, sorry: it's set to October 11 :-/

thank you Joulinar
(... and sorry indeed once more)

Joulinar, 2020-10-19:
System time can be updated manually using `/boot/dietpi/func/run_ntpd 1`

plsvn, 2020-10-19:
issue is... it was set to manual
solved now :)

Joulinar, 2020-10-19:
ok I will close the topic for now, feel free to reopen if needed.

closed 2020-10-19 by Joulinar