MichaIng/DietPi #3840: PiHole Dashboar 403 Forbidden

jimmyhenne, 2020-10-19:
Hello,
after I updated myMPD 6.7.0 and DietPi 6.33 I can't access the PiHole dashboard anymore There is the 403 forbidden message in the explorer when I try to access it. Before these updates, the PiHole dashboard worked fine. I have reinstalled lighttpd several times and edited the lighttpd.conf according to "https://github.com/pi-hole/pi-hole/blob/master/advanced/lighttpd.conf.debian". Unfortunately all without success so far
I will appreciate any help.

MichaIng, 2020-10-19:
Please add some general information like your device model, distro version and such, basically please fill the bug report template to allow us helping more effectively 😉.

Joulinar, 2020-10-19:
Hi,

many thanks for your message. Usually it's not needed to configure `lighttpd` following any guide. On DietPi whole configurations is invoked via `/etc/lighttpd/conf-enabled`. One question, which URL you are using to access PiHole?

jimmyhenne, 2020-10-19:
Thanks for your quick responses.

#### Required Information
- DietPi version | 6.33.3
- Distro version | buster
- Kernel version | uname -a
- SBC model | Linux DietPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
- Power supply used | original RasPi PS
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | PiHole
- Was the software title installed freshly or updated/migrated? PiHole has been running flawlessly for a year but dashoard broke after myMPD update to 6.7 or DietPi update 6.33.3
- Can this issue be replicated on a fresh installation of DietPi? Haven't done a fresh installation yet.

#### Actual behaviour
- When adressing `http://pi.hole/admin/` in firefox I get 403 Forbidden.

#### Extra details
- I already tried `sudo dpkg -P lighttpd`, `sudo apt install --reinstall lighttpd`, `pihole -r` several times

**systemctl status lighttpd:**
```lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-10-19 21:26:56 BST; 51min ago
  Process: 935 ExecReload=/bin/kill -USR1 $MAINPID (code=exited, status=0/SUCCESS)
 Main PID: 899 (lighttpd)
    Tasks: 1 (limit: 2094)
   CGroup: /system.slice/lighttpd.service
           └─899 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf

Okt 19 21:26:55 DietPi systemd[1]: Starting Lighttpd Daemon...
Okt 19 21:26:56 DietPi systemd[1]: Started Lighttpd Daemon.
Okt 19 21:27:05 DietPi systemd[1]: Reloading Lighttpd Daemon.
Okt 19 21:27:05 DietPi systemd[1]: Reloaded Lighttpd Daemon.
```

**cat /var/log/lighttpd/error.log:**
```
no output
```

**journalctl -u pihole-FTL:**
```-- Logs begin at Mon 2020-10-19 21:23:00 BST, end at Mon 2020-10-19 22:20:02 BST. --
Okt 19 21:23:13 DietPi systemd[1]: Starting LSB: pihole-FTL daemon...
Okt 19 21:23:13 DietPi pihole-FTL[441]: Not running
Okt 19 21:23:14 DietPi su[473]: (to pihole) root on none
Okt 19 21:23:14 DietPi su[473]: pam_unix(su:session): session opened for user pihole by (uid=0)
Okt 19 21:23:14 DietPi pihole-FTL[441]: FTL started!
Okt 19 21:23:14 DietPi su[473]: pam_unix(su:session): session closed for user pihole
Okt 19 21:23:14 DietPi systemd[1]: Started LSB: pihole-FTL daemon.
```

MichaIng, 2020-10-19:
When using `pihole -r`, did you select to reconfigure Pi-hole, and if so, did you choose to install/setup the Lighttpd?

Basically the DietPi webserver implementation and the one of Pi-hole conflict. The Pi-hole installed sets up a Pi-hole-only system which does not work with any other web application (that uses the webserver) aside of Pi-hole. DietPi sets up things so that one can run multiple web applications concurrently on the same webserver, and, it allows to install Pi-hole with Nginx and Apache2 as well. So it is important to never choose installing Lighttpd from within its installer. I recently added this info to our online docs: https://dietpi.com/phpbb/viewtopic.php?p=174#p174
We call the installer with the `--disable-install-webserver` flag which omits the question.

`sudo dpkg -P lighttpd` purged our webserver config as well and with `sudo apt install --reinstall lighttpd` you installed the package default which does not work with Pi-hole, if not configured by the Pi-hole installer or the `dietpi-software`.

But all not difficult to repair, can you please paste:
```
cat /etc/pihole/setupVars.conf
```

jimmyhenne, 2020-10-19:
Okay, I think I got your explanation.
When ran `pihole -r`, I just chose repair, not reconfigure.
**cat /etc/pihole/setupVars.conf**:
```
WEBPASSWORD=************
DNSMASQ_LISTENING=single
DNS_FQDN_REQUIRED=false
DNS_BOGUS_PRIV=false
DNSSEC=false
REV_SERVER=true
REV_SERVER_DOMAIN=fritz.box
REV_SERVER_TARGET=192.168.178.1
REV_SERVER_CIDR=192.168.178.0/24
BLOCKING_ENABLED=true
PIHOLE_INTERFACE=eth0
IPV4_ADDRESS=192.168.178.37/24
IPV6_ADDRESS=fd00::ba27:ebff:fefe:791b
PIHOLE_DNS_1=127.0.0.1#5353
PIHOLE_DNS_2=
QUERY_LOGGING=false
INSTALL_WEB_SERVER=false
INSTALL_WEB_INTERFACE=true
LIGHTTPD_ENABLED=true
```

The discourse topic is from me 😊

MichaIng, 2020-10-19:
Yes `repair` would be the first attempt, if that fails, `reconfigure` could be tested, but when it's asked about the webserver, "No" must be selected. But I see you did that, or chose `repair` since setup variable `INSTALL_WEB_SERVER=false` is still correct 🙂.

> The discourse topic is from me 😊

Yes, I'm subscribed to `#dietpi` tag on that forum, but was too slow to answer there before you opened the issue here 😄.

`WEBPASSWORD=************` oh, the admin panel password is stored in plain text there? Good to know, maybe that should be stored hashed 🤔.

Back to topic, please try to reinstall Lightpd + Pi-hole through dietpi-software:
```
dietpi-software install 84 93
```

jimmyhenne, 2020-10-19:
Don't get me wrong concerning the panel password. It's not stored in clear text, but in a encrypted way. I just wasn't sure if one can decrypt it, so I simply overwrote it 😁 
Will now reinstall and report..

jimmyhenne, 2020-10-19:
I did `dietpi-software reinstall 84 93` and rebootet the Pi. Entering `pi.hole/admin/` (resp. I'm using a shortcut to URL http://pi.hole/admin/ for accessing the dashboard) in Firefox still results in 403 Forbidden but then I tried accessing the dashboard by entering the IP address `192.168.178.37/admin/` and that is working again. So it seems that pi.hole is not redirected correctly. But as long as everything else runs flawless, and it seems so, I'm fine with that and just change my shortcut to the IP address.

Joulinar, 2020-10-29:
Hi guys, something still open on this or can it be closed?

jimmyhenne, 2020-10-29:
Sorry, can/will be closed.

closed 2020-10-29 by jimmyhenne

ravenclaw900, 2021-01-08:
I'm basically having the same problem, though with an easy workaround. Putting in `192.168.1.254` or `pi.hole` gives me the 403 Forbidden, but adding `/admin` to the end of either url let's me access the page.

Joulinar, 2021-01-08:
I guess this is as expected. There should be no `index.php`, `index.html`, `index.lighttpd.html` inside `/var/www` resulting in the 403 message.

MichaIng, 2021-01-08:
Exactly. Automatic redirection to the blocking page, which then shows a link to the admin panel, only works when using the blocking page as 404 handler, which again only makes sense when enabling the blocking page. But it neither is that recommended (performance, server load) nor does that work in all cases (only for non-HTTPS resources) and hence the blocking page will be removed with next Pi-hole update.

We could add an automatic redirect when the `pi.hole` host is used without path element?

ravenclaw900, 2021-01-08:
That is the usual usage of it, so I think that would be a good idea.