MichaIng/DietPi #3841: DL Allo Gui Image Bugs

Rickyrivits, 2020-10-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux DietPi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (armv7l)

- Power supply used | (LABIsts)
- SDcard used | (EG: SanDisk extreme)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
Freshly updated
- Can this issue be replicated on a fresh installation of DietPi?
Yes

                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              RPi 4 Model B (armv7l) | IP: 10.0.0.214 DietPi-Config  Please select an option:                                                                                                                                                         Mode     : [CPU Burn: (CPU only)]                                             Duration : [1 Minutes]                                                        Start    : Launch the stress test                                                                                                                                                                                                        <Ok>                         <Back>                                                                                                                                                                                   Mode     : [CPU Burn: (CPU only)]Duration : [1 Minutes]           Duration : [1 Minutes]           Start    : Launch the stress test

[ INFO ] DietPi-Config | Checking for required APT packages: stress

[ INFO ] DietPi-Config | Flagged for install: stress

[ INFO ] DietPi-Config | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Reading package lists...
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 38d 6h 19min 42s). Updates for this repository will not be applied.
E: Release file for https://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 35d 10h 54min 24s). Updates for this repository will not be applied.

[FAILED] DietPi-Config | APT update
 - Command: apt-get -q update
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 RPi 4 Model B (armv7l) | IP: 10.0.0.214 | Use up/down buttons to scroll text APT update
  - Command: apt-get -q update
  - Exit code: 100
  - DietPi version: v6.32.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 2 |
 DISTRO: 5
  - Image creator: DietPi Core Team
  - Pre-image: Raspbian Lite
  - Error log:
 Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
 Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
 Reading package lists...
 E: Release file for
 http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid
 yet (invalid for another 38d 6h 19min 42s). Updates for this repository will
 not be applied.

    Retry          : Re-run the last command that failed                 
    Open subshell  : Open a subshell to investigate or solve the issue   

                     <Ok>                         <Exit>
                                                                               E: Release file forRetry          : Re-run the last command that failed               Retry          : Re-run the last command that failed               Open subshell  : Open a subshell to investigate or solve the issue Open subshell  : Open a subshell to investigate or solve the issueSend report    : Uploads bugreport containing system info to DietPi

[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...

[ INFO ] DietPi-Bugreport | Checking URL: ssh.dietpi.com, please wait... (1/2)
[... 106 lines cut ...]
                                                                              Send report    : Uploads bugreport containing system info to DietPiSend report    : Uploads bugreport containing system info to DietPi<Ok><Ok><Exit>

[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

Should run test
- ...

#### Actual behaviour

See error report aforementioned
- ...

#### Extra details

I used "DL Allo Gui Image"
- ...

Rickyrivits, 2020-10-19:
https://github.com/MichaIng/DietPi/issues

closed 2020-10-19 by Rickyrivits

Joulinar, 2020-10-19:
Hi,

even if you closed it. Pls check your time/date settings. It seems your system time is not up-to-date