MichaIng/DietPi #3842: Medusa broken in version 6.33.3

Luan7805, 2020-10-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.33.3
- Distro version | buster 
- Kernel version | Linux DietPi 5.4.51-v8+ 1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- SBC model | RPi 3 Model B (aarch64)

#### Additional Information (if applicable)
Install broken in dietpi 6.33.3.
Work perfect in dietpi 6.33.2

#### Steps to reproduce
1. I did a clean install.
2. install Medusa

#### Expected behaviour
Medusa work

#### Actual behaviour
The process does not start

Joulinar, 2020-10-20:
Hi,

pls do following as user `root`

```
G_CONFIG_INJECT 'Group=' 'Group=dietpi' /etc/systemd/system/medusa.service
systemctl daemon-reload
systemctl restart medusa.service
systemctl status medusa.service
```

Luan7805, 2020-10-20:
Now works perfectly 👌

MichaIng, 2020-10-20:
Quick fix:
```
sed -i 's|:]]Group=/#Group=/|:]]*Group=/#Group=/|' /boot/dietpi/dietpi-software
```
This has the benefit that permissions through subsequent group memberships are still given. E.g. if group permissions of group `medusa` are required, `Group=dietpi` breaks this (and the other way round) as is not only overrides the primary group but also all subsequent group memberships.

aermak, 2020-12-22:
DietPi v6.34.3 : 16:00 - Tue 22/12/20

DietPi-Services
─────────────────────────────────────────────────────
 Mode: restart medusa

[FAILED] DietPi-Services | restart : medusa

─────────────────────────────────────────────────────
root@DietPi:~# systemctl status medusa.service
● medusa.service - Medusa Daemon
   Loaded: loaded (/etc/systemd/system/medusa.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-12-22 15:59:36 CET; 2min 28s ago
  Process: 1662 ExecStart=/usr/bin/python /mnt/dietpi_userdata/medusa/start.py -q --daemon --nolaunch --datadir=/mnt/dietpi_userdata/medusa (code=exited, status=1/FAILURE)

Dec 22 15:59:36 DietPi python[1662]:     from feedparser.api import parse
Dec 22 15:59:36 DietPi python[1662]:   File "/mnt/dietpi_userdata/medusa/ext/feedparser/__init__.py", line 28, in <module>
Dec 22 15:59:36 DietPi python[1662]:     from .api import parse
Dec 22 15:59:36 DietPi python[1662]:   File "/mnt/dietpi_userdata/medusa/ext/feedparser/api.py", line 30, in <module>
Dec 22 15:59:36 DietPi python[1662]:     import urllib.parse
Dec 22 15:59:36 DietPi python[1662]: ImportError: No module named parse
Dec 22 15:59:36 DietPi systemd[1]: medusa.service: Control process exited, code=exited status=1
Dec 22 15:59:36 DietPi systemd[1]: Failed to start Medusa Daemon.
Dec 22 15:59:36 DietPi systemd[1]: medusa.service: Unit entered failed state.
Dec 22 15:59:36 DietPi systemd[1]: medusa.service: Failed with result 'exit-code'.

MichaIng, 2020-12-22:
@aermak 
Many thanks for your report. Since it is a different error on different DietPi version, could you open a new issue, please?

closed 2020-12-22 by MichaIng