MichaIng/DietPi #3844: dietpi software install stuck when installing software (pivpn)

kelliegator, 2020-10-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster
10.6

- Kernel version | `uname -a`
Linux DietPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
Standard 5V power supply.

- SDcard used | (EG: SanDisk ultra)
Sandisk 16GB Ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
PiVPN

- Was the software title installed freshly or updated/migrated?
Installed freshly.

- Can this issue be replicated on a fresh installation of DietPi?
I don't think so.

- Bug report ID | `echo $G_HW_UUID`

Doing a bugreport just resulted in this screen, so I don't know if I'm doing anything wrong.
[image]

#### Steps to reproduce

1. Install PiVPN with OpenVPN.
2. Uninstall PiVPN.
3. Install it again (with the intent to install Wireguard instead).

#### Expected behaviour

The installation should go normally and let me install PiVPN again but with Wireguard instead.

#### Actual behaviour

The installation gets stuck at "./install.bash".  I've even tried to install this software and leave the computer on overnight, but when I checked the next morning it was still stuck at the same step, so I don't know what that is about.

#### Extra details
I had originally installed PiVPN with OpenVPN, but I uninstalled 'cause I wanted to reinstall PiVPN try Wireguard instead, but every time I try to install PiVPN it gets stuck. I also tried just installing Wireguard and it was very slow, but I didn't have time to wait and check if it also got stuck like with PiVPN. I'm a little worried that I've broken my Pi, or something. I'm not the most tech savvy in the world even though I've learned a lot over time.

But everything else with my Pi seems to work fine, like Nextcloud and samba, so I don't know what to believe.

I hope I've been able to provide all the necessary info. If I've missed something, feel free to point it out.

Joulinar, 2020-10-20:
Hi,

many thanks for your message. Pls can you have a look to your system time? Usually certificate issues happen if system time is not up-to-date. But this did not explain the issue with installation of PiVPN. Pls can you start installation again and let it run for a couple of minutes. Open a 2nd SSH connection and check if there is a `whiptail` process

`ps -ef |grep whiptail`

If you have such a process, PiVPN installer is waiting on user input.

At least on my very quit testing, it seems the SSH session was hanging. I hit a couple of times "ENTER" key and out of the sudden, the `whiptail` was showing up 🤔

kelliegator, 2020-10-20:
> Hi,
>
> many thanks for your message. Pls can you have a look to your system time? Usually certificate issues happen if system time is not up-to-date. But this did not explain the issue with [...]

Hey. I feel really stupid here but I actually don't quite know how to check the system time on my Dietpi. I'm still a bit of a newbie at this.

As for whiptail, this is the result I got:

`root     10253  9393  0 15:52 pts/0    00:00:00 whiptail --msgbox --backtitle Welcome --title PiVPN Automated Installer This installer will transform your Raspberry Pi into an OpenVPN or WireGuard server! 20 70
root     10432 10404  0 15:52 pts/1    00:00:00 grep whiptail`

MichaIng, 2020-10-20:
This is a bug our side, we wrapped the installer into an error handler but forgot to keep printing it's output to console.
This fixes it: #3845

Joulinar, 2020-10-20:
@kelliegator 
as workaround, you cold continue hitting "ENTER" key. This should move you through the installer once stuck. At least on my testing, it was needed to hit "ENTER" 6,7,8 times before it's going to show up. 😃

MichaIng, 2020-10-20:
Changelog: https://github.com/MichaIng/DietPi/commit/de269e92ca55e0459ef4ec92ca203beb7b58daf5
Quick fix:
```
sed -i 's|^[[:blank:]]*G_EXEC_NOEXIT=1 G_EXEC ./install.bash|\t\t\tG_EXEC_OUTPUT=1 G_EXEC_NOEXIT=1 G_EXEC ./install.bash|' /boot/dietpi/dietpi-software
```

Joulinar, 2020-10-20:
thx @MichaIng 
even better than hitting "ENTER" on a blank screen 🤣

MichaIng, 2020-10-20:
MOTD patch added (and removed all the obsolete pre-v6.33 ones 😉):
```
[[ -w '/boot/dietpi/dietpi-software' ]] && grep '^[[:blank:]]*G_EXEC_NOEXIT=1 G_EXEC \./install\.bash' /boot/dietpi/dietpi-software &> /dev/null && sed -i 's|^[[:blank:]]*G_EXEC_NOEXIT=1 G_EXEC \./install\.bash|\t\t\tG_EXEC_OUTPUT=1 G_EXEC_NOEXIT=1 G_EXEC ./install.bash|' /boot/dietpi/dietpi-software &> /dev/null
```

Joulinar, 2020-10-20:
yes, I have seen a couple of reports regarding `iptables-persistent` as well on PiVPN side. However it's usually working if your run the installer again.

MichaIng, 2020-10-20:
We could debug this, running manual `iptables-persistent` install on the machine where the installed failed with this error, then step-wise applying the environment of the PiVPN installer 🤔.

kelliegator, 2020-10-21:
> Changelog: [de269e9](https://github.com/MichaIng/DietPi/commit/de269e92ca55e0459ef4ec92ca203beb7b58daf5)
> Quick fix:
> 
> ```
> sed -i 's|^[[:blank:]]*G_EXEC_NOEXIT=1 G_EXEC \./install\.bash|\t\t\tG_EXEC_OUTPUT=1 G_EXEC_NOEXIT=1 G_EXEC ./install.bash|' /boot/dietpi/dietpi-software
> ```

This worked for me. Thanks!

mgagnequebec, 2020-12-14:
6.33.3 - this just happened to me when selecting multiple installs (netdata, rpi-monitor, webmin). The Enter key trick got me through the first package but then I got this: 

https://imgur.com/a/Bi7mu2J

I know these screenshots are not great, but I chose the option "send log to the dev" also. 

while it was stuck, I also tried to ssh with my other pc that's on the same network/subnet  (to try the "ps -ef |grep whiptail" command) but it timed out after asking me for the ssh password (using openssh, not bear)

Not sure what else I can put in this comment, but I hope this helps

Joulinar, 2020-12-14:
Hi,

your screen print is more a network issue as you are not able to connect to `raspbian.raspberrypi.org`. This could be a local challenge or some temporary issue on Raspbian side.

mgagnequebec, 2020-12-14:
strange, i can ping google just fine in the terminal. must be something idk how to fix or, yea " temporary issue "

closed 2020-12-20 by MichaIng