MichaIng/DietPi #3847: Update from v6.28.0 to 6.33.3 fails on RaspberryPi B

JanolovV, 2020-10-20:
Hi,

Update from v6.28.0 to 6.33.3 fails on RaspberryPi B

The update looked fine until unpacking of python 3.7 (3.7.3-2+deb10u2) over (3.7.3-2)
dpkg: error processing archive /tmp/apt-dpkg-install-PicVGF/06-python3.7_3.7.3-2+deb10u2_armhf.deb (--unpack):tee: /var/tmp/logs/dietpi-software_apt.log: Read-only file system

Restart and rerun of update gave the message: run 'dpkg --configure -a' to correct the problem. I did so, restarted and got back to the same message again. 

System is running on 32 MB SD-card not write protected (checked).

Thank you

Jan-Olov

[image]
[image]
[image]

Joulinar, 2020-10-20:
Hi,

many thanks for you report. Looks like one of the partitions is set to r/o. Can you run following

`lsblk -o name,fstype,label,size,ro,type,mountpoint`

MichaIng, 2020-10-20:
Please also check for file system errors, as those can lead to root being remounted R/O:
```
dmesg -l emerg,alert,crit,err
```

JanolovV, 2020-10-21:
Thank you for your help,

I've done as you described and here is a picture of the result. It looks ok
to me but I'm no expert in Linux.

Thank you!

Jan-Olov
[image: 20201021_094301.jpg]

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfritt.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Joulinar, 2020-10-21:
Hi,

I don't see any pictures. Pls try to attach on GitHub directly. Thx

JanolovV, 2020-10-21:
Sorry, Here's the picture

[image]

Joulinar, 2020-10-21:
pls can you try running `apt update && apt upgrade`. pls run same 2 commands if it fails again with same error regarding r/o file system. Pls don't do a reboot in between

JanolovV, 2020-10-21:
Do you mean both commands on the same line or one command first and the other when the first is executede?

MichaIng, 2020-10-21:
Both works, the `&&` is shell syntax to run the first command first, and only if succeeds, the second command runs. This makes sense since when updating the package list fails already, that should be fixed before attempting to upgrade the packages 😉.

JanolovV, 2020-10-21:
apt update worked fine
apt upgrade gave an errormessage: dpkg interrupted. You have to run dpkg --configure -a manually to correcty the problem
When that is done is says "Please reboot the system when convinient"

Joulinar, 2020-10-21:
pls do `dpkg --configure -a`

JanolovV, 2020-10-21:
I did the dpkg and thought it was finished when I saw the reboot comment. But while I was writing the last message it went on with execution of dpkg. Now I've done the upgrade as well and it looks a lot better. Unsatisfied dependencies has to be corrected with apt --fix-broken install. I do that now.

Joulinar, 2020-10-21:
ok, at the end `apt update && apt upgrade` should be running smoothly. If this finished and no further packages are open, you could continue running `dietpi-update`.

JanolovV, 2020-10-21:
Sorry, it does not look good at all. It stopped as before when running the apt upgrade asking for the dkpg... When running the dkpg this time I got lots of errors indicating the disc is read only. Can it be a corrupted SD-card? I guess you have a good suggestion how to check it?

Joulinar, 2020-10-21:
Could be a corruption of the SD card. You could have a check running following

`fsck -a /dev/mmcblk0p2`

JanolovV, 2020-10-21:
I get the following:
fsck from util-linux 2.33.1
/dev/mmcblk0p2 is mounted.
e2fsck: Cannot continue, aborting

JanolovV, 2020-10-21:
I made a test with "badblocks -v /dev/mmcblk0p2" which found no errors.

Looks lika I'm stuck. Upgrading and testing the disk gets me nowhere. On the other hand, the system is up running and I don't see any other malfunctions yet. A small installation with sqlite, apache, some python and some php. Would feel better though if the update worked.

MichaIng, 2020-10-22:
Strange, no errors, `lsblk` does not report R/O, other parts of the system work well but only APT (including our wrapper function) face a R/O root file system?

Let's check another way, please run and paste: `mount`

And try to manually write to those locations:
```
> /testfile
rm /testfile
> /var/tmp/dietpi/logs/testfile
rm /var/tmp/dietpi/logs/testfile
> /var/share/testfile
rm /var/share/testfile
```

JanolovV, 2020-10-22:
Hi,

That all works fine, only I don't have a share directory in /var.

[image]

MichaIng, 2020-10-22:
Ah sorry, a typo from my side, should have been `/usr/share/testfile`.

Btw just to have an `fsck`, this needs to be done on reboot:
```
> /forcefsck
reboot
```

If `apt upgrade` then still fails, lets test if manual install via `dpkg` fails the same of if something with `apt` is broken. For that please show an example package that failed, e.g.:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/python3.7_3.7.3-2+deb10u2_armhf.deb
dpkg -i --force-depends python3.7_3.7.3-2+deb10u2_armhf.deb
rm python3.7_3.7.3-2+deb10u2_armhf.deb
```

JanolovV, 2020-10-22:
Hi,

apt aupgrade failed as before but the last try with dpkg went right through with no issues.

MichaIng, 2020-10-22:
Let's try to reinstall apt itself:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-pkg5.0_1.8.2.1_armhf.deb
dpkg -i apt_1.8.2.1_armhf.deb libapt-pkg5.0_1.8.2.1_armhf.deb
rm apt_1.8.2.1_armhf.deb libapt-pkg5.0_1.8.2.1_armhf.deb
```

JanolovV, 2020-10-22:
OK I will do that but not tonight. It is well past 10 PM here and I've had a long day...

JanolovV, 2020-10-23:
Restart today does not look good at all. This looks like a hard drive failure to me:

[image]

Joulinar, 2020-10-23:
could be, at least it would fit to your issues with the r/o file system messages

If possible, you could mount your SD card into another Linux box and run the `fsck` command there

JanolovV, 2020-10-23:
Sorry, I don't have another Linux machine, it's all Windows. Only chance is if it works with the linux box in Windows 10, I don't really know but I doubt it. I will try anyway.

JanolovV, 2020-10-23:
I can dig one out, whats the plan?

Joulinar, 2020-10-23:
1. remove 1st current SD card from RPi
2. take 2nd SD card and flash DietPi on it
3. place 2nd SD card and boot RPi
4. use an USB connector to connect 1st SD card to USB port
5. perform `fsck` on the 1st SD card connected to USB port 

😃

JanolovV, 2020-10-23:
OK, working on it

JanolovV, 2020-10-23:
Sorry for the delay but those SD-cards are quite slow...
When I run fsck on /dev/sda2 which is the active partition of the first disk I get "clean, 57158/941616 files, 898746/3822976 blocks" Thats all.

MichaIng, 2020-10-23:
Can you try:
```
fsck -f /dev/sda2
```

JanolovV, 2020-10-23:
fsck -f resulted in this:

[image]

I'm also running fsck -c and so far it found (2/0/0) errors ~50% done

JanolovV, 2020-10-23:
I don't see any signs of poor contact surfaces and the card sits ok in the slot. Now with errors found it sits in a card reader.

MichaIng, 2020-10-23:
The errors it found are related to a previous corruption, so at least it is possible that the previous corruption happened due to bad contact (or any other issue with SD card in slot) and now the remains of this are found.

According to the bad blocks check, `(2/0/0)` means that it found two bad blocks? Good idea to run `fsck -c`, seems to be the recommended way to check for bad blocks and have them added to a special inode so that they are not used anymore for common file/directory writes.

JanolovV, 2020-10-23:
There was a lot of errors and after a restart I tried the apt update/apt upgrade again and it failed this time to.

I think this has come to the end of this road. It is my experience that if you feel like you are up to your knees in shit, stop digging. Disk problem have a tendency to grow with the effort you put into solving hem.

Now I will try to save what I can from this disk and make a fresh install. Any suggestions that makes the process easier?

I also want to thank you for your kind help. You have been very supportive and I appreciate it very much. Thank you!

MichaIng, 2020-10-23:
> It is my experience that if you feel like you are up to your knees in shit, stop digging. Disk problem have a tendency to grow with the effort you put into solving hem.

Something I still need to learn 😅.

> Now I will try to save what I can from this disk and make a fresh install. Any suggestions that makes the process easier?

At least `/mnt/dietpi_userdata` can be copied as is, same with `/boot/dietpi.txt`. The good thing: If there is a corrupted file, it won't be copied, even if data/config files are missing, most software would recreate them, so that is usually safe to do.

Otherwise as usual: In `/etc` you find package configs. If you configured one heavily, could be worth it to copy the configs over, same for `/var/lib` where you find package data.

Copy everything in place on the new system **as early as possible**, especially **before doing any install** via `dietpi-software`. Most APT/DEB packages and as well DietPi software installs prevent customised configs and preserve/migrate existing data.

And of course, it's always good to keep the old SD card around for a while, so if you find anything missing, you have a change to recover it.

Joulinar, 2020-10-29:
@JanolovV 
did you managed to get a new/fresh installation done?

JanolovV, 2020-10-29:
Yes I did. I got everything up and running as before so thank you again for your help. But I ran into some other problems. I am going to have my RaspberryPi as the spider in a network of measuring equippment communicating over short range radio (nRF24L) and I got problem with the drivers and example code. The suggestion was to run "make" which didn't work at all since my installation was missing the make command...

After digging around I came to the conclusion that perhaps my Pi is to old. It is version B of the first generation, one of the few maodels out of production, with an old processor so probably the "make" is not compiled for that processor? I don't really know but I decided to by a new Pi and start fresh on that one. I also think that I will go for Raspbian. Just beacuse it is more main stream. Sometimes it is wise to go by the rule of the big numbers. 

But I think I will keep the Dietpi on the old one. I like the small footprint and it has been working well until the failed update which really was a disk problem.

Maybe this, in  the end, will make me run more of Linux. I'm really a Microsoft guy beacuse I've used most of there operating systems since DOS 3.0. Simply beacuse that's what we had at work. I've been doing some Linux off and on since the late nineties but it's been more off than on. If you do anything with a computer you run inte problems sooner or later. A problem that take a day or more for me to solve in Linux I can fix on a coffe break i Windows. But in the Pi I don't have the choice and I'm forced to learn.

Keep up the good work!
Jan-Olov

Joulinar, 2020-10-29:
well that's the idea behind DietPi, to have a system running with a reduced amount of software packages. That's why it is called **Diet**Pi 😉 

By default DietPi contains software components only, that are needed to boot your system. Everything else would need to be installed in addition and on demand. This include `make` command es well. On DietPi you could select `Build-Essentials` package from `dietpi-software` catalogue. This should install `make` as well as everything else needed to compile stuff. 😄

MichaIng, 2020-10-29:
Generally DietPi is a minimal Debian (Raspbian in case of RPi 32-bit images), which means that you have the full Debian APT repository available with > 51,000 software packages available for easy install on (mostly) all devices and architectures. Often, if a command is missing, there is a package which is named by the command or contains it, and there is an online library to search through them, like in case of `make`: https://packages.debian.org/make

If no package matches the command, then you can find it out through the manpage: https://manpages.debian.org/autoreconf
=> At the top you see `/ buster / autoconf / autoreconf(1)` which means this command `autoreconf` is available in the Debian `Buster` via package name `autoconf`.

JanolovV, 2020-10-30:
Of course you are right, it's all there if you know how to. For you guys who handle this daily it's a piece of cake. But the problem is I don't have that knowledge and every time anything is missing or does not work as I expected it's work and time consuming finding out how to solve it. That's why I decided to go for Raspbian for my project, to save time and work for me. But as I wrote, I like the small footprint of Dietpi and will continue using it. Probably, as I get more into it, I will be able to use it in the way you describe. It's me and my limited knowledge that is the problem, not the distribution.

And I really appreciate the answers you give me here, I  will give it a try and see where it leads. To be honest I didn't expect to get the amount of help I got here. You are both engaged and pedagogic and seem to have all the time in the world to solve my problems. I'm not used to that. Usually you communicate with a guy who is trying his best to convince you that it's your own fault if anything goes wrong. Code 30, the problem sits 30 cm from your keyboard. In this case it might me right. By the way, where in the world are you sitting? I'm in Kalmar, Sweden.

MichaIng, 2020-10-30:
No complains about limited knowledge, we all started at zero at some point 🙂. This is why I added this hopefully useful basic info that also helps on Raspbian and any other Debian-based distribution, like Ubuntu etc. Those have more packages pre-installed but never all of them of course. Raspbian Lite not even has `make`, AFAIK? But if so, when compiling software (what `make` does) one quickly ends up with missing development headers or other build tools which brings you back to searching the package repository for those 😉.

DietPi actually explicitly addresses users with less Linux knowledge, by providing ready-to-run software installs that automate all the dependency installs, compiling, if required and reasonable setup. The lack of a pre-installed desktop (although very easy to do) makes it a bid harder compared to a desktop OS image when coming from Windows or macOS, but compared to Raspbian Lite you should be able to setup most stuff much quicker and easier. `make` however is for compiling software, which is generally a more advanced thing that often requires more experience. Some instructions may work OOTB but usually there are development headers and libraries required that you need to gather and install manually. So the case where having `make` pre-installed makes the life for Linux beginners easier is veeery rare, which is why we don't have it 😉.

All this might find a place in your new documentation: https://dietpi.com/docs/
Feel free to open an issue/request over there if you miss something, else the above generic "How do I search and find missing commands/packages in the APT repository" will be added definitely.

JanolovV, 2020-11-02:
Hi guys,

After reading your engaged desciption I decided to try with Dietpi in the old RPI and DiePi anyway and with Raspbian in the new, in parallell. Could be interesting to see the difference.

Joulinar, 2020-11-04:
Hi Guys,

is there still something open on this topic or can it be closed?

JanolovV, 2020-11-04:
Hi,

No reason to keep this open anymore. I got more help than expected, thank you again!

/Jan-Olov

closed 2020-11-04 by JanolovV