MichaIng/DietPi #3851: Unexpected behaviour

punktilend, 2020-10-21:
#### Details:
- Date           | Wed Oct 21 12:52:08 EDT 2020
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Archives directory /var/cache/apt/archives/partial is missing. - Acquire (2: No such file or directory)
```
---------------------------------------------------------------------

Joulinar, 2020-10-21:
Hi,

many thanks for the message. You simple can create the missing directory

```
mkdir /var/cache/apt/archives/partial
chown chown _apt:root /var/cache/apt/archives/partial/
chmod 700 /var/cache/apt/archives/partial/
```

Should looks like this at the end
```
root@DietPi4:~# ls -la /var/cache/apt/archives/ |grep partial
drwx------ 2 _apt root     4096 Oct  9 23:38 partial
root@DietPi4:~#
```

MichaIng, 2020-10-21:
Was this a fresh image or did you change something in `dietpi-config` > `Advanced Options` > `APT Cache`?

When APT cache is disabled or moved to RAM (or any other non-default location) and this default cache dir is aggressively cleaned/removed, this can happen. APT cache is disabled by default on DietPi and we have an aggressive clean function but that function assures that this dir is re-created whenever the cache is disabled or in RAM: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-set_software#L248-L253

The partial subdir btw does not need to exist, `apt-get` does this, only the parent dir `archives` needs to, with default 755 root:root permissions, so `mkdir -p /var/cache/apt/archives` or `sudo mkdir -p /var/cache/apt/archives` is enough, ah even `/var/cache/apt` would be enough.

closed 2020-12-17 by MichaIng