MichaIng/DietPi #3853: ssh.dietpi.com | SSL certificate not valid

Joulinar, 2020-10-21:
@MichaIng 

I have seen a couple of issues uploading bug reports the last days. Therefore I checked checked the certificate

```
root@DietPi4:~# wget --spider -t 1 -T 5 ssh.dietpi.com
URL transformed to HTTPS due to an HSTS policy
Spider mode enabled. Check if remote file exists.
--2020-10-21 21:42:11--  https://ssh.dietpi.com/
Resolving ssh.dietpi.com (ssh.dietpi.com)... failed: Connection timed out.
wget: unable to resolve host address ‘ssh.dietpi.com’
root@DietPi4:~# openssl s_client -showcerts -connect ssh.dietpi.com:443
CONNECTED(00000003)
depth=1 C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = CA, CN = protect@DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
verify error:num=19:self signed certificate in certificate chain
verify return:1
depth=1 C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = CA, CN = protect@DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
verify return:1
depth=0 C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = DDoS-Filter, CN = DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
verify return:1
---
Certificate chain
 0 s:C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = DDoS-Filter, CN = DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
   i:C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = CA, CN = protect@DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
 1 s:C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = CA, CN = protect@DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
   i:C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = CA, CN = protect@DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
---
Server certificate
subject=C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = DDoS-Filter, CN = DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain

issuer=C = AA, ST = DDoS-Filter, L = DDoS-Filter, O = DDoS-Filter, OU = CA, CN = protect@DDoS-Filter.domain, emailAddress = protect@DDoS-Filter.domain

---
No client certificate CA names sent
Peer signing digest: SHA512
Peer signature type: RSA
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 2753 bytes and written 432 bytes
Verification error: self signed certificate in certificate chain
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES256-GCM-SHA384
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256-GCM-SHA384
    Session-ID: 21EFD10F49987A63D583D74C97D8B68A489ACAF72F487F92FAAE625CCF2798DE
    Session-ID-ctx:
    Master-Key: DC94F3105279273C926A807470490FF6F503EE41EBB3FB2513FD33A343170BBB6DA63139F9FF7745FB382BF9D407DE58
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 120 (seconds)
    TLS session ticket:
    0000 - d8 f9 4b 25 3a 8f 9f 41-35 fc d3 49 79 86 1a 4f   ..K%:..A5..Iy..O
    0010 - a8 82 4a 6c 90 d4 1e 76-79 20 6d 70 2d cf 09 49   ..Jl...vy mp-..I
    0020 - 0c dd 63 01 93 dc 36 07-06 05 57 ee 1d cf e9 05   ..c...6...W.....
    0030 - ac 1d 7b 26 ef f0 6b 47-e8 f4 13 97 cc d8 fb dd   ..{&..kG........
    0040 - 8c f9 9f 51 fd da d3 10-c4 c7 62 c7 27 85 f2 a2   ...Q......b.'...
    0050 - 51 4e 26 68 14 4d 38 6e-8b 17 f5 94 74 21 90 e2   QN&h.M8n....t!..
    0060 - f9 ee 3f 4f 36 1c c9 38-b5 27 2b 80 01 0a 36 07   ..?O6..8.'+...6.
    0070 - e1 d9 45 10 6c b9 a8 1e-eb 27 04 11 0d fa d2 6f   ..E.l....'.....o
    0080 - 79 bb e2 4e f0 da 93 3d-83 53 c5 75 a6 7a 93 5c   y..N...=.S.u.z.\
    0090 - 5f d2 ab b5 1f 6a 33 88-c2 93 2b d9 e3 1f 26 de   _....j3...+...&.
    00a0 - 7a bd 62 ff 87 93 2f a4-f0 18 8e 75 31 14 2f 23   z.b.../....u1./#
    00b0 - 11 e3 58 73 f1 36 74 4e-2b 0c 2f db f9 ad e2 d1   ..Xs.6tN+./.....

    Start Time: 1603309403
    Timeout   : 7200 (sec)
    Verify return code: 19 (self signed certificate in certificate chain)
    Extended master secret: no
---
read:errno=0
root@DietPi4:~#
```

looks like it's self signed certificate
`Verify return code: 19 (self signed certificate in certificate chain)`

MichaIng, 2020-10-21:
This is the MVS DDoS protection. It kicks in lately, by switching from "Sensor" to "Permanent" mode, not sure if this is the intended way. I think it is triggered by the large number connection done by our new GitHub Actions check for docs and website which a.o. checks all links, i.e. does a large number of requests to our server in a short time.

I disabled it now completely, we're behind CF DDoS protection already with the parent domain, so should be okay.

Joulinar, 2020-10-21:
I guess the issue with the bug report was the part checking the connection before uploading the report

```
Checking URL: ssh.dietpi.com
│  - Command: wget --spider -t 1 -T 10 ssh.dietpi.com
```

and this is done via https 😉

MichaIng, 2020-10-21:
Fixed with: #3855

MichaIng, 2020-10-24:
The DDoS protection is somehow re-enabled automatically. I have to ask MVS how this works and if they could disable their DDoS protection on our VPS completely, on our own risk. At least I hope that the host servers and network is decoupled enough that that even if our VPS is DDoS'ed to death, their infrastructure is not affected at all.

MichaIng, 2020-11-11:
Okay, DDoS protections stays enabled now but the layer 7 proxy on SSL, which was responsible for the invalid certificate, stays disabled. Marking as closed.

closed 2020-11-11 by MichaIng