MichaIng/DietPi #3862: aliasses - difference between NanoPI NEO2 and RPi-Zero-W

maartenlangeveld, 2020-10-23:
v6.33.3

My NanoPI NEO2 has, among others, following aliasses defined after boot:
```
alias l='ls -CF'
alias la='ls -A'
alias ll='ls -l'
ls='ls --color=auto'
```
However my RPi-Zero-W has those aliasses not defined.

Is there a special reason for this difference in behaviour? Is it possible to have those aliasses also defined on my RPi-Zero-W during install?

Joulinar, 2020-10-23:
Looks like the privat alias of @MichaIng 
probably forgot to remove during image creation?

maartenlangeveld, 2020-10-23:
I would like to have them included ! ;-)

MichaIng, 2020-10-24:
There is a difference between the default `/root/.bashrc` shipped by Raspbian images and a few others, probably coming from `debootstrap` itself, not sure, and the default for new users that is coming from `/etc/skel/`. The first is very small with the aliases presented (similar at least) but commented, while the second is huge with a lot of stuff defined and more commented.

We do not touch the file, so which one is present depends on the pre-image and I don't have a preference so never decided to prefer one of those. On my personal systems I have own stuff defined via `/etc/bashrc.d/micha.sh`, so for all users, while the home dirs have no `.bashrc` at all.

If we want to have a consistent default, then we should add it to `/etc/bashrc.d/dietpi.bash`. The benefit is that this is only loaded for interactive shells while `~/.bashrc` is loaded on every bash shell, so as well when doing an SFTP upload, I think rsync remove transfer and some other transfer protocols which include a regular UNIX user authentication and load the users default shell then, where you don't want aliases and colour codes messing with transfer or automated command execution. Of course adding something like `[[ $PS1 ]] || return` can solve it, but `/etc/bashrc.d` has this natively, as it's loaded from `/etc/bash.bashrc` where you find this return code with the default file version.

`ls='ls --color=auto'` is btw only really good when you load some colour set: `eval "$(dircolours)"`
```
2020-10-24 02:11:41 root@micha:/etc/bashrc.d# cat micha.sh
# Add timestamps to every terminal/screen window: \D{%F %T}
PS1='\D{%F %T} ${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '

# Colorize "ls" command
eval "$(dircolors)"
alias ls='ls -A --color=auto'
alias l='ls -lh'
alias ll='l -L'

# Easier APT upgrade alias
alias APT='apt update && apt list --upgradeable && apt full-upgrade'

# Prevent accidental removals
alias rm='rm -RiIv'
alias mv='mv -iv'
alias cp='cp -iv'
```
Since I use [GNU Screen](https://packages.debian.org/buster/screen), the verbosity and timestamps allow me to review my SSH sessions for a long time.

maartenlangeveld, 2020-10-24:
@MichaIng 
Again many thanks for your help. Learning a lot ! Have adopted your settings ! I hereby close this thread, okay?

closed 2020-10-24 by maartenlangeveld