MichaIng/DietPi #3863: btrfs with compress for dietpi-backup

mrbluecoat, 2020-10-24:
### Creating a feature request

Not sure if this is supported but it would be nice to have the transparent compression of btrfs for the dietpi-backup mounted directory. That way transferring to cloud offsite storage and restores would be optimized.

https://btrfs.wiki.kernel.org/index.php/Compression#How_do_I_enable_compression.3F

Joulinar, 2020-10-24:
Hi,

by default `dietpi-backup` is a directory and not mount point. Means, you would need an external `btrfs` formatted device to get this working.

MichaIng, 2020-10-24:
Yes, exactly, so not is not something that would be added to `dietpi-backup` but instead to `DietPi-Drive_Manager`:
- On Btrfs-formatted drives, depending on Linux version, allow to set compression and choose compression method, and if programmer is in mood, compression level where supported.

Not bad to start supporting the special features that make Btrfs a nice file system type 👍.

cocoflan, 2020-10-25:
Good idea!

Maybe an idea to setup a backup system with btrfs with timeshift!

https://github.com/teejee2008/timeshift

MichaIng, 2020-10-25:
We'll keep `dietpi-backup` as slim solution to work with `rsync` only and no compression (aside of what the file system itself does i.e. Btrfs compression). However if someone requires more, additional/alternative backup solutions can of course be installed or added to `dietpi-software`.