MichaIng/DietPi #3864: Reinstalling No Machine

Ianszh, 2020-10-25:
#### Details:
- Date           | Sun Oct 25 07:14:22 GMT 2020
- Bug report     | 54453262-fc65-4629-98c4-5919cf034d0e
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux EnviroPlus 5.4.72+ #1356 Thu Oct 22 13:56:00 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install ./30.deb
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
(Reading database ... 51916 files and directories currently installed.)
Preparing to unpack /tmp/DietPi-Software/30.deb ...
dpkg: error processing archive /tmp/DietPi-Software/30.deb (--unpack):
 new nomachine package pre-installation script subprocess returned error exit status 132
Errors were encountered while processing:
 /tmp/DietPi-Software/30.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2020-10-25:
Many thanks for your report.

Let me see what this post install script is aiming to do.

MichaIng, 2020-10-25:
My goodness, a monster of a script, no chance to quickly identify possible failure reasons...
Please run it with debug flag and paste the output here. This may be a massive amount of output, in case the last ~20 lines should be the relevant ones:
```
bash -x /var/lib/dpkg/info/nomachine.postinst
```

MichaIng, 2020-12-17:
I'll mark this as closed. Feel free to reopen if issue exists.

closed 2020-12-17 by MichaIng