MichaIng/DietPi #3869: /boot/dietpi/func/dietpi-set_swapfile 128 /dev/zram0 fails

hansjuergenmay, 2020-10-26:
#### Details:
- Date           | Tue Oct 27 00:08:37 CET 2020
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux raspi0w 5.4.72+ #1356 Thu Oct 22 13:56:00 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | swapon /dev/zram0
- Exit code      | 255
- Software title | DietPi-Set_swapfile
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
swapon: /dev/zram0: insecure permissions 0644, 0600 suggested.
swapon: /dev/zram0: swapon failed: Invalid argument
```

MichaIng, 2020-10-26:
Many thanks for your report.

Interesting, AFAIK permissions were correct automatically when we implemented this. I'll try to replicate on VM, else it doesn't hurt to re-set the permissions, `600` simply makes sense for RAM content.

MichaIng, 2020-10-26:
This should fix it: #3870

closed 2020-12-20 by MichaIng