MichaIng/DietPi #3871: Automatic Backup schedule

Lycidias93, 2020-10-27:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...No

#### Describe the solution you'd like:

- ... I would like to have an option in dietpi-backup to automate my backup with time settings. autodelete for max backups would be nice too or uploading to cloud  option.

#### Describe alternatives you've considered:

- ... cron...

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

StephanStS, 2020-11-01:
@Lycidias93: Could be timeshift meet your functional requirements?
Up to now, you need a manual installation. 

See:
- https://www.fossmint.com/backup-restore-linux-with-timeshift/
- https://www.linuxlinks.com/essential-system-tools-timeshift-reliable-system-restore-tool-linux/

anvarit, 2021-09-16:
Easiest solution would be to add this per default in the cron.daily script with similar options as the line "DietPi-Sync daily".
If you would add a toggle in the GUI to enable daily backup then the daily cron script can then read this out and execute the /boot/dietpi/dietpi-backup 1 script.
While it is currently possible with manual different steps, I could be made easier this way

MichaIng, 2021-09-16:
While earlier or later we want to merge the two scripts, actually adding this now is simple enough until we find time for the merge. It simply makes sense! I'll add it to next milestone.

MichaIng, 2021-11-03:
A PR is up to add this feature: #4922

closed 2021-11-08 by MichaIng