MichaIng/DietPi #3873: Installing qbittorrent from dietpi-software fails on useradd command

tandy-1000, 2020-10-27:
#### Details:
- Date           | Tue Oct 27 11:13:26 GMT 2020
- Bug report     | 31a1e253-c104-4241-879d-e5ef87729c1e
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCKPro64 (aarch64) (ID=42)
- Kernel version | Linux rockpro64 5.8.16-rockchip64 #20.08.14 SMP PREEMPT Tue Oct 20 22:37:51 CEST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | useradd -rMN -g dietpi -d /home/qbittorrent -s  qbittorrent
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install qbittorrent from the dietpi-software utility

#### Expected behaviour:

- It should install..

#### Actual behaviour:

- It fails on the useradd command, saying the command is not found, if I rerun the command as root, it says qbittorrent is an invalid shell

#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 937: useradd: command not found
```

MichaIng, 2020-10-27:
Many thanks for your report.

Something with your PATH variable was not in order, or for some other reason essential system commands were not found. There should have been `useradd` and `nologin`. Interesting that a rerun solved it, however `$(command -v nologin)` was already expended to empty string, which is why the `-s` argument was lacking the expected `/usr/sbin/nologin` value and instead the user name `qbittorrent` was the next.

Not sure if it's worth it to do something about the script/function to at least allow the rerun to succeed, e.g. by hard-coding `/usr/sbin/nologin`. But actually I like the possibility to customise the nologin-message or action by creating `/usr/local/sbin/nologin` 🤔.

However, long story short: A simply rerun of the install should fix it. You could assure first that the commands are there:
```
command -v useradd
command -v nologin
```
And if something similar happens again, we'd need to have a closer look what happens to your environment or root mount so that those commands are not found temporarily.

tandy-1000, 2020-10-27:
no worries, thank you for your work on this software. a rerun did not fix the problem, however I have just managed to fix it now. I was in `su`, when I exited I was able to install the program without errors. thank you

Joulinar, 2020-10-27:
mean you switched from user `dietpi` to `root` using `su` and there it was not working?

MichaIng, 2020-10-27:
Ah that all makes sense. Never use `su` for anything, always use `sudo`. If you need an interactive shell, use `sudo -s`. `su` by default passes the whole environment of the current users session, including the limited `PATH` but also overrides, which potentially is an attack vector when someone managed pass a faulty variable to your unprivileged user environment that is then passed to gain root privileges. And `su` does not produce clear log entries to see which originating user was gaining root privileges when, but with `su` its like `root` itself logged in regularly.

I was thinking to force a default PATH with all our scripts in the past, as theoretically it could have been broken with other ways and none of our scripts requires custom entries 🤔.

Joulinar, 2020-10-29:
@MichaIng 
do you like to keep the issue open or can it be closed?

MichaIng, 2020-10-29:
I now added the mentioned idea: https://github.com/MichaIng/DietPi/commit/a1f7bb928d67d42414040440ffc63b42bbb4306e
There are a bunch of other thinkable issues but this workaround has no downsides that I can think of. We want and need all commands from those locations in this order. We don't need any custom locations like often used for Go or Rust builds if a admin really needs to override a system command system wide, this needs to be done via `/usr/local/(s)bin` anyway to assure it's loaded in all cases, since especially `sudo` (without `-E`) does not pass `PATH` or anything else.

closed 2020-12-20 by MichaIng