MichaIng/DietPi #3874: OpenSSH Sftp connection cannot be established.

IamLunchbox, 2020-10-27:
### Creating a bug report/issue
Dear Dietpi-Team,
i cannot access any other filesystem via sftp using OpenSSH installed on a current Dietpi-Build on a Raspberry 4.  

#### Required Information
DietPi version
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version
`10.6`
- Kernel version
`Linux DietPi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux`
- SBC model |
Rpi4
- Power supply used / SDcard used 
Unkown / irrelevant?

#### Additional Information (if applicable)
- Software title
```
ssh -V 
OpenSSH_7.9p1 Raspbian-10+deb10u2, OpenSSL 1.1.1d  10 Sep 2019
```
- Was the software title installed freshly or updated/migrated?
I used sftp connections to backup a public-facing server. The most recent timestamp of the backup-files is 26.08.2020.
- Can this issue be replicated on a fresh installation of DietPi?
I can ssh with the dietpi to every other system, sftp connections from others systems between them succeed all the time. Only sftp seems broken. Using sftp -vvv -i the/cert/file anon@IP-ADRESS generates the following output:
```
OpenSSH_7.9p1 Raspbian-10+deb10u2, OpenSSL 1.1.1d  10 Sep 2019
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug2: resolve_canonicalize: hostname ANONYMOUS is address
debug2: ssh_connect_direct
debug1: Connecting to ANONYMOUS [ANONYMOUS] port 22.
debug1: Connection established.
debug1: identity file scripts/script_auth.key type -1
debug1: identity file scripts/script_auth.key-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_7.9p1 Raspbian-10+deb10u2
debug1: Remote protocol version 2.0, remote software version OpenSSH_7.9p1 Debian-10+deb10u2
debug1: match: OpenSSH_7.9p1 Debian-10+deb10u2 pat OpenSSH* compat 0x04000000
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to ANONYMOUS:22 as 'ANONYMOUS'
debug3: hostkeys_foreach: reading file "/home/dietpi/.ssh/known_hosts"
debug3: record_hostkey: found key type ECDSA in file /home/dietpi/.ssh/known_hosts:3
debug3: load_hostkeys: loaded 1 keys from ANONYMOUS
debug3: order_hostkeyalgs: prefer hostkeyalgs: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group14-sha256,diffie-hellman-group14-sha1,ext-info-c
debug2: host key algorithms: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,ssh-ed25519-cert-v01@openssh.com,rsa-sha2-512-cert-v01@openssh.com,rsa-sha2-256-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,ssh-ed25519,rsa-sha2-512,rsa-sha2-256,ssh-rsa
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group14-sha256,diffie-hellman-group14-sha1
[... 110 lines cut ...]
debug1: Sending env LC_ALL = C.UTF-8
debug2: channel 0: request env confirm 0
debug3: send packet: type 98
debug3: Ignored env SSH_CLIENT
debug3: Ignored env SSH_CONNECTION
debug3: Ignored env SSH_TTY
debug3: Ignored env SHLVL
debug3: Ignored env PWD
debug3: Ignored env OLDPWD
debug3: Ignored env _
debug1: Sending subsystem: sftp
debug2: channel 0: request subsystem confirm 1
debug3: send packet: type 98
debug2: channel_input_open_confirmation: channel 0: callback done
debug2: channel 0: open confirm rwindow 0 rmax 32768
```

#### Steps to reproduce
Try to connect to an openssh-server using sftp interactively or script based.

#### Expected behaviour
An interactive session opens or files are transferred.

#### Actual behaviour
The connection stalls after "Sending subsystem: sftp"

MichaIng, 2020-10-27:
Many thanks for your report.

We pull OpenSSH on RPi (32-bit image) from Raspbian which should match the one from the Debian repository. Last update of the `openssh-client` package was in January, so your script worked well afterwards.

The last lines look like something needs to be confirmed:
```
debug2: channel 0: request subsystem confirm 1
debug3: send packet: type 98
debug2: channel_input_open_confirmation: channel 0: callback done
debug2: channel 0: open confirm rwindow 0 rmax 32768
```

Searching for a stall/hang at this point leads to a large amount of results, best guess of what is close: https://stackoverflow.com/questions/5929552/ssh-command-execution-hangs-although-interactive-shell-functions-fine
In this case it was a `.bashrc` entry that was executed during SFTP connection. This is generally something dangerous since `~/.bashrc` is executed even for non-interactive shells, even that it is stated differently often. Some defaults have something like `[ -z "$PS1" ] && return` at the top of the script to not execute something on non-interactive shell access like SFTP or rsync and some others, but e.g. the Raspbian > DietPi default doesn't have it. Since you tested it with different target systems (?) it is probably not the reason, but at least worth to have a look. And did you try to connect from one of the other systems?

IamLunchbox, 2020-10-28:
Interestingly I use bash and zsh on the mentioned systems. Anyways, without any further interaction by my side (except the hours invested before making this report), the problem doesnt persist. 

I dont know what happened, I didn't reboot and since I disabled the logging mechanism to preserve the SD-card, I have no idea what changed.

closed 2020-10-28 by IamLunchbox

MichaIng, 2020-10-28:
Strange, however great that it works.

For the future these scripts which are automatically loaded on shell sessions is what I would have a look at when getting a similar issue. `zsh` has a bunch of more of these scripts, loaded at different stages during shell init. SSH and SFTP and SCP load the default shell of the user first, then in case the forced command from that shell, e.g. `sftp`/`sftp-server` or `scp` commands. In very most cases it is not wanted that those shell init scripts are loaded on such kind of connections where no interactive shell is wanted but only a specific command executed. So one needs to take care that they are not, which is usually done via `[ -z "$PS1" ] && return` as can be seen in `/etc/profile` and `/etc/bash.bashrc`. Also `/etc/skel/.bashrc` shows an alternative approach by checking the bash flags via `$-` special variable where the character `i` stands for "interactive".

IamLunchbox, 2020-10-28:
Thank you for your advice! I will change my .rc-files in that regard.