MichaIng/DietPi #3875: Ping -c 1 -W 10 dns9.quad9.net failed during Setup

mjhymowitz, 2020-10-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.6

- Kernel version | `uname -a`
Linux DietPi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
AC 100-240V 50/60Hz input
DC 5.1V 3A output
15.3W maximum output power
1.5m 18 AWG captive cable
USB-C connector

- SDcard used | (EG: SanDisk ultra)
don't know, what ever comes in the Raspberry Pi 4 Computer desktop kit

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
dietPi

- Was the software title installed freshly or updated/migrated?
fresh install

- Can this issue be replicated on a fresh installation of DietPi?
I'm using a fresh install so I think so.

- Bug report ID | `echo $G_HW_UUID`
ea661fbe-98b7-4f54-ab78-7b4fdfe7dd28

#### Steps to reproduce

1. flash install to SD using balenaEtcher
2. take out SD and put micro SD in PI
3. power on PI.
4. unplug power from pi
5. plug power into pi 
6. Connect to pi
7. use putty to log into pi and follow setup proccess, dont change any setting

#### Expected behaviour
I exspected to reach a screen to install installation for selected software

#### Actual behaviour
got an error saying can't  ping -c 1 -W 10 dns9.quad9.net and get no responces causing it to fail
rerunning the command also failed

#### Extra details
I opened the command line on my pc and ping dns9.quad9.net. I got a responce, but when I ping -c 1 -W 10 dns9.quad9.net I get Access denied. Option -c requires administrative privileges on my laptop. When I'm the admin of my computer. How do I proceed with the installation? From what I can tell this dns9.quad9.net has a problem with it.

[image]

Joulinar, 2020-10-29:
Hi,

Many thanks for your message. Pls can you try to disable IPv6 and try again.

mjhymowitz, 2020-10-29:
it worked. Thank you

closed 2020-10-29 by mjhymowitz