MichaIng/DietPi #3876: Software install - wireguard

Cepadues, 2020-10-29:
#### Details:
- Date           | Thu Oct 29 10:34:01 CET 2020
- Bug report     | 41f7614c-a17a-4ecf-aa5a-fc9f20f4d297
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | dpkg-reconfigure -f noninteractive wireguard-dkms
- Exit code      | 10
- Software title | DietPi-Software
#### Steps to reproduce:

1. run wireguard install through dietpi-software
#### Expected behaviour:

- The command shouldn't return error
#### Actual behaviour:

- The command return error
#### Extra details:

- I have pi-hole already installed. The command fail on a fresh install of diet pi aswell.
#### Additional logs:

------------------------------
Deleting module version: 1.0.20200908
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-1.0.20200908 DKMS files...
It is likely that 5.4.72-v8+ belongs to a chroot's host
Building for 5.4.72+, 5.4.72-v7+, 5.4.72-v7l+ and 5.4.72-v8+
Building initial module for 5.4.72+
Error! Bad return status for module build on kernel: 5.4.72+ (aarch64)
Consult /var/lib/dkms/wireguard/1.0.20200908/build/make.log for more information :

DKMS make.log for wireguard-1.0.20200908 for kernel 5.4.72+ (aarch64)
Thu Oct 29 10:34:24 CET 2020
make: Entering directory '/usr/src/linux-headers-5.4.72+'
  AR      /var/lib/dkms/wireguard/1.0.20200908/build/built-in.a
  CC [M]  /var/lib/dkms/wireguard/1.0.20200908/build/main.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200908/build/noise.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200908/build/device.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200908/build/peer.o
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/wireguard/1.0.20200908/build/compat/compat.h:11,
                 from <command-line>:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/wireguard/1.0.20200908/build/main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/wireguard/1.0.20200908/build/compat/compat.h:11,
                 from <command-line>:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/wireguard/1.0.20200908/build/noise.o] Error 1
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/wireguard/1.0.20200908/build/compat/compat.h:11,
                 from <command-line>:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/wireguard/1.0.20200908/build/device.o] Error 1
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/wireguard/1.0.20200908/build/compat/compat.h:11,
                 from <command-line>:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/wireguard/1.0.20200908/build/peer.o] Error 1
make: *** [Makefile:1732: /var/lib/dkms/wireguard/1.0.20200908/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.72+'

Joulinar, 2020-10-29:
Hi,

many thanks for your message. This is a know downside of the Raspberry 64bit OS Beta. Unfortunately there is nothing we can do from DietPi side because the Kernel Header Package is still missing and not yet developed by Raspberry Foundation. The entire challenge with WireGuard and Raspberry OS 64bit Beta is descriped on this Github post https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4

There are as well some workaround described. 

Keep in mind, the Raspberry 64bit OS is still in Beta status and under development by Raspberry Foundation. It's not intent for production usage.

Cepadues, 2020-10-29:
Hello,
Thank you very much for your answer, I understand the problem and I will try on the stable 32 bit version of the OS.

Joulinar, 2020-10-29:
Ok I'm going to close the topic

closed 2020-10-29 by Joulinar