MichaIng/DietPi #3877: DietPi-Update | missing directories /run/dietpi as well as /run/php after update to v6.33.3

Joulinar, 2020-10-29:
@MichaIng 
There are 2 reports on our forum, reporting about missing directories` /run/dietpi` as well as `/run/php` after updating to `v6.33.3`

This leads to PHP failing on reboot as well as some error massages on login

https://dietpi.com/phpbb/viewtopic.php?f=11&t=8203&start=10

Strange thing, /run is not totally empty. It's just missing these 2 directories.

Joulinar, 2020-11-06:
@MichaIng 
should we keep this open?

MichaIng, 2020-11-06:
Let's close for now, we have it in mind when a similar report happens, to in case further investigate.

Hsewdot, 2020-11-30:
Hello,

First of all, many thanks for all the great work you are doing on Dietpi !

After updating my dietpi to 6.33, I noticed an issue about php-fpm which prevents dietpi from running my nextcloud server. After some researches, I discover that I am in the same case as the two others users in the linked forum post. 

However, I didn't succeed to fix it.

As I see your interest in investigating more about this issue, I let you know my current situation : 

My folders `/run/dietpi` &`/run/php` are not created after boot.

In the banner, I had the following messages for instance : 

`LAN IP : mawk: cannot open /run/dietpi/.network (No such file or directory)`

&

`/boot/dietpi/func/dietpi-banner: line 223: /run/dietpi/.dietpi_motd: No such file or directory`

I have the same output as the users in the link for requested commands : 

*> root@DietPi:/run# journalctl -u systemd-tmpfiles-setup*
> -- Logs begin at Thu 2020-11-26 22:14:04 CET, end at Sun 2020-11-29 22:24:01 CET. --
> -- No entries --

*> root@DietPi:/run# systemctl restart systemd-tmpfiles-setup*
> Failed to restart systemd-tmpfiles-setup.service: Operation refused, unit systemd-tmpfiles-setup.service may be requested by dependency only (it is configured to refuse manual start/stop).
> See system logs and 'systemctl status systemd-tmpfiles-setup.service' for details.
> 

*> root@DietPi:/run# systemctl status systemd-tmpfiles-setup.service*
> ● systemd-tmpfiles-setup.service - Create Volatile Files and Directories
>    Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-setup.service; static; vendor preset: enabled)
>    Active: inactive (dead)
>      Docs: man:tmpfiles.d(5)
>            man:systemd-tmpfiles(8)

MichaIng, 2020-11-30:
Many thanks for your report. The issue appeared after first reboot after the update and another reboot doesn't fix it?

The service is pulled in by `sysinit.target`. Can you show:
```
systemctl status sysinit.target
```

Probably some APT package upgrade is responsible 🤔.

Hsewdot, 2020-12-01:
Hello,

Here is the result of the command you asked me :

```
root@DietPi:/home/dietpi# systemctl status sysinit.target
● sysinit.target - System Initialization
   Loaded: loaded (/lib/systemd/system/sysinit.target; static; vendor preset: enabled)
   Active: active since Tue 2020-12-01 23:36:45 CET; 4min 55s ago
     Docs: man:systemd.special(7)

déc. 01 23:36:45 DietPi systemd[1]: Reached target System Initialization.
```

Unfortunately, I don't have an exact memory about when it happened. So, I looked for when the issue appeared.

As it is mentioned in the first post. Both `/run/dietpi` & `/run/php` are affected.

Then, I found when php7.3-fpm logged some errors. I may have found something relevant, I paste the log below.

Indeed, there was APT package upgrade near the error, I also paste my APT history log when PHP error appeared. 

I hope these informations would be interesting and may have a relevant clue.

**PHP7.3-FPM LOG**
```

[24-Nov-2020 23:58:58] NOTICE: fpm is running, pid 24897
[24-Nov-2020 23:58:58] NOTICE: ready to handle connections
[24-Nov-2020 23:58:58] NOTICE: systemd monitor interval set to 10000ms
[25-Nov-2020 00:00:55] NOTICE: Terminating ...
[25-Nov-2020 00:00:55] NOTICE: exiting, bye-bye!
[25-Nov-2020 00:03:00] ERROR: unable to bind listening socket for address '/run/php/php7.3-fpm.sock': No such file or directory (2)
[25-Nov-2020 00:03:00] ERROR: FPM initialization failed
[25-Nov-2020 00:06:03] ERROR: unable to bind listening socket for address '/run/php/php7.3-fpm.sock': No such file or directory (2)
[25-Nov-2020 00:06:03] ERROR: FPM initialization failed
[25-Nov-2020 00:07:51] ERROR: unable to bind listening socket for address '/run/php/php7.3-fpm.sock': No such file or directory (2)
[25-Nov-2020 00:07:51] ERROR: FPM initialization failed
```
**APT HISTORY LOG**

```
Start-Date: 2020-11-24  11:45:59
Commandline: apt-get upgrade
Upgrade: libldap-2.4-2:arm64 (2.4.47+dfsg-3+deb10u3, 2.4.47+dfsg-3+deb10u4), libldap-common:arm64 (2.4.47+dfsg-3+deb10u3, 2.4.47+dfsg-3+deb10u4)
End-Date: 2020-11-24  11:46:03

Start-Date: 2020-11-24  23:52:53
Commandline: apt-get -qq --allow-change-held-packages purge python3-certbot-nginx certbot
Requested-By: dietpi (1000)
Purge: certbot:arm64 (0.31.0-1), python-certbot-nginx:arm64 (0.31.0-1), python3-certbot-nginx:arm64 (0.31.0-1)
End-Date: 2020-11-24  23:53:02

Start-Date: 2020-11-24  23:53:16
Commandline: apt-get -qq --purge autoremove
Requested-By: dietpi (1000)
Purge: python3-zope.interface:arm64 (4.3.2-1+b2), python3-distutils:arm64 (3.7.3-1), python3-configargparse:arm64 (0.13.0-1), python3-josepy:arm64 (1.1.0-2), python3-asn1crypto:arm64 (0.24.0-1), python3-urllib3:arm64 (1.24.1-1), python3-zope.hookable:arm64 (4.0.4-4+b4), python3-pbr:arm64 (4.2.0-5), python3-idna:arm64 (2.6-1), python3-zope.event:arm64 (4.2.0-1), python3-six:arm64 (1.12.0-1), python3-zope.component:arm64 (4.3.0-1), python3-mock:arm64 (2.0.0-4), python3-requests:arm64 (2.21.0-1), python3-pyparsing:arm64 (2.2.0+dfsg1-2), python3-future:arm64 (0.16.0-1), python3-cryptography:arm64 (2.6.1-3+deb10u2), python3-cffi-backend:arm64 (1.12.2-1), python3-acme:arm64 (0.31.0-2), python3-configobj:arm64 (5.0.6-3), python3-lib2to3:arm64 (3.7.3-1), python3-openssl:arm64 (19.0.0-1), python3-rfc3339:arm64 (1.1-1), python3-requests-toolbelt:arm64 (0.8.0-1), python3-certbot:arm64 (0.31.0-1), python3-tz:arm64 (2019.1-1), python3-certifi:arm64 (2018.8.24-1), python3-parsedatetime:arm64 (2.4-2), python3-chardet:arm64 (3.0.4-3), python3-setuptools:arm64 (40.8.0-1)
End-Date: 2020-11-24  23:53:31

Start-Date: 2020-11-24  23:54:18
Commandline: apt-get -qq --allow-change-held-packages purge php-apcu php-common php-igbinary php-imagick php-redis php7.3-bcmath php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-intl php7.3-json php7.3-mbstring php7.3-mysql php7.3-opcache php7.3-readline php7.3-xml php7.3-zip libapache2-mod-php7.3
Requested-By: dietpi (1000)
Purge: php7.3-xml:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-zip:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-mbstring:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-readline:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php-igbinary:arm64 (3.1.2+2.0.8-1+0~20200518.16+debian10~1.gbp1a098b), php-common:arm64 (2:79+0~20201020.27+debian10~1.gbp95ec92), php7.3:arm64(7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-json:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-mysql:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-gd:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-curl:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-common:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-intl:arm64 (7.3.24-3+0~20201100201103.72+debian10~1.gbp945915), php7.3-bcmath:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-opcache:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-cli:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php7.3-fpm:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php-redis:arm64 (5.3.1+4.3.0-1+0~20200708.25+debian10~1.gbp800f71), php-imagick:arm64 (3.4.4-4+0~20200804.15+debian10~1.gbp2925ae), libapache2-mod-php7.3:arm64 (7.3.24-3+0~20201103.72+debian10~1.gbp945915), php-apcu:arm64 (5.1.18+4.0.11-1+0~20200518.14+debian10~1.gbpa99079)
End-Date: 2020-11-24  23:54:47

Start-Date: 2020-11-24  23:55:01
Commandline: apt-get -qq --purge autoremove
Requested-By: dietpi (1000)
[... cut ...]
Error: Sub-process /usr/bin/dpkg returned an error code (1)
End-Date: 2020-11-25  00:09:59

Start-Date: 2020-11-25  00:11:27
Commandline: apt-get install python-setuptools
Requested-By: dietpi (1000)
Install: python-setuptools:arm64 (40.8.0-1)
Error: Sub-process /usr/bin/dpkg returned an error code (1)
End-Date: 2020-11-25  00:11:28

Start-Date: 2020-11-26  15:30:41
Commandline: apt-get -qq --purge autoremove
Error: Sub-process /usr/bin/dpkg returned an error code (1)
End-Date: 2020-11-26  15:30:41
```

MichaIng, 2020-12-02:
PHP-FPM fails since `/run/php` does not exist, so this is expected. You are on a Stretch system (`echo $G_DISTRO_NAME`), right? I'll do a test PHP install there, probably the tmpfiles configuration by the PHP package is malformed so that `systemd-tmpfiles-setup` fails or stops at that point. I'm just wondering why there is not single log entry, no even that the systemd unit is about to be starting 🤔.

Hsewdot, 2020-12-02:
When I installed dietpi, it was a stretch system. Then, I upgraded it to Buster system last summer. That may explain why I should have both stretch & buster. I followed the official guide of Debian to do the upgrade.

Here is the result of the commands:

```
root@DietPi:/home/dietpi# for i in /etc/apt/sources.list{,.d/*.list}; do
>   echo "File: $i"
>   cat "$i"
>   echo
>   echo
> done
File: /etc/apt/sources.list
# deb https://deb.debian.org/debian/ stretch main contrib non-free
# deb https://deb.debian.org/debian/ stretch-updates main contrib non-free
# deb https://deb.debian.org/debian-security/ stretch/updates main contrib non-free
# deb https://deb.debian.org/debian/ stretch-backports main contrib non-free
# deb http://ftp.debian.org/debian/ stretch-backports main
# deb-src deb http://ftp.debian.org/debian/ stretch-backports main
# deb https://deb.torproject.org/torproject.org stretch main
# deb-src https://deb.torproject.org/torproject.org stretch main

deb https://deb.debian.org/debian/ buster main contrib non-free
deb https://deb.debian.org/debian/ buster-updates main contrib non-free
deb https://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb https://deb.debian.org/debian/ buster-backports main contrib non-free
deb-src http://deb.debian.org/debian/ buster-backports main

File: /etc/apt/sources.list.d/dietpi-php.list
# deb https://packages.sury.org/php/ stretch main
deb https://packages.sury.org/php/ buster main

File: /etc/apt/sources.list.d/meveric-all-C2.list
deb https://dietpi.com/meveric all c2

File: /etc/apt/sources.list.d/meveric-all-main.list
deb https://dietpi.com/meveric all main

File: /etc/apt/sources.list.d/meveric-all-testing.list
deb https://dietpi.com/meveric all testing

File: /etc/apt/sources.list.d/meveric-stretch-backports.list
#deb https://dietpi.com/meveric stretch backports
deb https://dietpi.com/meveric buster backports

File: /etc/apt/sources.list.d/meveric-stretch-main.list
# deb https://dietpi.com/meveric stretch main
deb https://dietpi.com/meveric buster main

```

```
root@DietPi:/home/dietpi# cat /usr/lib/tmpfiles.d/php7.3-fpm.conf
#Type Path                  Mode UID      GID      Age Argument
    d /run/php              0755 www-data www-data -   -
root@DietPi:/home/dietpi# systemd-tmpfiles --create /usr/lib/tmpfiles.d/php7.3-fpm.conf
root@DietPi:/home/dietpi# 
```

MichaIng, 2020-12-03:
Okay, that is all fine. I also see that the only case of a listed Stretch package was for autoremoval, so an obsolete left library.

tmpfiles looks fine, command succeeded, `/run/php` should hence exist and `systemctl restart php7.3-fpm` should succeed now.

However, this extremely essential task must succeed during boot reliably.

Hsewdot, 2020-12-03:
Yes, php7.3-fpm is working now.

```

root@DietPi:/home/dietpi# systemctl restart php7.3-fpm
root@DietPi:/home/dietpi# systemctl status php7.3-fpm
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-12-03 14:22:56 CET; 7s ago
     Docs: man:php-fpm7.3(8)
  Process: 1265 ExecStartPost=/usr/lib/php/php-fpm-socket-helper install /run/php/php-fpm.sock /etc/php/7.3/fpm/pool.d/www.conf 73 (code=exited, status=0/SUCCESS)
 Main PID: 1259 (php-fpm7.3)
   Status: "Ready to handle connections"
   Memory: 8.0M
   CGroup: /system.slice/php7.3-fpm.service
           ├─1259 php-fpm: master process (/etc/php/7.3/fpm/php-fpm.conf)
           ├─1261 php-fpm: pool www
           ├─1262 php-fpm: pool www
           ├─1263 php-fpm: pool www
           └─1264 php-fpm: pool www

déc. 03 14:22:56 DietPi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
déc. 03 14:22:56 DietPi systemd[1]: Started The PHP 7.3 FastCGI Process Manager.
```

Unfortunately, after reboot the folder has disappeared again :/

```
root@DietPi:~# systemctl status php7.3-fpm
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-12-03 14:25:29 CET; 1min 7s ago
     Docs: man:php-fpm7.3(8)
  Process: 1622 ExecStart=/usr/sbin/php-fpm7.3 --nodaemonize --fpm-config /etc/php/7.3/fpm/php-fpm.conf (code=exited, status=78)
  Process: 1863 ExecStopPost=/usr/lib/php/php-fpm-socket-helper remove /run/php/php-fpm.sock /etc/php/7.3/fpm/pool.d/www.conf 73 (code=exited, status=0/SUCCESS)
 Main PID: 1622 (code=exited, status=78)

déc. 03 14:25:28 DietPi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
déc. 03 14:25:29 DietPi php-fpm7.3[1622]: [03-Dec-2020 14:25:29] ERROR: unable to bind listening socket for address '/run/php/php7.3-fpm.sock': No such file or directory (2)
déc. 03 14:25:29 DietPi php-fpm7.3[1622]: [03-Dec-2020 14:25:29] ERROR: FPM initialization failed
déc. 03 14:25:29 DietPi systemd[1]: php7.3-fpm.service: Main process exited, code=exited, status=78/CONFIG
déc. 03 14:25:29 DietPi systemd[1]: php7.3-fpm.service: Failed with result 'exit-code'.
déc. 03 14:25:29 DietPi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
```

MichaIng, 2020-12-03:
Let's see:
```
journalctl -u systemd-tmpfiles-setup
systemctl show -p WantedBy systemd-tmpfiles-setup
ls -l /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
```

This might break some running processes, but you could also try to manually call the command:
```
/bin/systemd-tmpfiles --verbose --create --remove --boot --exclude-prefix=/dev
```
I'm just not sure if it prints any error output, it has no verbose flag at least.

Hsewdot, 2020-12-03:
Here it is is : 

```
root@DietPi:/var/log# journalctl -u systemd-tmpfiles-setup
-- Logs begin at Thu 2020-12-03 22:56:31 CET, end at Thu 2020-12-03 22:57:49 CET. --
-- No entries --
```

```
root@DietPi:/var/log# systemctl show -p WantedBy systemd-tmpfiles-setup
WantedBy=
```
```

root@DietPi:/var/log# ls -l /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
ls: impossible d'accéder à '/lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service': Aucun fichier ou dossier de ce type
```

_English translation : unable to access to '/lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service':  None file or folder of this type_

Only the folder "systemd-tmp-files-setup.service" is missing, all parent folders exist.

In the next command, I deleted the "--verbose" option as it isn't recognized. I hope the verbosity is enough.
```
root@DietPi:/# /bin/systemd-tmpfiles --create --remove --boot --exclude-prefix=/dev
[/usr/lib/tmpfiles.d/fail2ban-tmpfiles.conf:1] Line references path below legacy directory /var/run/, updating /var/run/fail2ban → /run/fail2ban; please update the tmpfiles.d/ drop-in file.
```

Following these commands, I checked the `/run` folder and `/run/dietpi` exists now and it is empty.

MichaIng, 2020-12-04:
`/lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service` is missing, that is the reason. So now we need to find out why it is missing as it is a fixed part of the `systemd` package and should never ever be removed for any reason: https://packages.debian.org/buster/systemd
Check via:
```
dpkg -S /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
```
which should show
```
systemd: /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
```
means that the package `systemd` contains this file.

First of all to restore it:
´``
ln -s ../systemd-tmpfiles-setup.service /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
```

I have sadly not really an idea how to debug this 🤔, I mean how to find out what removed that file (or better symlink).

Joulinar, 2020-12-04:
and it seems to be an extrem rar case. According statistics, we have more than 5.5K systems running v6.33.3 and 3 reports on this topic only 🤔

MichaIng, 2020-12-04:
There was the originating reports in the forum, although strangely those solved themselves with another reboot 🤔. I just don't think it is a coincidence, but still not sure if/how to proceed with this. I'll go through other bug report uploads to check for other cases, at least.

Btw, 5.5k systems from the 14% which do actually send survey data. So it's more like ~39k 😉.

Joulinar, 2020-12-04:
well that's the thing with statistics.Never trust them as long as your are not the creator 🤣 

> I'll go through other bug report uploads to check for other cases, at least.

good one, let's see if there is something.

MichaIng, 2020-12-04:
Especially with those numbers one has to be careful anyway, as they accumulate while we don't know how many systems are not used anymore of have been reinstalled on the older end 😄.

I want to implement an optional monthly upload and promote `dietpi-survey` a bid by times to get more accurate and recent data, and remove everything older than a month then. Since we do not even have any access logs enabled + SFTP upload bypasses Cloudflare and the only identifying UUID is a randomly generated string not tied to any hardware info, and the upload file is usually less than a KiB (based on amount of installed software), I hope users agree that there is no privacy impact. I just know from myself that I hit "No" to any such user experience program automatically, just started recently to question this reflex and started to agree at least for some open-source projects where the exact way and data being uploaded is shown transparently (Nextcloud as a shiny example IMO).

Hsewdot, 2020-12-04:
Hello,

I did what you mention in your last post. And I get the result you expected 

```
dpkg -S /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
systemd: /lib/systemd/system/sysinit.target.wants/systemd-tmpfiles-setup.service
```
Regarding the second command it tells me it can't create the link as the filesystem is read only.

Now, I explain you the issue which happened since last time.

My pihole service had an unexpected behavior as it failed to connect to my database and it tried again every miliseconds and wrote it down my log...
```
[2020-12-03 14:49:29.584 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.584 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("UPDATE network SET numQueries = (SELECT SUM(numQueries) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;") called but database is not available!
[2020-12-03 14:49:29.585 10033M] dbquery("DELETE FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE AND id != 15;") called but database is not available!
[2020-12-03 14:49:29.586 10033M] dbquery("UPDATE network SET firstSeen = (SELECT MIN(firstSeen) FROM network WHERE hwaddr = 'xxxxxx' COLLATE NOCASE) WHERE id = 15;
```

...until the pihole-FTL.log file filled my whole sd care, ending with a 21 GB size.

As I discovered that afterwards, when I saw that my freespace (Rootfs) was only 4 kB in the banner, I did a reboot because it seemed very rare. 

Then, Dietpi-login didn't start anymore and I discovered that my SD card was on read-only mode.

To solve that issue, I mounted the SD card partition in rw mode. I runned fsck on another linux system to check if there was any error and I deleted the 21 gb log file.

_Results of fsck:_
/rootfs was ok

/boot was damaged (log below)
```

ubuntu@ubuntu:~$ sudo fsck /dev/mmcblk0p1
fsck de util-linux 2.34
fsck.fat 4.1 (2017-01-24)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1)
Cluster 2062 out of range (1). Setting to EOF.
Cluster 2318 out of range (1). Setting to EOF.
Cluster 2574 out of range (1). Setting to EOF.
Cluster 2830 out of range (1). Setting to EOF.
Cluster 3086 out of range (1). Setting to EOF.
Cluster 3342 out of range (1). Setting to EOF.
Cluster 3598 out of range (1). Setting to EOF.
Reclaimed 3157 unused clusters (6465536 bytes).
Perform changes ? (y/n) y
/dev/mmcblk0p1: 87 files, 40940/65399 clusters
```
To sum up, my current state is that dietpi-login does'nt work and my SD card seems to be locked on readonly mode

MichaIng, 2020-12-04:
That sounds very similar to an endless logging loop I faced with a newly installed phpBB extension for automated daily database backup: https://github.com/david63/autodbbackup/pull/8
A presumably changed behaviour in PHP7.4, a not well implemented while loop check without exit path in case of failure (instead of regular negative result) and our server ended up with a 100 GiB log file. Luckily the VPS somehow didn't claim that space from the disk. Not sure if there is some smart compression (unlimited repeating identical log) or long deferred allocation the reason, but it saved us two times from a server crash until I found out the exact issue with a solution for the extension 😄.

Please report your issue to Pi-hole ([forum](https://discourse.pi-hole.net/) or [GitHub](https://github.com/pi-hole/pi-hole/issues)) as there should be an exit path in case of failed database access instead of such an endless loop. _(link me @MichaIng on the forum, and/or add the **#dietpi** tag)_
__________
Okay to solve your issue, the first thing is to get rid of any file system errors. Please redo:
```
dosfsck -a /dev/mmcblk0p1
sync
```
until there is no further error found. I know that sometimes this does not work somehow, then a forced fsck on reboot might help: `echo > /forcefsck`
Check result via:
```
journalctl -t systemd-fsck
cat /run/initramfs/fsck.log
```
_(Most likely the second file does not exist on your RPi, but for completeness in case of non-RPi users read this.)_

The above method to check on boot also forces the check regardless it the file system has been marked dirty or not. When checking on an external system, assure to use `e2fsck -f /dev/...` with the `-f` flag to force a whole scan.

Hsewdot, 2020-12-04:
I created the report on Pihole github : [Issue link](https://github.com/pi-hole/pi-hole/issues/3899)

----- 

```
root@DietPi:/lib/systemd/system# dosfsck -a /dev/mmcblk0p1
fsck.fat 4.1 (2017-01-24)
/dev/mmcblk0p1: 87 files, 40940/65399 clusters
```
```
root@DietPi:/lib/systemd/system# sync
root@DietPi:/lib/systemd/system# echo > /forcefsck
-bash: /forcefsck: Système de fichiers accessible en lecture seulement
```
_Translation : Filesystem readonly_

My dietpi is running on an Odroid C2, so I have the non-Rpi file :)

```
root@DietPi:/# cat /run/initramfs/fsck.log
Log of fsck -C -y -V -t ext4 /dev/mmcblk0p2
Thu Jan  1 00:00:02 1970

fsck from util-linux 2.33.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p2] fsck.ext4 -y -C0 /dev/mmcblk0p2
e2fsck 1.44.5 (15-Dec-2018)
rootfs: clean, 215788/1941504 files, 2236377/7758720 blocks

Thu Jan  1 00:00:02 1970

```

-----

I also did `e2fsck -f `on the mmcblk0p2 and it found some orphaned inode that I repaired.

MichaIng, 2020-12-05:
> root@DietPi:/lib/systemd/system# echo > /forcefsck
-bash: /forcefsck: Système de fichiers accessible en lecture seulement

Ah yes of course this cannot work as long as the file system is read-only.

> My dietpi is running on an Odroid C2, so I have the non-Rpi file :)

Ah okay 👍, I think I mixed it up with another issue report.

> I also did `e2fsck -f` on the mmcblk0p2 and it found some orphaned inode that I repaired.

Did you repeat it until no further error was found?
_______________
After having fixed all file system errors, does the system boot again in read-write mode? Else `dmesg` and `journalctl` are the tools to check boot logs. Or:
```
systemctl status '-.mount'
```

Hsewdot, 2020-12-06:
Hello,

Yes I repeated the `e2fsck -f `and everything is fine on that point.

My system keep booting in readonly mode. 

I paste the `dmesg `log below if it can help.

```
root@DietPi:/var/log# systemctl status -- -.mount
● -.mount - /
   Loaded: loaded
   Active: active (mounted) since Thu 2019-02-14 11:11:58 CET; 9h left
    Where: /
     What: /dev/mmcblk0p2

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```

By the way, my system doesn't have the right date
The following match with 22:55  UTC + 1 ; 06 december 2020. 

```
root@DietPi:/var/log/supervisor# date
Thursday 14 February 2019, 01:56:16 (UTC+0100)
```

---------------

I tried to pick up what may be the most relevant : 
```
`[    0.000000] Kernel command line: root=/dev/mmcblk0p2 rootwait ro console=tty0 no_console_suspend hdmimode=2160p60hz m_bpp=32 vout= fsck.repair=yes net.ifnames=0 elevator=noop disablehpd=true`

[    2.823117] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
```
----------
Full dmesg

```
root@DietPi:/home/dietpi# dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.16.85+ (root@odroid-stretch64) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) ) #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020
[    0.000000] CPU: AArch64 Processor [410fd034] revision 4
[    0.000000] secmon: share mem setup
[    0.000000] Reserved memory: initialized node linux,secmon, compatible id aml_secmon_memory
[    0.000000] __reserved_mem_alloc_size: linux,meson-fb
[    0.000000] fb: Reserved memory: created fb at 0x0000000078000000, size 128 MiB
[    0.000000] Reserved memory: initialized node linux,meson-fb, compatible id amlogic, fb-memory
[    0.000000] __reserved_mem_alloc_size: linux,di
[    0.000000] DI reserved memory: created CMA memory pool at 0x0000000076200000, size 30 MiB
[    0.000000] Reserved memory: initialized node linux,di, compatible id amlogic, di-mem
[    0.000000] __reserved_mem_alloc_size: linux,ion-dev
[    0.000000] linux,ion-dev add multi user:ffffffc001d75580
[    0.000000] ion_dev mem setup
[    0.000000] Reserved memory: initialized node linux,ion-dev, compatible id amlogic, idev-mem
[    0.000000] ppmgr share mem setup
[    0.000000] linux,ion-dev add multi user:ffffffc001d7aa20
[    0.000000] Reserved memory: initialized node linux,ion-dev, compatible id amlogic, idev-mem
[    0.000000] __reserved_mem_alloc_size: linux,codec_mm_cma
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000067c00000, size 188 MiB
[    0.000000] Reserved memory: initialized node linux,codec_mm_cma, compatible id shared-dma-pool
[    0.000000] __reserved_mem_alloc_size: linux,picdec
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000064c00000, size 48 MiB
[    0.000000] Reserved memory: initialized node linux,picdec, compatible id shared-dma-pool
[    0.000000] __reserved_mem_alloc_size: linux,codec_mm_reserved
[    0.000000] codec_mm:vdec: reserved mem setup
[    0.000000] Reserved memory: initialized node linux,codec_mm_reserved, compatible id amlogic, codec-mm-reserved
[    0.000000] __reserved_mem_alloc_size: linux,jpegenc_cma
[... 809 lines cut ...]
[    5.353704] scsi 0:0:0:1: Attached scsi generic sg1 type 13
[    5.373347] ionvideo open
[    5.455921] systemd-journald[219]: Received request to flush runtime journal from PID 1
[    5.573308] ionvideo_stop_generating!!!!
[    5.578402] ionvideo release
[    5.931075] 8021q: 802.1Q VLAN Support v1.8
[    6.003025] .
[    6.021160] cfg80211: Calling CRDA to update world regulatory domain
[    6.061588] eth0: device MAC address xx:xx:xx:xx:xx:xx
[    7.013042] .....
[   11.063416] stmmaceth c9410000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.063067] ........ready
[   19.133940] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[   19.134565] sd 0:0:0:0: [sda] 15628052480 512-byte logical blocks: (8.00 TB/7.27 TiB)
[   19.134711] sd 0:0:0:0: [sda] 4096-byte physical blocks
[   19.135425] sd 0:0:0:0: [sda] Write Protect is off
[   19.135525] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[   19.136181] sd 0:0:0:0: [sda] No Caching mode page found
[   19.136294] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   19.137425] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[   20.512954]  sda: sda1
[   20.514404] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[   20.516155] sd 0:0:0:0: [sda] Attached SCSI disk

```

MichaIng, 2020-12-06:
Very strange:
```
[    2.823117] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
```
This is not normal, as ext4 is a journaling file system. Like: https://superuser.com/q/855547/1183051

```
[    4.704222] systemd[1]: Started Remount Root and Kernel File Systems.
```
That should be the point where the root file system is mounted R/W as defined in /etc/fstab:
```
systemctl cat -- -.mount
```
This should show `Options=noatime,lazytime,rw`.

And does it work to manually remount the rootfs with proper options? `mount -o remount,rw /`

Another topic: The haveged daemon fails to start, not sure if it's related:
```
journalctl -u haveged
```

Hsewdot, 2020-12-07:
I have read your link, then if I understand well, I have to format the partition correctly. 🤔 

I saw that my /etc/fstab file is empty. Then I filled it up following debian guide

I tried to modify it with the options you told me but I didn't solve the problem.
```

root@DietPi:/home/dietpi# cat /etc/fstab
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
UUID=e139ce78-9841-40fe-8823-96a304a09859rootfs / ext4 noatime,lazytime,rw 0 1  
```

Yes using `mount -o remount,rw /` works but it doesn't persist after reboot.

-------
Unfortunately, there is no log entry for havegd (the readonly mode may prevent writing ones)

```
root@DietPi:/home/dietpi# journalctl -u haveged
-- Logs begin at Thu 2019-02-14 13:14:35 CET, end at Thu 2019-02-14 13:15:38 CET. --
-- No entries --
```
-----
<details>

<summary> dmesg with new fstab file </summary>

```
dietpi@DietPi:~$ dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.16.85+ (root@odroid-stretch64) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) ) #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020
[    0.000000] CPU: AArch64 Processor [410fd034] revision 4
[    0.000000] secmon: share mem setup
[    0.000000] Reserved memory: initialized node linux,secmon, compatible id aml_secmon_memory
[    0.000000] __reserved_mem_alloc_size: linux,meson-fb
[    0.000000] fb: Reserved memory: created fb at 0x0000000078000000, size 128 MiB
[    0.000000] Reserved memory: initialized node linux,meson-fb, compatible id amlogic, fb-memory
[    0.000000] __reserved_mem_alloc_size: linux,di
[    0.000000] DI reserved memory: created CMA memory pool at 0x0000000076200000, size 30 MiB
[    0.000000] Reserved memory: initialized node linux,di, compatible id amlogic, di-mem
[    0.000000] __reserved_mem_alloc_size: linux,ion-dev
[    0.000000] linux,ion-dev add multi user:ffffffc001d75580
[    0.000000] ion_dev mem setup
[    0.000000] Reserved memory: initialized node linux,ion-dev, compatible id amlogic, idev-mem
[    0.000000] ppmgr share mem setup
[    0.000000] linux,ion-dev add multi user:ffffffc001d7aa20
[    0.000000] Reserved memory: initialized node linux,ion-dev, compatible id amlogic, idev-mem
[    0.000000] __reserved_mem_alloc_size: linux,codec_mm_cma
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000067c00000, size 188 MiB
[    0.000000] Reserved memory: initialized node linux,codec_mm_cma, compatible id shared-dma-pool
[    0.000000] __reserved_mem_alloc_size: linux,picdec
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000064c00000, size 48 MiB
[    0.000000] Reserved memory: initialized node linux,picdec, compatible id shared-dma-pool
[    0.000000] __reserved_mem_alloc_size: linux,codec_mm_reserved
[    0.000000] codec_mm:vdec: reserved mem setup
[    0.000000] Reserved memory: initialized node linux,codec_mm_reserved, compatible id amlogic, codec-mm-reserved
[    0.000000] __reserved_mem_alloc_size: linux,jpegenc_cma
[    0.000000] Reserved memory: created CMA memory pool at 0x000000005e400000, size 36 MiB
[    0.000000] Reserved memory: initialized node linux,jpegenc_cma, compatible id shared-dma-pool
[    0.000000] cma: CMA: reserved 8 MiB at 5dc00000
[    0.000000] On node 0 totalpages: 486912
[    0.000000]   Normal zone: 6664 pages used for memmap
[... 808 lines cut ...]
[    4.886079] scsi 0:0:0:1: Attached scsi generic sg1 type 13
[    4.887359] ionvideo open
[    4.993033] .
[    5.079608] systemd-journald[226]: Received request to flush runtime journal from PID 1
[    5.089941] ionvideo_stop_generating!!!!
[    5.095372] ionvideo release
[    5.453680] 8021q: 802.1Q VLAN Support v1.8
[    5.573196] cfg80211: Calling CRDA to update world regulatory domain
[    5.667895] eth0: device MAC address xx:xx:xx:xx:xx:xx
[    6.003008] .....
[   10.663311] stmmaceth c9410000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   11.053047] ......ready
[   16.103853] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[   16.104337] sd 0:0:0:0: [sda] 15628052480 512-byte logical blocks: (8.00 TB/7.27 TiB)
[   16.104473] sd 0:0:0:0: [sda] 4096-byte physical blocks
[   16.105211] sd 0:0:0:0: [sda] Write Protect is off
[   16.105304] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[   16.105988] sd 0:0:0:0: [sda] No Caching mode page found
[   16.106099] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   16.107237] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[   17.271994]  sda: sda1
[   17.273565] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
```

</details>

MichaIng, 2020-12-07:
> I have read your link, then if I understand well, I have to format the partition correctly. 🤔

That would be a radical solution (basically flash DietPi freshly). Not sure if it is possible to recover the journaling function.

> I saw that my /etc/fstab file is empty. Then I filled it up following debian guide

Okay that is a good reason. Please reset your fstab via: `dietpi-drive_manager 4` (after mounting rootfs R/W via `mount -o remount,rw /`)

> Unfortunately, there is no log entry for havegd (the readonly mode may prevent writing ones)

Let's first recover the correct fstab with above step and see if this recovers R/W after reboot.

Then we need to fix your system time: `/boot/dietpi/func/run_ntpd 1`
Currently not sure if it did not run due to R/O mount, or did you manually disable it?

And then we re-check haveged and see if the C2 probably has a hardware random generator that can be used more effectively (if you are in mood). Not related to your issue, but could be a general enhancements for our C2 systems.

Hsewdot, 2020-12-07:
Ok, I deleted what was written in fstab 

However, my system doesn't find the command 
```

root@DietPi:/home/dietpi# dietpi-drive_manager 4

bash: dietpi-drive_manager : commande introuvable
```

For system time, I think it is due to R/O mount because I don't know how to disable it and I don't get the interest.

I would like to add that my folders `/boot` and '/DietPi' are empty.

MichaIng, 2020-12-07:
Ah, please do:
```
mount -o remount,rw / # if not done yet
mount /dev/mmcblk0p1 /boot
. /boot/dietpi/func/dietpi-globals
dietpi-drive_manager 4
```

Hsewdot, 2020-12-07:
Thank you, now I can do all the commands :

```
root@DietPi:/etc# dietpi-drive_manager 4
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager |  - Detected mounted drive: /dev/mmcblk0p1 > /boot
[ INFO ] DietPi-Drive_Manager |  - Detected unmounted drive: /dev/sda1
```

<details>

<summary> fstab file </summary>

```
# Please use "dietpi-drive_manager" to setup mounts
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAPFILE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / ext4 noatime,lazytime,rw 0 1
UUID=0966-01D4 /boot vfat noatime,lazytime,rw 0 2
#UUID=8e408268-1231-4e5b-9ead-06979431bc60 /mnt/8e408268-1231-4e5b-9ead-06979431bc60 ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount
```

</details>

-------

About system time
_First_ 
```
root@DietPi:/etc# /boot/dietpi/func/run_ntpd 1
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
```
_Second_ 
```

[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/60)
```

I think It's normal knowing my system only achieve to communicate with local network.

I have a bug as my dietpi config doesn't detect any ethernet software despite having an ethernet cable plugged in.
```

  ●─ Adapter Options ─────────────●                                      │                            
                            │                                      Ethernet : Not Found | [On] | Disconnected                                      │                            
                            │                                      WiFi     : Not Found | [Off] | Disconnected 

```
&

`No Ethernet hardware was detected on your device.  `

-----

And now haveged seems to work !

```
root@DietPi:/etc# systemctl start haveged
root@DietPi:/etc# journalctl -u haveged
-- Logs begin at Thu 2020-12-03 14:26:51 CET, end at Thu 2020-12-03 14:28:09 CET. --
déc. 03 14:28:08 DietPi systemd[1]: Started Entropy Daemon based on the HAVEGE algorithm.
déc. 03 14:28:08 DietPi haveged[1327]: haveged: listening socket at 3
déc. 03 14:28:09 DietPi haveged[1327]: haveged: ver: 1.9.8; arch: generic; vend: ; build: (gcc 9.2.1 CTV); collect: 128K
déc. 03 14:28:09 DietPi haveged[1327]: haveged: cpu: (VC); data: 16K (D); inst: 16K (D); idx: 11/40; sz: 15456/64452
déc. 03 14:28:09 DietPi haveged[1327]: haveged: tot tests(BA8): A:1/1 B:1/1 continuous tests(B):  last entropy estimate 8.00294
déc. 03 14:28:09 DietPi haveged[1327]: haveged: fills: 0, generated: 0
```

MichaIng, 2020-12-07:
Okay, fstab is fine now, great that also haveged works now.

> No Ethernet hardware was detected on your device.

This means the Ethernet adapter is indeed not recognised by the kernel, otherwise it would be listed in `ls -l /sys/class/net/` as `eth0` or something. This is also new, isn't it? I mean you did the DietPi update and APT upgrades before.
Hmm in dmesg it states `[    5.667895] eth0: device MAC address xx:xx:xx:xx:xx:xx`. I hope that not the network interfaces are listed elsewhere with this older kernel version: `find /sys -name eth0`

Hsewdot, 2020-12-07:
Yes I think it is linked because it's what leads me to post here.

because from my understanding of Dietpi-config code, it gets the info here `ETH_DEV_INDEX=$(mawk 'NR==1' /run/dietpi/.network)` but as my folder `/run/dietpi` doesn't exist, it can't get the info.  

I did the command `/bin/systemd-tmpfiles --verbose --create --remove --boot --exclude-prefix=/dev` to recreate `/run/dietpi` and now my ethernet device is detected.

----

Though, the real issue is that when I put my ethernet card in DHCP mode as I have DHCP server running on my router. The gateway adress is wrong.

When I changed of ISP a few month ago my router ip adress changed from 192.168.0.1 to 192.168.1.1. However, in my system my old gateway adress is kept.

In addition, the dns is 127.0.0.1 but as my pihole is broken too, it can't resolve anything.

---------

To sum up:

 the system doesn't boot anymore in R/O mode (thanks :) ).

Pihole has started acting crazy again but I stopped the service to prevent from doing the same thing.

MichaIng, 2020-12-07:
Hmm, the Ethernet hardware detection is done by traversing `/sys/class/net/eth*`. Only the active/chosen Ethernet interface and IP is stored to `/run/dietpi/.network`.

> When I changed of ISP a few month ago my router ip address changed from 192.168.0.1 to 192.168.1.1. However, in my system my old gateway address is kept.

Do you use the router as DHCP server or Pi-hole? Since the router IP changed to a different subnet, all devices, including Pi-hole need to change the IP range as well. And if Pi-hole is acting as DHCP server, then it makes sense that it uses the old stored IP range as defined in it's config files.
But, I am pretty sure you can simply change the IP range in your router settings? This would be simpler than re-configuring Pi-hole and probably other devices which used fixed IP addresses somewhere. Else: `pihole -r` > `reconfigure`, but assure that you do **NOT** chose to install Lighttpd as webserver (if you installed it via `dietpi-software`). This would break our webserver setup, as Pi-hole implements it a different way. Choosing "No" there will leave the existing webserver (including Pi-hole web interface) setup in place and functional.

> In addition, the DNS is 127.0.0.1

If you use that DietPi machine as server only, not as desktop client with browser, then change the DNS server to something upstream, like Cloudflare at 1.1.1.1. Ah I see, with DHCP this is not always possible. This is a good reason to change to static IP, which should work well in most cases. E.g. make a DHCP IP reservation for the Pi-hole machine in the DHCP server, so that it does not assign that IP address to any other device. Then use that IP as static IP on the Pi-hole machine. Then you can set the DNS server as well to an upstream one.

In the past the Pi-hole installer configured the own host machine to use itself for DNS resolving (127.0.0.1), but that is actually no good idea (maintenance or crashes of Pi-hole will lock itself out, as you perfectly recognised) and with recent version as well the installer leaves the hosts DNS as it was.

Hsewdot, 2020-12-07:
I got some debug and technical help from my IRL friend @TtuxX (he was following this strange issue) :)

--------

In fact, changing the IP range in my router seemed harder than expected.

I changed my IP address on my router after router reboot & plug/unplug the Dietpi. 
However after this, the Dietpi kept the 192.168.1.17 address, whatever the range was (in .0.x or in .1.x) on my router control panel.

After some manipulations, I noticed that with a router set with a "1.x" range, using 192.168.0.1, my internet connection started working again on my Dietpi.

Starting from there where I had a working but weird internet connection, I wanted to change my Dietpi network settings manually, so as to make them match "by force" my router configuration. I did it from dietpi-config with a static lease in dietpi-config with cloudflare dns.

At the same time I reconfigured pi-hole with `pihole -r` as you indicated (but don't ask me why haha).

However, doing that I discovered that its IP was still not right and still ".1.x" so i investigated further and typed `ip route show`. I got:

[image]

That was completely incoherent.

So I ended-up looking for that IP in the system as I had a conviction that it was static somewhere in the config files. I did a `grep -lri '192.168.0.1' /etc/` to find the files where this offending IP would still be located, and I found "`/etc/dhcpcd.conf`".

I found to my great surprise that there were duplicated and conflicting entries in the file, with conflict from Dietpi and Pi-hole.

[image]

I edited the dhcpcd.conf file to remove duplicate entries and Dietpi-config network entry worked again as expected and all my network issues were solved.

I know that my Pi-hole is too old and I must update it in case they fixed conflicting duplicate configurations in the dhcpcd.conf file.

However, do you know if it is possible that Dietpi and possibly Pi-hole do not catch conflicting duplicate configurations for the same interfaces in the dhcpcd.conf file ? Should I report it to Pi-hole (if this kind of check is still missing in the latest Pi-hole versions) ?

---- 

So finally for me (apart from the Pi-hole spamming log write), the original issue was solved thanks to your instructions:
https://github.com/MichaIng/DietPi/issues/3877#issuecomment-738240814 then https://github.com/MichaIng/DietPi/issues/3877#issuecomment-738463143

Thank you again for your kind help @MichaIng ! <3

MichaIng, 2020-12-08:
Wow wait, dhcpcd did actually configure your network adapter? It should actually not do that for any network interfaces that is listed in `/etc/network/interfaces` without the "`manual`" suffix, as then tools from the `ifupdown` package, which pulls in the isc-dhcp-server (dhclient) in case, do the configuration.

`dhcpcd` is installed by Pi-hole, but should remain inactive with our configuration. Especially do we disable the service after Pi-hole install, but possibly it reactivates itself on package upgrades. Luckily Pi-hole is about to remove dhcpcd from dependencies: https://github.com/pi-hole/pi-hole/pull/3715
For you this means, purge dhcpcd: `apt purge dhcpcd5`

Hsewdot, 2020-12-09:
That's quite strange indeed because I don't have any memorie of adding or deleting manual suffix.

Here is my interfaces file : 

root@DietPi:/boot# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.1.17
netmask 255.255.255.0
gateway 192.168.1.1
#dns-nameservers 192.168.1.1

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 0.0.0.0
netmask 0.0.0.0
gateway 0.0.0.0
#dns-nameservers 0.0.0.0
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

Good to know that the issue won't appear in the future. :)

-------

As an epilogue, I think I would have to reformat my system as both packages `python-pkg-resources` & `python-setuptools` are in "a very bad inconsistent state" following that reinstall [Comment](https://github.com/MichaIng/DietPi/issues/3877#issuecomment-736874163) in the apt history log. Though, it may be my fault as I possibly ordered dietpi-software to uninstall the wrong software.
After some research on google, I tried a lot of commands with apt/dpkg/aptitude, unfortunately they didn't solve this issue. As you have a lot of knowledge, have you ever encountered such an issue? I would be happy to know your ideas if it doesn't take you much time. 

Though, I am aware here may not be the right place. So, if you would like to investigate it further, I may create a topic on dietpi forum or we can talk about in dm.

MichaIng, 2020-12-09:
> That's quite strange indeed because I don't have any "manual" suffix written in my file.

What does the following show?
```
ip l
ip a
journalctl -u dhcpcd
```

> Though, it may be my fault as I possibly ordered dietpi-software to uninstall the wrong software.

Whatever you uninstall/purge, it should never lead to a broken package state or failing APT. I usually bet for some file corruption in such case, or one that was fixed, but the fix included the removal of a file (unlink of an inode).

> Error: Sub-process /usr/bin/dpkg returned an error code (1)

I guess the postinst script cannot finish due to a broken or missing file. That could be solvable by removing the script and then reinstalling the package(s):
```
rm -f /var/lib/dpkg/info/{python-pkg-resources,python-setuptools}.postinst
apt -f install
apt install --reinstall python-pkg-resources python-setuptools
```
If it fails the same way, then one would need to have a deeper look into those `postinst` scripts to see where it is failing and why.

> Though, I am aware here may not be the right place.

We don't separate it strict. I personally prefer GitHub, but doesn't matter much. For documentation and to possibly reference issues/solutions at a later time, it is great to do it in a public thread, as long as investigation does not imply exposing personal data somehow.

So it is a bid up to you: We could try to track down the issues, which will likely cost some time and there is no guarantee that there will be no other related issue later (other missing or broken files due to the Pi-hole incidence) that we don't see yet. If there is not much custom configs or data to transfer, a fresh reinstall of DietPi could be quicker and sort out potential future issues, but less learning about the system depths 😃.

Hsewdot, 2020-12-09:
I think I want to carry on, at least I will understand better how it works :)

------
Here are the commands regarding the network device

```
root@DietPi:/boot# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
root@DietPi:/boot# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.17/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 71738sec preferred_lft 71738sec
```

------
Regarding the python issue

```
root@DietPi:~# apt -f install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
Need to get 0 B/182 kB of archives.
After this operation, 0 B of additional disk space will be used.
dpkg: error processing package python-pkg-resources (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
dpkg: dependency problems prevent configuration of python-setuptools:
 python-setuptools depends on python-pkg-resources (= 40.8.0-1); however:
  Package python-pkg-resources is not configured yet.

dpkg: error processing package python-setuptools (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 python-pkg-resources
 python-setuptools
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# apt install --reinstall python-pkg-resources python-setuptools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 2 reinstalled, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
Need to get 0 B/182 kB of archives.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for python-setuptools:arm64
```

MichaIng, 2020-12-09:
Please try the following:
```
dpkg -P --force-depends python-pkg-resources python-setuptools
apt install python-pkg-resources python-setuptools
```

Hsewdot, 2020-12-09:
```
root@DietPi:/home/dietpi# dpkg -P --force-depends python-pkg-resources python-setuptools
(Reading database ... 34618 files and directories currently installed.)
Removing python-setuptools (40.8.0-1) ...
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Traceback (most recent call last):
  File "/usr/lib/python2.7/site.py", line 554, in <module>
    main()
  File "/usr/lib/python2.7/site.py", line 536, in main
    known_paths = addusersitepackages(known_paths)
  File "/usr/lib/python2.7/site.py", line 272, in addusersitepackages
    user_site = getusersitepackages()
  File "/usr/lib/python2.7/site.py", line 247, in getusersitepackages
    user_base = getuserbase() # this will also set USER_BASE
  File "/usr/lib/python2.7/site.py", line 237, in getuserbase
    USER_BASE = get_config_var('userbase')
  File "/usr/lib/python2.7/sysconfig.py", line 587, in get_config_var
    return get_config_vars().get(name)
  File "/usr/lib/python2.7/sysconfig.py", line 533, in get_config_vars
    _init_posix(_CONFIG_VARS)
  File "/usr/lib/python2.7/sysconfig.py", line 417, in _init_posix
    from _sysconfigdata import build_time_vars
  File "/usr/lib/python2.7/_sysconfigdata.py", line 6, in <module>
    from _sysconfigdata_nd import *
ImportError: No module named _sysconfigdata_nd
dpkg: error processing package python-setuptools (--purge):
 installed python-setuptools package pre-removal script subprocess returned error exit status 1
dpkg: python-pkg-resources: dependency problems, but removing anyway as you requested:
 supervisor depends on python-pkg-resources.
 speedtest-cli depends on python-pkg-resources.
 python-setuptools depends on python-pkg-resources (= 40.8.0-1).

dpkg: error processing package python-pkg-resources (--purge):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting a removal
Errors were encountered while processing:
 python-setuptools
 python-pkg-resources
```

Then the apt install didn't work
<details>
<summary> apt install log </summary>

```
root@DietPi:/home/dietpi# apt install python-pkg-resources python-setuptools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-pkg-resources is already the newest version (40.8.0-1).
python-setuptools is already the newest version (40.8.0-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
Need to get 0 B/182 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
dpkg: error processing package python-pkg-resources (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
dpkg: dependency problems prevent configuration of python-setuptools:
 python-setuptools depends on python-pkg-resources (= 40.8.0-1); however:
  Package python-pkg-resources is not configured yet.

dpkg: error processing package python-setuptools (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 python-pkg-resources
 python-setuptools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

</details>

MichaIng, 2020-12-09:
As long as the purge fails, the reinstall of course fails as well. Hammer method, let's see:
```
rm -f /var/lib/dpkg/info/{python-pkg-resources,python-setuptools}*
dpkg -P --force-depends python-pkg-resources python-setuptools
# only if the purge succeeds go on with:
apt install python-pkg-resources python-setuptools
```

Hsewdot, 2020-12-09:
I did the first 3 commands with that 

The first command printed no specific message

Then the second

```
root@DietPi:/home/dietpi# dpkg -P --force-depends python-pkg-resources python-setuptools
dpkg: warning: files list file for package 'python-pkg-resources' missing; assuming package has no files currently installed
(Reading database ... 34490 files and directories currently installed.)
Removing python-setuptools (40.8.0-1) ...
dpkg: python-pkg-resources: dependency problems, but removing anyway as you requested:
 supervisor depends on python-pkg-resources.
 speedtest-cli depends on python-pkg-resources.

dpkg: error processing package python-pkg-resources (--purge):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting a removal
Errors were encountered while processing:
 python-pkg-resources
```
Even though it failed, I did the third command :

```
root@DietPi:/home/dietpi# apt install --reinstall python2.7-minimal
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
Need to get 1,292 kB/1,474 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 python2.7-minimal arm64 2.7.16-2+deb10u1 [1,292 kB]
Fetched 1,292 kB in 0s (4,817 kB/s)      
dpkg: warning: files list file for package 'python-pkg-resources' missing; assuming package has no files currently installed
(Reading database ... 34490 files and directories currently installed.)
Preparing to unpack .../python2.7-minimal_2.7.16-2+deb10u1_arm64.deb ...
Unpacking python2.7-minimal (2.7.16-2+deb10u1) over (2.7.16-2+deb10u1) ...
dpkg: error processing package python-pkg-resources (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Setting up python2.7-minimal (2.7.16-2+deb10u1) ...
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Traceback (most recent call last):
  File "/usr/lib/python2.7/py_compile.py", line 181, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/py_compile.py", line 173, in main
    compile(filename, doraise=True)
  File "/usr/lib/python2.7/py_compile.py", line 106, in compile
    with open(file, 'U') as f:
IOError: [Errno 2] No such file or directory: '/usr/lib/python2.7/encodings/__init__.py'
dpkg: error processing package python2.7-minimal (--configure):
 installed python2.7-minimal package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 python-pkg-resources
 python2.7-minimal
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

So it achieves to remove the `python-setuptools` but not the `python-pkg-resources`.

I didn't do the fourth command as you told me

MichaIng, 2020-12-09:
Try:
```
apt install --reinstall libpython2.7-minimal
```

Hsewdot, 2020-12-09:
python2.7-minimal doesn't return error anymore.

Now, only `python-pkg-resources` is problematic.

```
root@DietPi:/home/dietpi# apt install --reinstall libpython2.7-minimal
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
Need to get 395 kB/577 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 libpython2.7-minimal arm64 2.7.16-2+deb10u1 [395 kB]
Fetched 395 kB in 0s (1,767 kB/s)         
dpkg: warning: files list file for package 'python-pkg-resources' missing; assuming package has no files currently installed
(Reading database ... 34490 files and directories currently installed.)
Preparing to unpack .../libpython2.7-minimal_2.7.16-2+deb10u1_arm64.deb ...
Unpacking libpython2.7-minimal:arm64 (2.7.16-2+deb10u1) over (2.7.16-2+deb10u1) ...
dpkg: error processing package python-pkg-resources (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Setting up libpython2.7-minimal:arm64 (2.7.16-2+deb10u1) ...
Setting up python2.7-minimal (2.7.16-2+deb10u1) ...
Errors were encountered while processing:
 python-pkg-resources
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-12-09:
Okay probably now the other two packages can be reinstalled:
```
python-pkg-resources python-setuptools
```

It seems to have been some of the core libraries for Python missing that you reinstalled now. \*pressing thumbs\*

Hsewdot, 2020-12-09:
It did fail, but it gives new error message and may be a new clue. From what I understand the cause is a missing file `debpython.version`. I checked in /usr/bin, there is no file named like that.

```
root@DietPi:/home/dietpi# apt install --reinstall python-pkg-resources python-setuptools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  python-setuptools-doc
The following NEW packages will be installed:
  python-setuptools
0 upgraded, 1 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
Need to get 382 kB/564 kB of archives.
After this operation, 1,442 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 python-setuptools all 40.8.0-1 [382 kB]
Fetched 382 kB in 0s (1,950 kB/s)      
dpkg: warning: files list file for package 'python-pkg-resources' missing; assuming package has no files currently installed
(Reading database ... 34490 files and directories currently installed.)
Preparing to unpack .../python-pkg-resources_40.8.0-1_all.deb ...
Unpacking python-pkg-resources (40.8.0-1) over (40.8.0-1) ...
Selecting previously unselected package python-setuptools.
Preparing to unpack .../python-setuptools_40.8.0-1_all.deb ...
Unpacking python-setuptools (40.8.0-1) ...
Setting up python-pkg-resources (40.8.0-1) ...
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 35, in <module>
    from debpython.version import SUPPORTED, debsorted, vrepr, \
ImportError: No module named debpython.version
dpkg: error processing package python-pkg-resources (--configure):
 installed python-pkg-resources package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of python-setuptools:
 python-setuptools depends on python-pkg-resources (= 40.8.0-1); however:
  Package python-pkg-resources is not configured yet.

dpkg: error processing package python-setuptools (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 python-pkg-resources
 python-setuptools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-12-09:
Let's reinstall the other larger library
```
apt install --reinstall libpython2.7-stdlib
# and to be sure
apt install --reinstall python2.7
```

Hsewdot, 2020-12-09:
Both commands printed the same thing : (see below

I don't know if you have the following steps in mind yet. 

Else I have found that [case](https://askubuntu.com/questions/881811/apt-get-issues-python-related-packages-not-fully-installed-or-removed/881815) which seems similar. Do you think it's relevant in my case ? (mainly reinstalling libpython2.7-minimal)

```
root@DietPi:/usr/bin# apt install --reinstall python2.7
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
2 not fully installed or removed.
Need to get 305 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 python2.7 arm64 2.7.16-2+deb10u1 [305 kB]
Fetched 305 kB in 0s (1,850 kB/s)
(Reading database ... 34618 files and directories currently installed.)
Preparing to unpack .../python2.7_2.7.16-2+deb10u1_arm64.deb ...
Unpacking python2.7 (2.7.16-2+deb10u1) over (2.7.16-2+deb10u1) ...
Setting up python2.7 (2.7.16-2+deb10u1) ...
Setting up python-pkg-resources (40.8.0-1) ...
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 35, in <module>
    from debpython.version import SUPPORTED, debsorted, vrepr, \
ImportError: No module named debpython.version
dpkg: error processing package python-pkg-resources (--configure):
 installed python-pkg-resources package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of python-setuptools:
 python-setuptools depends on python-pkg-resources (= 40.8.0-1); however:
  Package python-pkg-resources is not configured yet.

dpkg: error processing package python-setuptools (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for mime-support (3.62) ...
Errors were encountered while processing:
 python-pkg-resources
 python-setuptools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-12-09:
> mainly reinstalling libpython2.7-minimal

You did that above already 😄: https://github.com/MichaIng/DietPi/issues/3877#issuecomment-742096757

Yeah, since we already reinstalled all basic Python environment, I'm also out of idea how to do it a soft way. Not sure where "debpython" is coming from but the `version.py` scripts is part of the above library.

The other way is to purge Python 2 completely. Can you see what would happen, if this would purge as well some software you need to keep (especially configuration and data)?
```
apt purge libpython2.7-minimal # Do not hit "y" if any end user software would be purged as well ;)
```
Purging this library will purge all other Python packages as well as everything depends on it, but hopefully no software you explicitly installed and need.

And not to forget local/pip installs: `find /usr/local -name python2.7`

Hsewdot, 2020-12-09:
Oh, I only checked the previous comment, I should have gone more above 

Regarding the purge, I identify 2 softwares samba & supervisor.

Reinstalling samba isn't an issue at the moment as I have a pretty basic configuration. However, I am more skeptical about supervisor. 

In all case, I have already a backup of my system just before we started to manipulate python package. :)

MichaIng, 2020-12-10:
Ah Samba is based on Python 2, that is true. However its config can be backed up easily: `mv /etc/samba/smb.conf{,_bak}`
Do you use supervisor for MineOS or something else? Because with DietPi v6.34 we switch to a systemd service for controlling MineOS.

Hsewdot, 2020-12-10:
I had installed MineOS a long time ago to see the performance on my C2 😄 

But now I don't need it. I was afraid that supervisor process break the system. 

My /etc/samba has disappeared very strange 🤔 

Hopefully, I find it in my dietpi-backup/data/etc folder.

Then I did the purge but there were errors. I did a second and I posted it logs only the package left which are on errors.

<details>
<summary> apt -log purge </summary>

```

root@DietPi:/mnt/dietpi-backup/data/etc/samba# apt purge libpython2.7-minimal
Reading package lists... Done
Building dependency tree      
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libarchive13 libjansson4 libldb1 librsync1 libtalloc2 libtdb1 libtevent0 libwbclient0 samba-common tdb-tools
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  libpython-stdlib* libpython2-stdlib* libpython2.7* libpython2.7-minimal* libpython2.7-stdlib* libsmbclient* python* python-crypto* python-dnspython* python-ldb* python-meld3* python-minimal* python-pkg-resources*
  python-samba* python-setuptools* python-talloc* python-tdb* python2* python2-minimal* python2.7* python2.7-minimal* rdiff-backup* samba* samba-common-bin* samba-libs* smbclient* speedtest-cli* supervisor*
0 upgraded, 0 newly installed, 28 to remove and 1 not upgraded.
2 not fully installed or removed.
After this operation, 86.8 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 34618 files and directories currently installed.)
Removing python-setuptools (40.8.0-1) ...
Removing supervisor (3.3.5-1) ...
Traceback (most recent call last):
  File "/usr/bin/pyclean", line 31, in <module>
    from debpython import files as dpf
ImportError: No module named debpython
dpkg: error processing package supervisor (--remove):
 installed supervisor package pre-removal script subprocess returned error exit status 1
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 35, in <module>
    from debpython.version import SUPPORTED, debsorted, vrepr, \
ImportError: No module named debpython.version
dpkg: error while cleaning up:
 installed supervisor package post-installation script subprocess returned error exit status 1
Removing samba (2:4.9.5+dfsg-5+deb10u1) ...
Removing samba-common-bin (2:4.9.5+dfsg-5+deb10u1) ...
Removing python-samba (2:4.9.5+dfsg-5+deb10u1) ...
Traceback (most recent call last):
  File "/usr/bin/pyclean", line 31, in <module>
    from debpython import files as dpf
ImportError: No module named debpython
dpkg: error processing package python-samba (--remove):
 installed python-samba package pre-removal script subprocess returned error exit status 1
Removing smbclient (2:4.9.5+dfsg-5+deb10u1) ...
Removing libsmbclient:arm64 (2:4.9.5+dfsg-5+deb10u1) ...
Removing python-dnspython (1.16.0-1) ...
Traceback (most recent call last):
  File "/usr/bin/pyclean", line 31, in <module>
    from debpython import files as dpf
ImportError: No module named debpython
dpkg: error processing package python-dnspython (--remove):
 installed python-dnspython package pre-removal script subprocess returned error exit status 1
Removing rdiff-backup (1.2.8-7) ...
Traceback (most recent call last):
  File "/usr/bin/pyclean", line 31, in <module>
    from debpython import files as dpf
ImportError: No module named debpython
dpkg: error processing package rdiff-backup (--remove):
 installed rdiff-backup package pre-removal script subprocess returned error exit status 1
Removing speedtest-cli (2.0.2-1+deb10u1) ...
[... 474 lines cut ...]
 python-meld3 : Depends: python:any (< 2.8)
                Depends: python:any (>= 2.7.5-5~)
 python-pkg-resources : Depends: python:any (< 2.8)
                        Depends: python:any (>= 2.7~)
 python-samba : Depends: samba-libs (= 2:4.9.5+dfsg-5+deb10u1) but it is not going to be installed
                Depends: python (< 2.8) but it is not going to be installed
                Depends: python (>= 2.7~) but it is not going to be installed
                Depends: python:any (< 2.8)
                Depends: python:any (>= 2.7~)
                Depends: libpython2.7 (>= 2.7) but it is not going to be installed
                Depends: python-talloc (>= 2.1.9) but it is not going to be installed
 python-tdb : Depends: python (< 2.8) but it is not going to be installed
              Depends: python (>= 2.7~) but it is not going to be installed
              Depends: python:any (< 2.8)
              Depends: python:any (>= 2.7~)
              Depends: libpython2.7 (>= 2.7) but it is not going to be installed
 python2-minimal : Depends: python2.7-minimal (>= 2.7.16-1~) but it is not going to be installed
 rdiff-backup : Depends: python (>= 2.7) but it is not going to be installed
                Depends: python (< 2.8) but it is not going to be installed
 supervisor : Depends: python:any (< 2.8)
              Depends: python:any (>= 2.7~)
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

</details>

MichaIng, 2020-12-10:
> My /etc/samba has disappeared very strange 🤔

Ah sorry, I think it's `/etc/smb.conf`, hopefully it's still there.

Now I see where this `debpython` is part of: https://packages.debian.org/buster/amd64/python2-minimal/filelist
I thought `python2-minimal` is just a meta package to pull the right subversion package (`python2.7-minimal`) but is contains scripts as well. It seems that `/usr/share/python` got lost which caused all those `debpython` errors, at least as part of the issue.

Okay, round-robin:
```
apt install --reinstall python2-minimal
apt purge python-crypto  python-dnspython python-ldb python-meld3 python-pkg-resources python-samba python-tdb rdiff-backup supervisor
apt purge python2-minimal libpython2.7-minimal
```

It's a real pain when the uninstall of a package invokes many other scripts and does not allow to proceed with the uninstall when those scripts fail. In this case, `pyclean` is used to remove the auto-generated .pyc and .pyo files. You install a Python module package, like `python-samba` that consists (aside of probably other assets, .so libraries and binaries) of .py files, which are raw Python scripts. When you execute those scripts, the compiler automatically creates a `__pycache__` sub directory with .pyc cached versions of the raw scripts, which are then faster to execute in the future. When using optimisation flags, .pyo files are created. When removing an APT package, APT will never remove any file that is not packed together with the package and on install stored as a file list in `/var/lib/dpkg/info/<package_name>.list`, which totally makes sense to not have something unwanted removed. However, due to the Python cache files, this means that all cached files together with the Python modules directory structure would remain on the system, not a clean uninstall. So the uninstall invokes scripts which do this cleanup: `/var/lib/dpkg/info/<package_name>.prerm` and/or `/var/lib/dpkg/info/<package_name>.postrm`, which invoke `pyclean` which does this job. `pyclean` however is a Python module as well, so if there is anything broken with the depths of the Python setup, one is not easily able to "simply" purge everything a start from scratch, and a reinstall fails for the same reason. There are a few other basic Python executables like `pyversions` which may cause similar issues. In the end, everything is a file, so if one knows where all files are located, one could remove them all manually. What I would wish for is a way to force an uninstall regardless of errors in the pre/post removal scripts, and a way to force a (re)install, regardless if existing files would be overwritten by it. While this of could could be unclean, it would make live would easier in cases like this where a single broken or missing file somewhere could break you whole APT functionality as every on every call it tries and fails to fix the affected packages.
**EDIT: Now I found `--force-remove-reinstreq` and `--force-overwrite` dpkg options, which could serve that needs, i.e. `dpkg -P --force-remove-reinstreq <package_name>`.**

Joulinar, 2020-12-10:
> Ah sorry, I think it's /etc/smb.conf, hopefully it's still there.

nope, SAMBA config file should be located at `/etc/samba/smb.conf`

MichaIng, 2020-12-10:
It looks like some more files have been lost them. Could be worth to have a look into `/lost+found`, e.g.:
```
grep -r '/var/log/samba' /lost+found/
```
to see if it lost the file system link and landed there.

Hsewdot, 2020-12-10:
Ok, thank your explanation it is very clear. I understand the source of the issue now. 

You'll find all the logs below :

<details>

<summary> apt install --reinstall python2-minimal log </summary>

```
root@DietPi:/home/dietpi# apt install --reinstall python2-minimal
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 python-crypto : Depends: python (< 2.8) but it is not going to be installed
                 Depends: python (>= 2.7~) but it is not going to be installed
                 Depends: python:any (< 2.8)
                 Depends: python:any (>= 2.7.5-5~)
 python-dnspython : Depends: python:any (< 2.8)
                    Depends: python:any (>= 2.7~)
 python-ldb : Depends: python (< 2.8) but it is not going to be installed
              Depends: python (>= 2.7~) but it is not going to be installed
              Depends: python:any (< 2.8)
              Depends: python:any (>= 2.7~)
              Depends: libpython2.7 (>= 2.7) but it is not going to be installed
 python-meld3 : Depends: python:any (< 2.8)
                Depends: python:any (>= 2.7.5-5~)
 python-pkg-resources : Depends: python:any (< 2.8)
                        Depends: python:any (>= 2.7~)
 python-samba : Depends: samba-libs (= 2:4.9.5+dfsg-5+deb10u1) but it is not going to be installed
                Depends: python (< 2.8) but it is not going to be installed
                Depends: python (>= 2.7~) but it is not going to be installed
                Depends: python:any (< 2.8)
                Depends: python:any (>= 2.7~)
                Depends: libpython2.7 (>= 2.7) but it is not going to be installed
                Depends: python-talloc (>= 2.1.9) but it is not going to be installed
 python-tdb : Depends: python (< 2.8) but it is not going to be installed
              Depends: python (>= 2.7~) but it is not going to be installed
              Depends: python:any (< 2.8)
              Depends: python:any (>= 2.7~)
              Depends: libpython2.7 (>= 2.7) but it is not going to be installed
 python2-minimal : Depends: python2.7-minimal (>= 2.7.16-1~) but it is not going to be installed
 rdiff-backup : Depends: python (>= 2.7) but it is not going to be installed
                Depends: python (< 2.8) but it is not going to be installed
 supervisor : Depends: python:any (< 2.8)
              Depends: python:any (>= 2.7~)
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

</details>

<details>

<summary> log for `apt purge python-crypto  python-dnspython ....` </summary>

```
root@DietPi:/home/dietpi# apt purge python-crypto  python-dnspython python-ldb python-meld3 python-pkg-resources python-samba python-tdb rdiff-backup supervisor
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 python2-minimal : Depends: python2.7-minimal (>= 2.7.16-1~) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

```

</details>

[... 85 lines cut ...]
-------

**Here is the command regarding samba part**

<details>

<summary>log for grep -r '/var/log/samba' /lost+found/</summary>

```
root@DietPi:/home/dietpi# grep -r '/var/log/samba' /lost+found/
/lost+found/#787212/logon/genlogon/genlogon.pl:# 1. Log the user connection to /var/log/samba/netlogon.log
/lost+found/#787212/logon/genlogon/genlogon.pl:open LOG, ">>/var/log/samba/netlogon.log";
/lost+found/#270584:    log file = /var/log/samba/log.%m
/lost+found/#270651:   log file = /var/log/samba/log.%m
/lost+found/#270651:# We want Samba to only log to /var/log/samba/log.{smbd,nmbd}.
/lost+found/#282688:    log file = /var/log/samba/log.%m
/lost+found/#655520:"/var/log/samba/log.smbd" 2020-11-21-11:23:37
/lost+found/#655520:"/var/log/samba/log.nmbd" 2020-11-22-0:0:2
/lost+found/#655520:"/var/log/samba/log.samba" 2020-5-19-0:0:0
/lost+found/#666908:    --logfile=/var/log/samba/user-syncpasswords.log
/lost+found/#666908:    --logfile=/var/log/samba/user-syncpasswords.log
Binary file /lost+found/#673442 matches
```

</details>

TtuxX, 2020-12-21:
Hi there @MichaIng, I hope you are doing great.

I just wanted to thank you the time and dedication you spent to solve my friend @Hsewdot's issue.

You are a great project leader and the best I've ever seen taking over the project of a (great) founder.
We the Dietpi community are incredibly fortunate to have you (as well as @Joulinar and the other Dietpi contributors) and you are doing a great work with new developments and bug fixing.
I often read you as well when I scroll through the Nextcloud Github or forum and your contribution is tremendous.

Thank you again for your sense of service and help, for your time and your gentleness.

I sent you 30€ through Dietpi's PayPal and I hope you will be able to treat yourself to a good bottle of your favourite drink, a small gift or something else.

I hope that you will be able to spend a lot of time with your loved ones. I wish you a Merry Christmas and a Happy New Year in advance and will be happy to continue to follow your contributions on Github and other platforms.

Kind regards

TtuxX

MichaIng, 2020-12-21:
Many thanks for your kind feedback and donation. Enjoy your Christmas and holidays as well, as much as possible with current circumstances 😷.

It seems like we didn't solve @Hsewdot's issue yet. Seeing this large amount of files in lost+found, including a binary, only looking for this which contain the samba log path, makes me think that it cannot reasonably be fixed. Even if we try on purging those nasty packages (`dpkg -P --force-all python2.7-minimal python2-minimal`, the `force-all` option includes all individual force options, last resort hammer 😉) likely a bunch of other issue will appear or cause non-visible issues already. My recommendation at this point is to do a fresh reinstall (if you didn't do that already), at best on a new SD card, so you can recover what is possible from the old one. Then doing some bad blocks check before re-using it, via Rufus from Windows or from Linux via:
```
badblocks -v /dev/xyz2
```

MichaIng, 2021-03-25:
I mark the issue as closed. Feel free to reopen if the issue persists with the latest update.

closed 2021-03-25 by MichaIng