MichaIng/DietPi #3878: jackett stuck on activating

pengocha, 2020-10-29:
os: DietPi v6.32.2

Since a few days my jackett won't start anymore, it just shows the status 'activating'. I usually can solve this by running 'sudo dietpi-software reinstall 147' but this gives me an error everytime i try:

 APT update                                                                                                           │
│  - Command: apt-get -q update                                                                                        │
│  - Exit code: 100                                                                                                    │
│  - DietPi version: v6.32.2 (MichaIng/master) | HW_MODEL:  | HW_ARCH:  | DISTRO:                                      │
│  - Image creator: DietPi Core Team                                                                                   │
│  - Pre-image: Raspbian Lite                                                                                          │
│  - Error log:                                                                                                        │
│ Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                      │
│ Hit:2 https://apt.sonarr.tv develop InRelease                                                                        │
│ Hit:3 https://download.mono-project.com/repo/debian raspbianbuster InRelease                                         │
│ Hit:4 https://archive.raspberrypi.org/debian buster InRelease                                                        │
│ Ign:5 https://download.webmin.com/download/repository sarge InRelease                                                │
│ Hit:6 https://download.webmin.com/download/repository sarge Release                                                  │
│ Unable to parse package file /var/lib/apt/extended_states (1)                                                        │
│ Can't call method "policy" on an undefined value at /usr/bin/apt-show-versions line 53.                              │
│ Reading package lists...                                                                                             │
│ E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ;        │
│ apt-show-versions -i'                                                                                                │
│ E: Sub-process returned an error code  

I hope this is enough information, any help is welcome!

Joulinar, 2020-10-29:
Hi,

many thanks for your message. You could clear apt cache and try again installation of Jackett.

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

pengocha, 2020-10-30:
Thanks for the reply, tried it out just now;
Getting this after i run it: 

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@DietPi:~# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Err:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Get:2 https://apt.sonarr.tv develop InRelease [13.3 kB]
Get:3 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Err:2 https://apt.sonarr.tv develop InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Get:4 https://download.mono-project.com/repo/debian raspbianbuster InRelease [2,238 B]
Err:3 https://archive.raspberrypi.org/debian buster InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Err:4 https://download.mono-project.com/repo/debian raspbianbuster InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Ign:5 https://download.webmin.com/download/repository sarge InRelease
Get:6 https://download.webmin.com/download/repository sarge Release [16.9 kB]
Get:7 https://download.webmin.com/download/repository sarge Release.gpg [173 B]
Err:7 https://download.webmin.com/download/repository sarge Release.gpg
  Signed file isn't valid, got 'NODATA' (does the network require authentication?)
Error: No information about packages! (Maybe no deb entries?)
Reading package lists... Done
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
E: The repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' is not signed.
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
E: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.
E: Failed to fetch https://apt.sonarr.tv/dists/develop/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'https://archive.raspberrypi.org/debian buster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://download.mono-project.com/repo/debian/dists/raspbianbuster/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'https://download.mono-project.com/repo/debian raspbianbuster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: GPG error: https://download.webmin.com/download/repository sarge Release: Signed file isn't valid, got 'NODATA' (does the network require authentication?)
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code
root@DietPi:~#

Joulinar, 2020-10-30:
Do you use a proxy or something like this to connect to the internet?

pengocha, 2020-10-31:
I'm behind a router but it has been that way for over a year now

Joulinar, 2020-10-31:
hmm usually these error is related to incorrect proxy settings. But if you just have a standard setup with normal internet router, this shouldn't be an issue.

@MichaIng 
any ideas?

pengocha, 2020-10-31:
I also can connect remotly from another device using ssh and access other services like sonarr radarr and transmission. Did a netwerk test from the dietpi config and everything was ok there. I'm connected by lan btw.

Joulinar, 2020-10-31:
that's fine. Somehow `apt` is not working correctly. Usually @MichaIng has good ideas 😄

MichaIng, 2020-10-31:
Proxy was also the only idea I got from searching for the error message, especially since all repositories are affected the same way. Does it work to access the file via curl directly?
```
curl -sSfL http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease
```

Joulinar, 2020-10-31:
or could it be some incorrect settings about an unused proxy on `/etc/apt/apt.conf.d` 🤔 

```
Acquire::http::proxy "http://<proxy>";
Acquire::https::proxy "https://<proxy>";
Acquire::ftp::proxy "ftp://<proxy>";
Acquire::socks::proxy "socks:<proxy>";
```

pengocha, 2020-11-01:
curl -sSfL http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease gives me a list of packages, if it helps i can paste them.
In my /etc/apt/apt.conf.d folder i have this:

root@DietPi:/etc/apt/apt.conf.d# ls
01autoremove-kernels  20apt-show-versions  70debconf  97dietpi  99-dietpi-cache  99-dietpi-force-ipv4

Joulinar, 2020-11-01:
what happen if you remove all external drives and just try to boot from SD card?

pengocha, 2020-11-01:
A message asking me for root password for maintenance, after entering the password i enter the terminal. All the services are deactivated. I run dietpi-software reinstall 147 the same error after some time.

MichaIng, 2020-11-01:
Which RPi model is it actually? I think RPi 3+ and 4 will automatically prefer SD card to boot from, while on RPi 3 (non-plus) I am not sure if programming the OTP to boot from USB implies that it boots from USB whenever a drive with bootloader found. Or is there some sort of overlayfs loaded?

Let's get an overview of what is attached and mounted:
```
df -T
lsblk
```

webmin was btw already/still installed in your first post. Uninstalling it through `dietpi-software` would have removed the APT repository your output shows that it is still present.

pengocha, 2020-11-01:
> root@DietPi:~# df -T
> Filesystem     Type      1K-blocks       Used Available Use% Mounted on
> /dev/root      ext4       15025172    8015852   6531500  56% /
> devtmpfs       devtmpfs     316932          0    316932   0% /dev
> tmpfs          tmpfs        350212        396    349816   1% /dev/shm
> tmpfs          tmpfs        350212       7200    343012   3% /run
> tmpfs          tmpfs          5120          4      5116   1% /run/lock
> tmpfs          tmpfs        350212          0    350212   0% /sys/fs/cgroup
> tmpfs          tmpfs         51200       1332     49868   3% /var/log
> tmpfs          tmpfs       1047552          0   1047552   0% /tmp
> /dev/sda1      fuseblk   976758784  668208896 308549888  69% /media/exfat
> /dev/mmcblk0p1 vfat         258096      65600    192496  26% /boot
> tmpfs          tmpfs         70040          0     70040   0% /run/user/992
> tmpfs          tmpfs         70040          0     70040   0% /run/user/0
> /dev/sdb1      fuseblk  1953480700 1938031876  15448824 100% /mnt/F474B7AA74B76DCC

> root@DietPi:~# lsblk
> NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
> sda           8:0    0 931.5G  0 disk
> └─sda1        8:1    0 931.5G  0 part /media/exfat
> sdb           8:16   0   1.8T  0 disk
> └─sdb1        8:17   0   1.8T  0 part /mnt/F474B7AA74B76DCC
> sdc           8:32   0 931.5G  0 disk
> └─sdc1        8:33   0 931.5G  0 part
> mmcblk0     179:0    0  14.9G  0 disk
> ├─mmcblk0p1 179:1    0   256M  0 part /boot
> └─mmcblk0p2 179:2    0  14.6G  0 part /
> root@DietPi:~#
>

pengocha, 2020-11-02:
I'm not 100% sure but it looks like a problem with the sd card. Even though i formated it multiple times over the last couple of days the problem wans't solved. I took a new sd card with a clean image file and my hdd's connected and the problem was gone. i'm reinstalling everything as i speak. Keep you posted.

Joulinar, 2020-11-02:
Yes that's a good reason that your SD card got damaged

pengocha, 2020-11-03:
Reinstalled everything on a new sd and its working again, thread can be closed, thank you all for helping!

Joulinar, 2020-11-03:
ok to bad that your SD card gave up. But at least everything working now using new SD card. Going to close the topic. Feel free to reopen if needed.

closed 2020-11-03 by Joulinar