MichaIng/DietPi #3881: Waveshare32 LCD blank screen

black00019, 2020-10-31:
The waveshare32 LCD have a blank screen with the last dietpi update.
If i install the new driver from waveshare site, the screen work
I think which kernel 5 is not compatible with waveshare driver on dietpi
please upgrade
thanks

MichaIng, 2020-10-31:
Many thanks for your report.

Please fill the bug report template, at least tell us which SBC and which Debian version you use:
```
echo $G_HW_MODEL_NAME $G_DISTRO_NAME
```

I assume now that you're using an RPi. Found that indeed last month waveshare updated their device tree overlays for Linux 5.4 and we ship an old overlay from our website. I'll update our script to point to the waveshare GitHub repository as resource: https://github.com/waveshare/LCD-show

MichaIng, 2020-10-31:
This fixes it: https://github.com/MichaIng/DietPi/commit/4f4c0b12361a347dfedf2be0e04192da12b024b5
Changelog: https://github.com/MichaIng/DietPi/commit/7c33908e8f2e56f1234303aeeab47bdb625790d9

closed 2020-12-20 by MichaIng