MichaIng/DietPi #3883: unable to Run now to update DietPi from v6.30.0 to v6.33.3.

ROMARICX, 2020-11-02:
#### Details:
- Date           | Mon Nov  2 13:40:10 CET 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | cp -Rf DietPi-master/dietpi /boot/
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
cp: cannot remove '/boot/dietpi/boot': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-autostart': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-backup': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-bugreport': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-cleaner': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-cloudshell': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-config': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-cpuinfo': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-cron': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-drive_manager': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-explorer': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-launcher': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-led_control': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-letsencrypt': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-login': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-morsecode': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-services': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-software': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-survey': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-sync': Read-only file system
cp: cannot remove '/boot/dietpi/dietpi-update': Read-only file system
cp: cannot remove '/boot/dietpi/func/change_hostname': Read-only file system
cp: cannot remove '/boot/dietpi/func/create_mysql_db': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-banner': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-benchmark': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-globals': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-logclear': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-obtain_hw_model': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-optimal_mtu': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-ramlog': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-set_cpu': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-set_hardware': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-set_software': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-set_swapfile': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-set_userdata': Read-only file system
cp: cannot remove '/boot/dietpi/func/dietpi-wifidb': Read-only file system
cp: cannot remove '/boot/dietpi/func/obtain_network_details': Read-only file system
cp: cannot remove '/boot/dietpi/func/run_ntpd': Read-only file system
cp: cannot remove '/boot/dietpi/misc/dietpi-arr_to_RAM': Read-only file system
cp: cannot remove '/boot/dietpi/misc/dietpi-justboom': Read-only file system
cp: cannot remove '/boot/dietpi/misc/dietpi-nordvpn': Read-only file system
cp: cannot remove '/boot/dietpi/misc/start_kodi': Read-only file system
cp: cannot create regular file '/boot/dietpi/patch_file': Read-only file system
cp: cannot remove '/boot/dietpi/postboot': Read-only file system
cp: cannot remove '/boot/dietpi/preboot': Read-only file system

Joulinar, 2020-11-02:
Many thanks for your report.

Your boot partition is set to r/o. Pls can you try to reboot your system.

ROMARICX, 2020-11-02:
No way. i wrote: mount -o remount,rw /
and then reboot Allo
But the dietpi-update results results was the same

Joulinar, 2020-11-02:
can you check some thing

```
cat /etc/fstab
dmesg -l emerg,alert,crit,err
lsblk -o name,fstype,label,size,ro,type,mountpoint
```

ROMARICX, 2020-11-02:
cat /etc/fstab
#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#FTP Client Mount--------------------------------------------------
#/mnt/ftp_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#NFS Client Mount--------------------------------------------------
#/mnt/nfs_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

proc            /proc           proc    defaults				0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0

#Internal Drives---------------------------------------------------
UUID=1435-2586    /boot    auto    defaults,noatime    0 0
UUID=b81a0b95-50e5-4f5f-8be9-0329475b1e85    /    auto    defaults,noatime    0 1
tmpfs /var/log tmpfs defaults,size=50m,noatime,nodev,nosuid,mode=1777 0 0
/var/swap    none    swap    sw    0   0

ROMARICX, 2020-11-02:
dmesg -l emerg,alert,crit,err
[    0.362308] i2c i2c-2: of_i2c: invalid reg on /i2c@b0178000/mpu6515@68
[    0.383179] atc260x_i2c 0-0065: icm_ohm not config, use default value
[    0.407578] atc260x-switch-ldo atc2603c-switch1.1: pmic_ver=0
[    0.412223] atc260x-pm atc2603c-pm.0: _set_s2_mode() can not get of_property s2_mode 
[    0.914625] owl_dss DSI: error!dsihw_parse_params, fail to get lcd reset gpio
[    0.964957] no blk dev
[    0.964969] read MiscInfoHeader failed
[    0.964979] read Head failed
[    0.964988] OWL_HDMI_ERR: failed to read hdcp key from secure storage
[    0.965210] Bluetooth: HCI Realtek H5 protocol registration failed
[    0.976890] pcm512x 2-004c: Failed to reset device: -6
[    0.977464] pcm512x 2-004d: Failed to reset device: -6
[    0.981928] [SNDRV]:[atc2603a_get_cfg] Fail to get device_node

[    0.981942] [SNDRV]:[atc2603a_init] audio get cfg failed!

[    0.982289] [SNDRV]:[atc2603c_platform_probe] atc2603c_platform_probe

[    0.982316] [SNDRV]:[atc2603c_platform_probe] what's my lucky draw 220

[    0.982763] [SNDRV]:[atm7059_dai_probe] it's ok 0

[    0.983508] [SNDRV]:[atm7059_link_init] atm7059_link_init

[    0.983557] [SNDRV]:[atm7059_audio_gpio_init] find i2s_switch_gpio property fail

[    0.983572] [SNDRV]:[atm7059_audio_gpio_init] find earphone_detect_gpios property fail

[    0.983596] [SNDRV]:[atm7059_audio_gpio_init] GPIO[126] request failed

[    0.983607] [SNDRV]:[atm7059_link_init] maybe using irq
[    1.129094] atc2603c-audio atc260x-audio: ASoC: Failed to add route AOUT FL FR Mixer -> AOUT MIXER -> ADC0 Mux
[    1.141723] train_Msize:1024, train_ces:1
[    1.162784] start:0xc0008000, end:0xc0bff2d8, checksum:0xf0b02035
[    2.968768] read_mi_head, line 429, magic 0x0, length 0, item num 0
[    2.968800] read Head null
[    2.968821] OWL_HDMI_ERR: failed to read hdcp key from secure storage
[    5.672070] FAT-fs (mmcblk0p1): codepage cp437 not found
[    5.673789] FAT-fs (mmcblk0p1): codepage cp437 not found
[    5.676484] EXFAT: Codepage cp437 not found
[    8.231794] EXT3-fs (mmcblk0p2): error: couldn't mount because of unsupported optional features (240)
[    8.232243] EXT2-fs (mmcblk0p2): error: couldn't mount because of unsupported optional features (240)
[    9.256445] atc260x-irkeypad atc2603c-irkeypad.0: Get ir_ch failed ret = -22 
[   10.483783] found sparky chip 10014639ca094994 provisioned
[   10.483957] idx 3077 newmac 17f7010c06
[   10.845520] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.848492] FAT-fs (mmcblk0p1): Filesystem has been set read-only
[   10.851348] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.854352] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.865037] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.872644] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.877186] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.880410] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.884063] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.887391] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   10.890329] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 69209)
[   11.756708] set port reset!80
[   11.876908] set port reset!80
[   12.176677] set port reset!80
[   12.296845] set port reset!80
[10075.695571] FAT-fs (mmcblk0p1): error, fat_free_clusters: deleting FAT entry beyond EOF
[10075.703686] FAT-fs (mmcblk0p1): Filesystem has been set read-only
[10610.807114] FAT-fs (mmcblk0p1): error, fat_free_clusters: deleting FAT entry beyond EOF
[10610.815289] FAT-fs (mmcblk0p1): Filesystem has been set read-only

ROMARICX, 2020-11-02:
lsblk -o name,fstype,label,size,ro,type,mountpoint
NAME        FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT
mmcblk0                   14.4G  0 disk 
├─mmcblk0p1 vfat   misc     48M  0 part /boot
└─mmcblk0p2 ext4   rootfs 14.4G  0 part /

Joulinar, 2020-11-02:
your boot partition doesn't looks healthy. Quite some error messages on the kernel log.

@MichaIng any ideas

ROMARICX, 2020-11-02:
Txs have you a suggestion?

MichaIng, 2020-11-03:
Please try to fix the file system:
```
dosfsck /dev/mmcblk0p1
```

ROMARICX, 2020-11-03:
dosfsck /dev/mmcblk0p1
fsck.fat 4.1 (2017-01-24)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action

ROMARICX, 2020-11-03:
Reclaimed 1124 unused clusters (2301952 bytes).
Perform changes ? (y/n) y
/dev/mmcblk0p1: 70 files, 5952/24519 clusters

MichaIng, 2020-11-03:
Let's see if that was enough:
```
mount -o remount,rw /boot
dietpi-update 1
```

ROMARICX, 2020-11-04:
Hello. Proble Solved! Update Ok.
Thanks a lot.

closed 2020-11-04 by ROMARICX