MichaIng/DietPi #3884: Cannot install Wireguard during initial install

jadaworth, 2020-11-02:
Hi all, I am trying to install Dietpi on my new Raspberry Pi 4. The only software I want to install is Wireguard (for use as a client). I am getting the following error. I am connecting via ssh on my LAN.

```
Setting up wireguard-dkms (1.0.20200908-1~bpo10+1) ...
Removing old wireguard-1.0.20200908 DKMS files...

------------------------------
Deleting module version: 1.0.20200908
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-1.0.20200908 DKMS files...
It is likely that 5.4.72-v8+ belongs to a chroot's host
Building for 5.4.72+, 5.4.72-v7+, 5.4.72-v7l+ and 5.4.72-v8+
Building initial module for 5.4.72+
Error! Bad return status for module build on kernel: 5.4.72+ (aarch64)
Consult /var/lib/dkms/wireguard/1.0.20200908/build/make.log for more information.
dpkg: error processing package wireguard-dkms (--configure):
 installed wireguard-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 wireguard-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Software | APT upgrade
 - Command: apt-get -qq upgrade
```

My current kernel is 5.4.72-v8+ according to "uname -r" if that helps. Please let me know if I can provide any more information. I am the noobiest of noobs when it comes to Linux.

Joulinar, 2020-11-02:
Hi,

many thanks for your message. This is a know downside of the Raspberry 64bit OS Beta. Unfortunately there is nothing we can do from DietPi side because the Kernel Header Package is still missing and not yet developed by Raspberry Foundation. The entire challenge with WireGuard and Raspberry OS 64bit Beta is descriped on this Github post https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4

There are as well some workaround described. 

Keep in mind, the Raspberry 64bit OS is still in Beta status and under development by Raspberry Foundation. It's not intent for production usage.

jadaworth, 2020-11-02:
> Hi,
> 
> many thanks for your message. This is a know downside of the Raspberry 64bit OS Beta. Unfortunately there is nothing we can do from DietPi side because the Kernel Header Package is still missing and not yet developed by Raspberry Foundation. The entire challenge with WireGuard and Raspberry OS 64bit Beta is descriped on this Github post [raspberrypi/Raspberry-Pi-OS-64bit#4](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4)
> 
> There are as well some workaround described.
> 
> Keep in mind, the Raspberry 64bit OS is still in Beta status and under development by Raspberry Foundation. It's not intent for production usage.

Wow thanks for the quick reply! You're the best. So can I use Raspberry 32bit OS? How much slower will it be than 64bit? And can I install the 32bit OS through DietPi?

Joulinar, 2020-11-02:
I don't think that there is any benefit of the 64bit version compare to 32bit. At least as long as you don't have a software that is specifically designed for 64bit. DietPi 32bit for RPi is the standard default image and stable for long time.

jadaworth, 2020-11-02:
Got wireguard up and running on 32-bit DietPi. Thanks again for your help. 

Now I have to figure how to install .Net Core on ARM architecture...might just build a 64bit Ubuntu machine with an Intel or AMD CPU in the near future. Much more expensive than a Pi but probably much easier to work with lol

Joulinar, 2020-11-02:
maybe this will help https://edi.wang/post/2019/9/29/setup-net-core-30-runtime-and-sdk-on-raspberry-pi-4

Anyway I still hope the Foundation will finish the 64bit image sooner or later. And hopefully they will fix the challenge with the header package. Or they will release kernel 5.9. We will see how it's going in 2021.

MichaIng, 2020-11-03:
RPi Linux 5.9 is in progress, it's being worked on in the firmware repository next branch already: https://github.com/raspberrypi/firmware/tree/next
But no WireGuard module included 😢: https://github.com/raspberrypi/firmware/tree/next/modules/5.9.2-v8%2B/kernel/net
I checked the kernel sources and it is actually enabled for RPi4 and RPi3 64-bit builds only it seems: https://github.com/raspberrypi/linux/tree/rpi-5.9.y/arch/arm64/configs
Here it is not present: https://github.com/raspberrypi/linux/tree/rpi-5.9.y/arch/arm/configs

I think we need to request this as being enabled on all models/architectures, a must-have IMO 😃.

Joulinar, 2020-11-03:
let's see how long it will take to get 5.9 ready. But this would mean quite some changes for our WG implementation. isn't it?

MichaIng, 2020-11-03:
Only a minor change required our side. We already detect WireGuard modules shipped with kernel packages to support the one shipped with Armbian kernels. But the location is different, which is why I didn't see it first: `/lib/modules/*/kernel/net/wireguard/wireguard.ko` vs `/lib/modules/*/kernel/drivers/net/wireguard/wireguard.ko`.

Just checked a current Armbian kernel and it's the same there (now). Not sure if it has changed or a symlink was present 🤔, however will change `dietpi-software` to detect it at both locations.

Ah probably a manually build WireGuard places it differently.

Joulinar, 2020-11-03:
ah ok good to know

MichaIng, 2020-11-03:
At least it was in the path without `/drivers/`: https://forum.armbian.com/topic/10390-cant-load-wireguard-kernel-module/
Now no symlink and in the path with `/drivers/`. Now it's detected at both locations: https://github.com/MichaIng/DietPi/commit/8ee591baa826860aad0454225418a03497fa2a96

Joulinar, 2020-11-06:
ok I'm going to close this issue. Feel free to reopen if needed.

closed 2020-11-06 by Joulinar