MichaIng/DietPi #3885: PineH64 Model B Ethernet fails on Start-up

rgreenosborne, 2020-11-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.3x
- Distro version | Buster
- Kernel version | many tried
- SBC model | Pine H64 Model B
- Power supply used | 5V 3A Barrel Plug
- SDcard used | Class 10
#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?

Fresh install.

- Can this issue be replicated on a fresh installation of DietPi?

Yes.  Ethernet fails 35 seconds into booting.

#### Steps to reproduce
1. Follow install guide.  
2. Boot current image on Pine H64 Model B

#### Expected behaviour
Normal install procedure

#### Actual behaviour
35 seconds after boot the ethernet stops working with a system message:

[35.xxx] dwmac-sun8i 5020000. ethernet eth0 is down

Could this be an artifact of using the same Armbian base for Model A and Model B variants? re #3286

MichaIng, 2020-11-03:
Many thanks for your report.

Since there is now an Armbian image for version B, I'll check and compare those, especially kernel and device tree. Although a first working and later failing Ethernet is not what I would expect.

Did the first boot setup finished, actually, I mean do the 35 seconds apply after first login or already during first boot setup? In case probably there is some faulty step.

I'll anyway create a new image now since ours is quite old (from February). It will be based on the model B pre-image. If any image works on both models, I'd assume the model B image to work better on model A than the model A image on model B, due to missing onboard WiFi.

MichaIng, 2020-11-03:
Image ready, would be awesome if you could test if it works better: https://dietpi.com/downloads/images/testing/DietPi_PINEH64-ARMv8-Buster.7z

rgreenosborne, 2020-11-03:
Followed identical writing process on the same SD card for testing. Appears to be working perfectly! I completed a very simple config and added Hotspot Wifi and it worked as well.

Rebooted and finishing configuration over ssh via eth0. Thanks @MichaIng

MichaIng, 2020-11-03:
Great, let's hope that is works on Model A as well, although even the initial image request was for model B 😉.

closed 2020-11-03 by MichaIng