MichaIng/DietPi #3886: Dietpi update from 6.30.0 to 6.33 - error no space left

nico3182, 2020-11-02:
hello

I did the update on ALLO Usbridge (Raspbian Lite/Rpi CM 3+ (armv71) )of Dietpi version 6.30 to 6.33
I have the error: 
error writing '/boot/dietpi/patch_file': No space left on device

What should I do ?
(i have a screenshot but it fails to upload....)

Thank you for your help

Best regards
Nicolas

Joulinar, 2020-11-02:
Hi,

many thanks for your message. Looks like you file system is full. Pls can you post `df -h`

nico3182, 2020-11-03:
Hi,

Thanks for your answer.
Here is the result .
Do you mean no space on the sd card ?
I recovered the Allo Usbridge with dietpi 6.30, yet now I can no more
access to QOBUZ playlists or favorites.
Thanks for help
Best regards
Nicolas

nico3182, 2020-11-03:
I also send you the screen with errror message
Nicolas

Joulinar, 2020-11-03:
Pls login to Github directly and post your output

nico3182, 2020-11-03:
Result

nico3182, 2020-11-03:
Did you receive the picture? I cannot see it on the discussion

ravenclaw900, 2020-11-03:
GitHub doesn't support e-mailing in pictures.  You need to post it on the website.

Joulinar, 2020-11-03:
or even better to login via SSH to your system and copy/past the content

nico3182, 2020-11-03:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  1.9G   13G  14% /
devtmpfs        484M     0  484M   0% /dev
tmpfs           488M     0  488M   0% /dev/shm
tmpfs           488M  7.1M  481M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           488M     0  488M   0% /sys/fs/cgroup
tmpfs          1023M  8.0K 1023M   1% /tmp
tmpfs            50M  144K   50M   1% /var/log
/dev/mmcblk0p1   42M   42M   512 100% /boot

Joulinar, 2020-11-03:
@MichaIng 
is that normal that ALLO images have 42 MB BootFS only?? That seems to be pretty small. On normal RPi we use 256 MB

nico3182, 2020-11-03:
hello can I just flash the sdcard with dietpi 6.33 and reboot the allo usbridge?

Joulinar, 2020-11-03:
Let's wait on the developer to have a look

MichaIng, 2020-11-03:
That is an older Raspbian Stretch image, isn't it? It's an issue with all older RPi images.

Those images were shipped with a smaller `/boot` partition which is now too small with the RPi4 and 64-bit kernel/dtb files. The kernel package install scripts have already been adjusted to skip RPi4 files on non-RPi4 and non-RPi4 files on RPi4, if a small `/boot` partition is found, but it stays close.

Probably we can get some space by removing non-essential files. Can you please show the content of the boot partition?
```
ls -Al /boot
```

nico3182, 2020-11-06:
hello,

this is the result:
bcm2708-rpi-0-w.dtb
bcm2708-rpi-b.dtb
bcm2708-rpi-b-plus.dtb
bcm2708-rpi-cm.dtb
bcm2708-rpi-zero.dtb
bcm2708-rpi-zero-w.dtb
bcm2709-rpi-2-b.dtb
bcm2710-rpi-3-b.dtb
bcm2710-rpi-3-b-plus.dtb
bcm2710-rpi-cm3.dtb
bcm2711-rpi-4-b.dtb
bootcode.bin
cmdline.txt
config.txt
COPYING.linux
dietpi
dietpi-README.md
dietpi.txt
.firmware_revision
fixup4cd.dat
fixup4.dat
fixup4db.dat
fixup4x.dat
fixup_cd.dat
fixup.dat
fixup_db.dat
fixup_x.dat
FSCK0000.REC
issue.txt
kernel7.img
kernel7l.img
kernel.img
LICENCE.broadcom
LICENSE.oracle
overlays
README.md
start4cd.elf
start4db.elf
start4.elf
start4x.elf
start_cd.elf
start_db.elf
start.elf
start_x.elf
System Volume Information

Thanks for your help
Nicolas

MichaIng, 2020-11-06:
Okay, probably you installed the first RPi4 supporting kernel at a time where it did not yet skip RPi4 stuff in case of low space. Please remove the following:
```
rm -Rf /boot/{fixup4*,kernel7l.img,start4*,System\ Volume\ Information,dietpi-README.md,README.md,bcm2711-rpi-4-b.dtb}
```

There is a file from a past file system check. Let's see which one it is:
```
head -10 /boot/FSCK0000.REC
```

nico3182, 2020-11-06:
In fact I've bought the Allo Usbridge and it was configured with Dietpi, I just installed the sofwares I needed to stream music.  Personnally, I don't relly understand how all that works....
I've entered both command lines as per your message. What should I do now ?
Thanks.

Joulinar, 2020-11-06:
can you retry the update?

MichaIng, 2020-11-06:
Also, could you paste the output of the second command, so we can check which file this was and if the original one is there or missing?

nico3182, 2020-11-06:
I had no output when I entered the command (I tried several times)
I did the update and it works fine ! very happy, a lot of thanks!
Do you want me to try something else?

MichaIng, 2020-11-06:
You could check the whole content:
```
more /boot/FSCK0000.REC
```
However, as everything required for boot is there and the `dietpi-update` recovered all DietPi scripts, to you may simply remove it:
```
rm /boot/FSCK0000.REC
```

nico3182, 2020-11-06:
when I entered more /boot/FSCK0000.REC, nothing was displayed.
Then I entered rm /boot/FSCK0000.REC.

nico3182, 2020-11-06:
all is working fine now.
Thanks a lot for your support, I am impressed by your reactivity and professionalism.
Best regards,
Nicolas

closed 2020-11-06 by nico3182