MichaIng/DietPi #3887: Unable to hear audio on analog 3.5 jack

corasaniti, 2020-11-02:
### Unable to hear audio on analog 3.5 jack

- DietPi version 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster 10.6 
- Kernel version | Linux DietPi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
- SBC model Rpi4
- SDcard used | SanDisk ultra

I set the option in the Audio Options of the DietPiConfig tool
rpi-bcm2835-3.5mm
I am not able to hear the sound from the speakers connected via the analog jack to my Rasp4
Thanks

MichaIng, 2020-11-03:
Many thanks for your report.

Could you please paste the output of the following commands:
```
amixer contents
cat /etc/asound.conf
```

And did you try it with `speaker-test` or with a different audio device to rule out a sound source/player issue?

corasaniti, 2020-11-03:
This is the output of my file 

`/etc/asound.conf`
```
pcm.!default {
        type hw
        card 0
        device 0
}
ctl.!default {
        type hw
        card 0
}

```
If I try speaker-test  I am not able to hear anything and in console I see the following log

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 512 to 65536
Period size range from 512 to 65536
Using max buffer size 65536
Periods = 4
was set period_size = 16384
was set buffer_size = 65536

  0 - Front Left
Time per period = 1.378552
  0 - Front Left
Time per period = 2.729755
  0 - Front Left
Time per period = 2.730159
  0 - Front Left
Time per period = 2.729889
  0 - Front Left`
`

Thanks

MichaIng, 2020-11-03:
Can you also paste the output of:
```
amixer contents
```

corasaniti, 2020-11-04:
Hi this is my amixer content
Thanks

root@DietPi:~# amixer
Simple mixer control 'HDMI',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]

MichaIng, 2020-11-04:
I think I found the reason: https://www.raspberrypi.org/blog/latest-raspberry-pi-os-update-may-2020/
It seems they changed the way how output channels are accessible. A shame since there seem to be no "auto" choice anymore, i.e. no way to have audio output to HDMI automatically, if plugged, else through 3.5mm, which was a good default choice before. Also it is now not possible to switch output with a simple amixer or alsamixer switch but the ALSA config file needs to be edited. What was meant to simplify things, IMO made it more complicated, aside of the fact that it broke all 3rd party configuration tools and personal scripts and the way users were used to the last years...
Another question is if this is RPi4 only or affects all RPi models, in prior case it would further complicate our scripts. And I was already wondering why the amixer instructions on official docs have gone and replaced by `raspi-config` instructions, i.e. too complicated now to do it with low-level tools.

@corasaniti 
Let's assure what I assume it is, please paste the output of the following:
```
aplay -l
```
It should show you two audio cards now, the first being HDMI output, the second being 3.5mm jack. If that is so, you should be able to get output to 3.5mm via:
```
sed -i 's/card 0/card 1/' /etc/asound.conf
```

I'll also check this on RPi2 later.

corasaniti, 2020-11-04:
Hi this is my output of
`aplay -l`

root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3

So with command 
`sed -i 's/card 0/card 1/' /etc/asound.conf`

I should be able to hear audio on analog 3.5jack

if I want to restore the configuration on the HDMI audio output, do I have to execute the following command? Correct?
`sed -i 's/card 1/card 0/' /etc/asound.conf`

right
Thanks

corasaniti, 2020-11-04:
Hi the command

`sed -i 's/card 0/card 1/' /etc/asound.conf`

it worked perfectly for me :-)

Now I am able to listen to the music on my speakers connected in analog on the 3.5mm jack and listen to the noise of the floppy drive emulated in Amiberry

Thanks: - /

MichaIng, 2020-11-05:
Okay, that is great.

I think 3.5mm jack for default sound output is fine as well.

MichaIng, 2020-11-05:
It is even worse. If no HDMI monitor is attached (or HDMI forcefully disabled), card 0 is the 3.5mm jack and the HDMI card is not available. It seems to make sense, but the problem is that is is extremely complicated (too complicated to implement) now to know which card ID which output channel will finally have. That alone could be solved with a two stage implemention:
1. A dedicated selector to enable/disable audio drivers: Onboard audio, dtoverlays for external cards etc.
2. Make the sound card selection an auto detection only, so only "available" sound cards are listed.

This assures that the correct sound card is chosen, regardless of its ID, which as well solves another issue better, where the KMS driver adds an additional sound card 0, which we currently work around by incrementing the IDs. But in general these sound card IDs are not fixed and it depends on the order in which all available output devices are loaded.

The big downside is that a sound card cannot be chosen with a single selection or even via dietpi.txt or scripted at boot, but includes mostly selecting the driver/overlay/module first, then the selecting the now available cards as a 2nd manual step.

But the worst thing is: If one selects the audio jack and then disables HDMI or unplugs the monitor or switches to `hdmi_drive=1` (DVI mode, no sound), the previously chosen sound card 1 does not exist anymore and it needs to be re-selected...

This could be worked around by directly setting `hdmi_drive=1` when selecting 3.5mm jack. It it is as I assume, then the card ID should be 0 as well. Sadly this does not work the other way round: `hdmi_drive=2` does not force the HDMI card 0 being available, it seems an HDMI monitor needs to be attached.

@corasaniti 
If you are in mood, could you do some tests to verify that RPi4 behaves the same:
```
G_CONFIG_INJECT 'hdmi_drive:0=' `hdmi_drive:0=1` /boot/config.txt # forces DVI mode on HDMI port 0
G_CONFIG_INJECT 'hdmi_drive:1=' `hdmi_drive:1=1` /boot/config.txt # forces DVI mode on HDMI port 1
reboot
cat /proc/asound/cards
```
Check if the last command now only lists "Headphones" (3.5mm jack) with ID 0, like:
```
2020-11-05 15:36:50 root@micha:~# cat /proc/asound/cards
 0 [Headphones     ]: bcm2835_headphonbcm2835 Headphones - bcm2835 Headphones
                      bcm2835 Headphones
```

If HDMI output is still shown, detach HDMI screen, then:
```
sed -i '/^hdmi_drive:[01]=/d' /boot/config.txt # Removes the HDMI drive entries to check effect of attached screen alone
reboot
cat /proc/asound/cards
```
The second test of course needs to be done via SSH then, when no HDMI monitor is attached 😉.

MichaIng, 2020-12-07:
Issue now solved via device tree overlay to disable HDMI or 3.5mm jack completely, if the other one is wanted: #3938
With this, the chosen onboard sound will always be on card0. The downside is that a reboot is required to chance it. But this is true in most cases anyway, e.g. enabled onboard audio the first time requires a reboot due to `dtparam=audio=off` => `on` and all other sound card selections are implemented via device tree overlays as well.

closed 2020-12-20 by MichaIng

Pisgah, 2025-03-08:
I also encountered the same situation, my solution is as follows:

[image]

and go to the bottom:

[image]

and then you have the sound on the speaker.

However, after each reboot, you will lose the speaker sound.
You must re-enter sudo dietpi-config --> Audio Options,
reset Sound card: [rpi-bcm2835-3.5mm],
then the system will reboot,

repeat the steps above, and the speaker sound will return.

Pisgah, 2025-11-22:
# BCM2835 3.5mm audio output fails on boot with DietPi on Raspberry Pi 4

## Description
On Raspberry Pi 4 running DietPi, the onboard BCM2835 3.5mm audio output does not produce sound immediately after boot. Selecting `rpi-bcm2835-3.5mm` in `dietpi-config → Audio` does not result in audio output. Manually specifying `hw:0,0` temporarily restores sound, but the setting is overwritten on reboot, creating a loop where audio is non-functional until manually corrected.

## Environment
- Device: Raspberry Pi 4  
- OS: DietPi (latest stable)  
- Audio devices:  
  - BCM2835 Onboard 3.5mm output  
  - Optional HDMI audio  
  - Optional USB audio DAC  

## Steps to Reproduce
1. Boot Raspberry Pi 4 with DietPi installed.  
2. Open `dietpi-config → Audio`.  
3. Select `rpi-bcm2835-3.5mm` as the default audio output.  
4. Reboot the system.  
5. Observe that no sound is produced.  
6. Manually setting `hw:0,0` restores audio temporarily.  
7. Reboot again — the setting is overwritten and audio fails again.

## Observed Behavior
- BCM2835 3.5mm output is unavailable immediately after boot.  
- DietPi auto-detection maps `rpi-bcm2835-3.5mm` to a virtual EQ card (`hw:0,0-eq`), which does not correspond to the real hardware until manually corrected.  
- Manual `hw:0,0` selection works temporarily but is overwritten on reboot.  

## Expected Behavior
- Selecting `rpi-bcm2835-3.5mm` in DietPi should produce functional audio on boot without manual intervention.  
- The mapping should persist across reboots.

## Impact
- Audio output fails immediately after boot without manual intervention.  
- Users must repeatedly select `hw:0,0` or otherwise intervene to restore sound.  
- Makes headless or automated deployments unreliable if 3.5mm audio is required.

## Logs
**aplay output:**
```bash
dietpi@PisgahIotServer:~$ sudo aplay -l
aplay: device_list:279: no soundcards found...
dmesg output (filtered for snd):

dietpi@PisgahIotServer:~$ sudo dmesg | grep snd
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave nvme.max_host_mem_size_mb=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 numa=fake=2 system_heap.max_order=0 smsc95xx.macaddr=XX:XX:XX:XX:XX:XX vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  root=PARTUUID=XXXX-XX rootfstype=ext4 rootwait net.ifnames=0 logo.nologo console=tty1 usbcore.autosuspend=-1 usb-storage.quirks=174c:55aa:u
Note: MAC address and PARTUUID have been masked for privacy.

Related Discussion
DietPi Issue #3887 comment 3314941167

Summary: On Pi4, BCM2835 3.5mm audio sometimes fails on boot due to ALSA driver initialization and DietPi auto-detection order. Manual hw:0,0 fixes temporarily, but may be overwritten on reboot. No universal solution reported; workarounds involve fixed card indices or disabling HDMI audio.

MichaIng, 2025-11-22:
I already explained above why switching to `hw:0,0` breaks it after reboot. It needs to stay at `rpi-bcm2835-3.5mm`, otherwise onboard audio is disabled in the firmware, as you can see in the kernel command-line.

If the card in the menu shows a `-eq` suffix, the equalizer in the "DietPi-JustBoom" submenu got enabled. Maybe it applies a too low volume. Select "EQ Configure" in that case and raise volume bars, or run `alsamixer -D eq` from console. Anyway, the equalizer is disabled when you (re)select a sound card. Hence try the following:

1. Select `rpi-bcm2835-3.5mm`, verify it does not show an `-eq` suffix. Never change to another sound card, it will break things expectedly on reboot, as explained.
2. Exit `dietpi-config` and start it again, to verify that there is still no `-eq` suffix. Also verify via `grep '^CONFIG_SOUNDCARD=' /boot/dietpi.txt` that it has no `-eq` suffix, so it is all persistent.
3. Reboot, verify above again. You can also check `/etc/asound.conf` that it shows
    ```
    pcm.!default {
    	type hw
    	card 0
    	device 0
    }
    
    ctl.!default {
    	type hw
    	card 0
    }
    ```
    Means no ALSA plugin enabled.
4. Test audio

Pisgah, 2025-11-23:
My current solution is to use the custom autostart script in dietpi-config:

```bash
#!/bin/bash
# Headless 3.5mm audio setup for DietPi AutoStart 14
LOGFILE="/xxxxxxxxxxx/headless_audio.log"
echo "$(date) - Starting headless audio setup..." >> $LOGFILE
sleep 10
# Load bcm2835 module if not loaded
if ! lsmod | grep -q snd_bcm2835; then
/sbin/modprobe snd_bcm2835 && echo "$(date) - snd_bcm2835 loaded." >> $LOGFILE
fi
# Set PCM volume
if aplay -l | grep -q "Headphones"; then
/usr/bin/amixer -c 1 sset 'PCM' 65% && echo "$(date) - PCM volume set to 65%." >> $LOGFILE
else
echo "$(date) - Headphones device not found." >> $LOGFILE
fi
echo "$(date) - Headless audio setup completed." >> $LOGFILE
exit 0 

This perfectly solves the problem; no need to touch dietpi-config-->audio``