MichaIng/DietPi #3891: Suggestion: Pi-Hole + DietPi Hotspot Compatibility

ReperakPro, 2020-11-06:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...I'd like to see the ability to automatically setup DietPi Hotspot to work with PiHole through DietPi-Software

#### Describe the solution you'd like:

- ...DietPi + Pi-Hole compatibility would make running a Hotspot with built-in adblocking easy as pie, as the current method to run Pi-Hole through a Hotspot is very complex and there's a severe lack of documentation on how to do this with any software or SBC.

#### Describe alternatives you've considered:

- ...I have tried setting the two softwares up together manually to no luck.

#### Additional context

- ...Nothing to say here.

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-11-06:
Many thanks for your suggestion.

Actually it should work OOTB. Assure to select the WiFi IP as local IP for Pi-hole so that clients connected to the AP are recognised as "local network" origins. Otherwise from within Pi-hole UI, assure that "all origins" are allowed to access the DNS resolver.

Joulinar, 2020-11-07:
yep it's that simple. During PiHole installation select `wlan0` as interface and once installation completed change DNS server settings inside `/etc/dhcp/dhcpd.conf` to point to `192.168.42.1` like this

```
ddns-update-style none;
default-lease-time 600;
max-lease-time 7200;
authoritative;
log-facility local7;

subnet 192.168.42.0 netmask 255.255.255.0 {
        range 192.168.42.10 192.168.42.50;
        option broadcast-address 192.168.42.255;
        option routers 192.168.42.1;
        option domain-name "local";
        option domain-name-servers 192.168.42.1;
}
```

restart dhcp server and you are done 
`systemctl restart isc-dhcp-server.service`

ReperakPro, 2020-11-09:
Thank you very much, this worked perfectly. Hopefully anybody else who would like to do the same thing as me can refer to this issue in the future.

closed 2020-11-09 by ReperakPro