MichaIng/DietPi #3892: dietpi-update gave error

tabishz, 2020-11-07:
#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux raspi4 5.4.51-v7l+ #1326 SMP Fri Jul 17 10:51:18 BST 2020 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | 5V 3A USB-C
- SDcard used | Samsung Evo 64GB

#### Additional Information (if applicable)
#### Details:
- Date           | Fri Nov  6 20:13:11 MST 2020
- Bug report     | e0a3d583-255d-467e-8bac-c29dd007521d
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux raspi4 5.4.51-v7l+ #1326 SMP Fri Jul 17 10:51:18 BST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:
1. sudo dietpi-update

#### Expected behaviour:
it should update the dietpi version
#### Actual behaviour:
Got the error as details pasted in Additional Details

Joulinar, 2020-11-07:
Hi,

Many thanks for your message.pls can you post an error message you are facing.

tabishz, 2020-11-07:
No specific error. At the end the dietpi-uodate highlighted some details and suggested I share those details on GitHub. I have shared those details above

It said apt-get -qq upgrade exited with code 100

Joulinar, 2020-11-07:
Pls run `apt update && apt upgrade` and share the output if there are errors

tabishz, 2020-11-09:
Thanks a lot for your help. Anyway, I tried the apt update && upgrade, no errors. After that, when I ran ```sudo dietpi-upgrade``` it went through without any errors.. so I guess it fixed itself.. Thanks.

Joulinar, 2020-11-09:
Usually there should have been some more error messages while facing the issue. But ok it's gone now. I will close the issue, feel free to reopen if needed.

closed 2020-11-09 by Joulinar