MichaIng/DietPi #3893: F2FS SSD over USB not booting since 6.33

Cjkeenan, 2020-11-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
root@BH-3B:~# cat /boot/dietpi/.version                                                                                                                                         
G_DIETPI_VERSION_CORE=6                                                                                                                                                         
G_DIETPI_VERSION_SUB=33                                                                                                                                                         
G_DIETPI_VERSION_RC=3                                                                                                                                                           
G_GITBRANCH='master'                                                                                                                                                            
G_GITOWNER='MichaIng'     
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
```
root@BH-3B:~# cat /etc/debian_version                                                                                                                                           
10.6                                                                                                                                                                            
root@BH-3B:~# echo $G_DISTRO_NAME                                                                                                                                               
buster 
```
- Kernel version | `uname -a`
```
root@BH-3B:~# uname -a                                                                                                                                                          
Linux BH-3B 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux   
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
root@BH-3B:~# echo $G_HW_MODEL_NAME                                                                                                                                             
RPi 3 Model B (armv7l) 
```
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.4A APC Power Strip
- SDcard used | (EG: SanDisk ultra)
WD Blue SSD 240GB

#### Additional Information (if applicable)

#### Steps to reproduce

1. Have a F2FS SSD over USB as OS drive
2. Update to v6.33 of DietPi
3. Reboot System

#### Expected behaviour

- System should reboot and start all services without any user intervention

#### Actual behaviour

[image]
- After pressing CTRL + D system boots fine and services start fine

#### Extra details

- If you need any more logs please let me know

Joulinar, 2020-11-07:
Many thanks for your message.

It might not be related but you have an `under voltage` message at the end. Means your system ist not powered enough. Do you use an external power supply? Did you tried using different USB port already?

Cjkeenan, 2020-11-07:
Yeah I tried via the Canakit 5V 2.5A power brick and that had no voltage warnings, but also had the same errors.

I entered the root password again and this is the specific output:
[image]

Joulinar, 2020-11-07:
your SSD is not mounted. Therefore you have all the error messages and long boot times. Let's check following

```
dmesg -l emerg,alert,crit,err
lsblk -o name,fstype,label,size,ro,type,mountpoint
```

Cjkeenan, 2020-11-07:
```
root@BH-3B:~# dmesg -l emerg,alert,crit,err                                                                                                                                                     
[    1.668161] vc_vchi_sm_init: failed to open VCHI service (-1)                                                                                                                                
[    1.668165] [vc_sm_connected_init]: failed to initialize shared memory service                                                                                                               
[  108.313382] vc_sm_cma_vchi_init: failed to open VCHI service (-1)                                                                                                                            
[  108.313396] [vc_sm_connected_init]: failed to initialize shared memory service                                                                                                               
[  108.563918] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)                                                                                                           
[  110.473947] Under-voltage detected! (0x00050000)                                                                                                                                             
[  125.034033] Under-voltage detected! (0x00050005)                                                                                                                                             
[  141.687467] Under-voltage detected! (0x00050000)                                                                                                                                             
[  929.993908] Under-voltage detected! (0x00050005)                                                                                                                                             
[  952.873872] Under-voltage detected! (0x00050005)                                                                                                                                             
[  992.393841] Under-voltage detected! (0x00050000)  
```
and 
```
root@BH-3B:~# lsblk -o name,fstype,label,size,ro,type,mountpoint                                                                                                                                
NAME   FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT                                                                                                                                                   
sda                 232.9G  0 disk                                                                                                                                                              
├─sda1 vfat   boot    256M  0 part /boot                                                                                                                                                        
└─sda2 f2fs         232.6G  0 part /  
```
These were both done after CTRL+D and the system boots as normal.

Joulinar, 2020-11-07:
hmm the issue seems to be with `BootFS` and this is `vfat` file system. At least that's the messages on your screen print above. Can you try d`osfsck /dev/sda1`

@MichaIng and ideas

Next to this you still have `Under-voltage detected` messages

Cjkeenan, 2020-11-07:
```
root@BH-3B:~# dosfsck /dev/sda1                                                                                                                                                                 
fsck.fat 4.1 (2017-01-24)                                                                                                                                                                       
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.                                                                                                             
1) Remove dirty bit                                                                                                                                                                             
2) No action                                                                                                                                                                                    
?       
```

Joulinar, 2020-11-07:
I know it's easier with having the SSD connected via USB only but your are running a high risk of running an unstable system. Maybe your issues in BootFS are already a result of this.

But let's wait on @MichaIng Usually he has good ideas if it comes down to file system issues.

MichaIng, 2020-11-07:
It is probably all related to the SSD powered via USB port. I am not 100% sure how the circuits are made on each RPi model, if USB device can have any effect on ARM voltage, but especially in case of peaks there is practically always an effect. So before investing time in any other investigations, first try to power the SSD via dedicated power supply and fix the under-voltage. Hard to not assume that failing external drive mounts, under-voltage warnings, and the external drive bring powered from RPi USB ports with 1.2A shared current are not related 😉. And of course random unmount and unstable voltage easily lead to file system errors. Of course it cannot be really said for a single error.

I just checked which dependencies the /boot mount actually ships, as I wasn't aware of any side of the parent mounts, which obviously was mounted fine (the root file system).
```
Requires=systemd-fsck@dev-disk-by\x2dpartuuid-907af7d0\x2d01.service
```
It seems that explicitly for FAT file systems, a file system check is mandatory at boot. So wile the above (dedicated power supply and fixing under-voltage warnings) stays true, fixing that file system error might (temporarily) solve your issue. Since the system boots, it should be safe to choose option `1)` in both cases, removing the dirty bit (mandatory) and copying the original boot sector to the backup. ... ah, verified something that I had in mind already: RPi does not need any boot sector, so it is totally irrelevant 😉.

Cjkeenan, 2020-11-07:
Ok so what course of action should I take? My SSD enclosure does not have any external power option. I have not seen any enclosure that has external power, except those for 3.5" HDD drives. Like I said though, even with the Canakit brick, the undervoltage errors go away, but all of the other errors remain, is the damage already done and I need to do some sort of repair?

MichaIng, 2020-11-08:
Did you fix the difference in boot sector as well? In case run `dosfsck -a /dev/sda1` to have fixes applied automatically.
... but ah, actually having another look at your screenshot, the failing fsck is not the issue but another symptom. Waiting for the block device itself times out, hence neither can it be mounted, not can the fsck run.
A simple solution might be to give it some more time to load: Add `boot_delay=5` or even `boot_delay=10` to `/boot/cmdline.txt` to give it some more time to be initialised.

Btw, the other kernel errors can be ignored btw. Those are expected when running the RPi with less than 32M GPU memory, where some GPU features are disabled (cut-down firmware files loaded). I already asked RPi devs if there is a way to prevent the attempt to load the vcsm device tree part as well as the kernel module, with cut-down firmware, since those are doomed to fail anyway and only produce error messages which might confuse or create worries, but it is not foreseen that the loaded firmware files can influence the device tree or loaded kernel modules. I really have to implement my personal workaround for this to disable both manually 😄.

Another thing: You have the serial/UART device disabled (`/dev/ttyS0`), but the serial console on that device enabled, obviously. So you might want to disable it, either via `dietpi-config` > `Advanced Options` > `Serial/UART` or manually via `systemctl disable serial-getty@ttyS0`.

Cjkeenan, 2020-11-08:
> A simple solution might be to give it some more time to load: Add `boot_delay=5` or even `boot_delay=10` to `/boot/cmdline.txt` to give it some more time to be initialised.

Does it matter where in the cmdline.txt this is inserted?

> Another thing: You have the serial/UART device disabled (`/dev/ttyS0`), but the serial console on that device enabled, obviously.

Will do and will report back.

Cjkeenan, 2020-11-08:
Update: No dice on adding a boot_delay=10 to the config.txt and disabling the serial console, same issues. Anything else you guys can think of or do you want to see any more logs? Is there any way to verify if that delay was indeed adhered to?

MichaIng, 2020-11-08:
<details>
<summary>Not really relevant in your case ;)</summary>
As far as I understand, you put a new (empty) file `timeout` onto the first partition of the SD card. The SD card is read/checked anyway, so it could see that file and then wait and see if a bootable USB drive is detected. What I don't understand in this regards is, that AFAIK the RPi boots from the SD card automatically if possible, also the docs state it:
> Make sure that you do not have an SD card inserted in the Pi, since if you do, it will boot from that first.

So then the SD cards must contain a FAT partition, likely on the same location as if an OS was installed, and contain that empty `timeout` file, but the SD card must not contain an OS aside of that, i.e. at least the `bootcode.bin` must not exist.

That docs are a scattered mess, heavily redundant, which makes it tedious to read. Apart of the `timeout` file, it is also possible to program this increased timeout via OTP bit: `program_usb_boot_timeout=1`

But before applying permanent changes to the OTP, I'd rule out everything else. The thing is that the RPi "boots" fine, bootloader, kernel, firmware from boot partition are read and loaded and root partition is mounted (?), it is only the Linux device connector that seems to be created very late somehow. By all I understand this timeout (2 vs 5 second) is only to allow more time for the very first step, the RPi ROM firmware to find a FAT partition with a `bootcode.bin` to start booting from. That stage has long been passed successfully.
</details>

Just to be sure, there is no SD card attached currently, is it?
And did you actually try to mount the `/boot` partition after CTRL+D boot? `mount /boot`
**Ah and I made a mistake above: `boot_delay=5`/`boot_delay=10` must not be added to `cmdline.txt `but to `config.txt`, sorry, I mixed that up. Fixed in above. Please try that first.**

Cjkeenan, 2020-11-08:
Sorry, it appears we are on opposite sleep schedules, but anyway, there is no SD card attached and I did add boot_delay=10 to the config.txt and it did not help.

```
root@BH-3B:/boot# mount /boot                                                                                                                                                                   
mount: /boot: /dev/sda1 already mounted on /boot. 
```
Also is it normal for the green LED be blinking at a 1s interval?

Cjkeenan, 2020-11-12:
Not trying to push, and if you guys are busy I get it, but any other ideas of how to fix or at least troubleshoot?

MichaIng, 2020-11-13:
The green LED beside the constantly lighting red led, right?
It should blink on disk writes, but you could actually check via `dietpi-led_control` what the trigger is and adjust it. So a constant 1s blink should actually not be the case. There are error codes, but AFAIK those only occur when boot fails completely: https://www.raspberrypi.org/documentation/configuration/led_blink_warnings.md

Is the error and solution you face still the same as in OP? Because `/boot` seems to have been successfully mounted according to your post 4 days ago?

Probably a fresh and complete `dmesg` and/or `journalctl` output can help to review everything that happens during boot.

Cjkeenan, 2020-11-13:
> `ietpi-led_control` what the trigger is and adjust it

I just reset all values to default and it has the same behavior, 1s blinks during and after boot

> Probably a fresh and complete `dmesg` and/or `journalctl` output can help to review everything that happens during boot.

[journalctl.txt](https://github.com/MichaIng/DietPi/files/5534386/journalctl.txt)
[dmesg.txt](https://github.com/MichaIng/DietPi/files/5534388/dmesg.txt)

ravenclaw900, 2020-11-13:
Not sure, but 
```text
[    3.995637] usb 1-1.5: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
[    3.995644] usb 1-1.5: required by the UAS driver. Please try an other USB controller if you wish to use UAS.```
looks like the problem to me.  Doing a search, it seems like some people were able to resolve it by adding `max_usb_current=1` to `/boot/config.txt`

Cjkeenan, 2020-11-13:
Issue still present even after including that max current definition in config.txt
New dmesg and journalctl files:
[dmesg2.txt](https://github.com/MichaIng/DietPi/files/5534791/dmesg2.txt)
[journalctl2.txt](https://github.com/MichaIng/DietPi/files/5534792/journalctl2.txt)

MichaIng, 2020-11-13:
`max_usb_current` has not effect anymore, maximum USB current is available on all RPi models by default now.

The issue with `dwc_otg` driver and UAS is known and expected as well.
<details>
<summary>Excursus about the UAS story</summary>
Until RPi4, the default USB driver is `dwc-otg` which supports "FIQ" but no scatter-gather, which is required for UAS. There is an alternative mainstream diver `dwc2`, which supports scatter-gather, hence UAS, but no FIQ. Don't ask me about details and benefits of those technology, but the UAS kernel module was not even available until not very long ago. It has been build now since it includes a list of USB quicks which makes booting/mounting (don't remember the exact issue) of a very rare/special USB SATA bridge case possible, but actually real UAS was/is still not possible. I don't remember the details but it was definitely not a regular SATA drive attached via regular USB case/docking station, only reported by a single user. So now the UAS module is there, the kernel attempts to load it, which is known to be not possible with the default USB driver, UNTIL RPi 4.

The RPi 4 now loads the `dwc2` by default, which enables UAS support. Not sure why scather-gather+UAS is now seen more important than FIQ only on this model, the only thing I see now is regular reports with RPi 4 where booting or even mounting USB drive fails, since there are lots of drives (respectively contained firmware versions) available which do not support UAS and then fail. For this another USB quick needs to be added to blacklist UAS for that USB device. So to me it looks like actually enabling UAS now introduced a similar issue (from symptoms) for a large amount of users than what adding the kernel module in the first place was aimed to resolve for a single user only.

But UAS can be enabled on all RPi models by switching the USB driver: `echo 'dtoverlay=dwc2' >> /boot/config.txt`
And UAS can be disabled for all USB drives: `echo -e 'blacklist uas\nblacklist sg' > /etc/modprobe.d/disable_uas.conf`
The last one does not mute the error messages, but it saves ~50 KiB kernel memory space, for free on RPi <4 since they have no effect anyway.

Actually I wanted to ask the RPi devs about this for a long time, what the actual benefits of UAS are, if this really outweighs the issue that several users run into a failing USB drives without manual action. If there are benefits, why it is only enabled on RPi 4 by default, if the required `dwc2` driver could be a reasonable default on all RPi models, and otherwise if there are any downsides of blacklisting the uas and sg kernel modules (there are none, I'm doing this for a year on my RPi). Currently it looks wrong to me but I want to get some official info/opinions on this before applying an enhancement on DietPi 🙂.
</details>

Okay so we see the kernel waiting for the root device:
```
[    1.728302] Waiting for root device PARTUUID=907af7d0-02
```
The whole drive is quickly found, attached, all partitions seen correctly, fsck done on root (no dirty bit, hence continues immediately), then init system (systemd) is loaded:
```
[    5.100966]  sda: sda1 sda2
[    5.102845] sd 0:0:0:0: [sda] Attached SCSI disk
[    5.586670] F2FS-fs (sda2): recover fsync data on readonly fs
[    5.587276] F2FS-fs (sda2): Mounted with checkpoint version = 4e672421
[    5.587348] VFS: Mounted root (f2fs filesystem) readonly on device 8:2.
[    5.588788] devtmpfs: mounted
[    5.592356] Freeing unused kernel memory: 1024K
[    5.623938] Run /sbin/init as init process
```
And then we see the boot file system device not appearing before it times out after 90 seconds, which finally loads to emergency mode:
```
Nov 12 20:44:34 BH-3B systemd[1]: dev-disk-by\x2dpartuuid-907af7d0\x2d01.device: Job dev-disk-by\x2dpartuuid-907af7d0\x2d01.device/start timed out.
Nov 12 20:44:34 BH-3B systemd[1]: Timed out waiting for device /dev/disk/by-partuuid/907af7d0-01.
Nov 12 20:44:34 BH-3B systemd[1]: Dependency failed for File System Check on /dev/disk/by-partuuid/907af7d0-01.
Nov 12 20:44:34 BH-3B systemd[1]: Dependency failed for /boot.
Nov 12 20:44:34 BH-3B systemd[1]: Dependency failed for Local File Systems.
Nov 12 20:44:34 BH-3B systemd[1]: local-fs.target: Job local-fs.target/start failed with result 'dependency'.
Nov 12 20:44:34 BH-3B systemd[1]: local-fs.target: Triggering OnFailure= dependencies.
Nov 12 20:44:34 BH-3B systemd[1]: boot.mount: Job boot.mount/start failed with result 'dependency'.
Nov 12 20:44:34 BH-3B systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-907af7d0\x2d01.service: Job systemd-fsck@dev-disk-by\x2dpartuuid-907af7d0\x2d01.service/start failed with result 'dependency'.
Nov 12 20:44:34 BH-3B systemd[1]: dev-disk-by\x2dpartuuid-907af7d0\x2d01.device: Job dev-disk-by\x2dpartuuid-907af7d0\x2d01.device/start failed with result 'timeout'.
```
Another fsck is done on the root fs (not the affected boot fs). I was wondering first about this:
```
Nov 12 20:44:45 BH-3B systemd-fsck[122]: Info: MKFS version
Nov 12 20:44:45 BH-3B systemd-fsck[122]:   "Linux version 5.8.0-1-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.0-6) 10.2.0, GNU ld (GNU Binutils for Debian) 2.35) #1 SMP Debian 5.8.7-1 (2020-09-05)"
```
But I think this just tells us about the host system which created the F2FS file system and this of course does not need to match the RPi.
[... cut ...] t can have an effect, especially when boot partition is initially read fine (kernel, bootloader etc) and root partition mounts fine etc, and especially since it worked perfectly fine before, but just to rule it out.
- External power supply: Probably the new kernel 5.4 changed the way how power is drawn across the device. I know a dedicated power supply for a 2.5" case is quite uncommon, but for SBCs actually reasonable. Otherwise a 3.5" case/docking station usually can take a 2.5" drive as well.

Cjkeenan, 2020-11-13:
Thank you so much for the detailed response, I will continue updating this comment as I go through the steps.

> From which DIetPi version were you upgrading, respectively, do you remember if it was still Linux 4.19?

The initial build was the F2FS image you made here: https://github.com/MichaIng/DietPi/issues/606#issuecomment-695821494

Cjkeenan, 2020-11-26:
I am very thankful for all of the troubleshooting steps you guys have provided but at this point should I just backup dietpi onto a flash drive, then reflash the F2FS image and then restore from backup or will that likely restore the bug as well? Like I said, while running everything seems fine, just in the event of a power outage, and I am not home, the RPi will not boot on its own and that is not great.

Joulinar, 2020-11-26:
Usually a `dietpi-backup` is a full backup. Probably the issue will be backed up as well.

Cjkeenan, 2021-01-05:
So when I was looking into tips for making the RPi more reliable, I found this article and this specific tip. I wonder, do you think this is what is going on here, something with my clock is screwed up and it isn't allowing boot?

https://blog.ronnyvdb.net/2019/01/20/howto-make-a-raspberry-pi-truly-read-only-reliable-and-trouble-free/

step 4.4 and 4.5

Joulinar, 2021-01-05:
usually system time is synchronized during boot. You can check it using `systemctl status systemd-timesyncd.service`

MichaIng, 2021-01-05:
That guide is btw pretty outdated, as it handles things as there were no systemd. Most things do not apply on DietPi or are for the specific use case of the author (SSH tunnel) or standard/implemented in DietPi already.

Use `dietpi-config` > `Advanced Options` > `Time sync mode` > `4 : Daemon + Drift` to have `systemd-timesyncd` enabled permanent, like NTP or htpdate daemons would be. It starts slightly earlier on boot and of course keeps the accuracy higher compared to running it on hourly/daily basis, but otherwise there will be no significant difference. The basic problem that system clock is out of sync until the first network sync has been done, remains of course with a read-only file system.

Btw, instead of dealing with all the error messages when mounting the rootfs read-only, mounting an overlayfs that emulates R/W might be better. See here the instructions by @Joulinar: https://dietpi.com/phpbb/viewtopic.php?p=24920#p24920

Cjkeenan, 2021-01-09:
No dice on it being the Time Sync Mode. For reference, I did not follow that guide, but rather I was just looking around for similar symptoms and came across that. Do you guys have any inclination as to what is going on? It seems rather odd to me that it times out waiting for the device, and then after forcing it via CTRL+D, it magically mounts and boots fine no?

MichaIng, 2021-01-10:
Can you please show your `cmdline.txt`: `cat /boot/cmdline.txt`
It has the `rootfstype` boot parameter, if that is set correctly to "f2fs", probably removing that option helps.

Ah, and can you show `lsmod`, to see if F2FS is built into the kernel or loaded as module. In the letter case, it would make sense that it cannot be mounted at boot without initramfs (default on RPi). Would be strange then that it works after CTRL+D, but let's simply rule it out 😉.

Cjkeenan, 2021-01-11:
```
dietpi@BH-3B:~$ cat /boot/cmdline.txt                                                                                                                                                           
 console=tty1 root=PARTUUID=907af7d0-02 rootfstype=f2fs elevator=deadline fsck.repair=yes rootwait net.ifnames=0     
 ```
 
```
dietpi@BH-3B:~$ lsmod                                                                                                                                                                           
Module                  Size  Used by                                                                                                                                                           
xt_nat                 16384  0                                                                                                                                                                 
xt_tcpudp              16384  0                                                                                                                                                                 
veth                   28672  0                                                                                                                                                                 
xt_conntrack           16384  2                                                                                                                                                                 
nf_conntrack_netlink    45056  0                                                                                                                                                                
xt_addrtype            16384  2                                                                                                                                                                 
br_netfilter           28672  0                                                                                                                                                                 
bridge                139264  1 br_netfilter                                                                                                                                                    
stp                    16384  1 bridge                                                                                                                                                          
llc                    16384  2 bridge,stp                                                                                                                                                      
wireguard             131072  0                                                                                                                                                                 
ip6_udp_tunnel         16384  1 wireguard                                                                                                                                                       
udp_tunnel             16384  1 wireguard                                                                                                                                                       
overlay               102400  0                                                                                                                                                                 
sg                     28672  0                                                                                                                                                                 
joydev                 20480  0                                                                                                                                                                 
uas                    24576  0                                                                                                                                                                 
bcm2835_isp            32768  0                                                                                                                                                                 
bcm2835_mmal_vchiq     28672  1 bcm2835_isp                                                                                                                                                     
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq                                                                                                                                  
videobuf2_dma_contig    20480  1 bcm2835_isp                                                                                                                                                    
videobuf2_memops       16384  1 videobuf2_dma_contig                                                                                                                                            
videobuf2_v4l2         28672  1 bcm2835_isp                                                                                                                                                     
videobuf2_common       57344  2 bcm2835_isp,videobuf2_v4l2                                                                                                                                      
[... cut ...]
nfnetlink              16384  4 nft_compat,nf_conntrack_netlink,nf_tables                                                                                                                       
ip_tables              28672  0                                                                                                                                                                 
x_tables               32768  8 xt_comment,xt_nat,ip_tables,nft_compat,xt_tcpudp,xt_MASQUERADE,xt_addrtype,xt_conntrack                                                                         
ipv6                  458752  40 br_netfilter,wireguard,bridge
```

MichaIng, 2021-01-11:
Okay, also here I see it being as kernel builtin:
```
2021-01-11 16:39:53 root@micha:~# modinfo f2fs
name:           f2fs
filename:       (builtin)
license:        GPL
file:           fs/f2fs/f2fs
description:    Flash Friendly File System
author:         Samsung Electronics's Praesto Team
alias:          fs-f2fs
```
This means no initramfs required. Okay I'm out of ideas what the reason could be, but it most likely has to do with F2FS. I'm not finding time to test myself soon, probably search machines can help to check for similar issues or requirements when booting an RPi from F2FS.

Cjkeenan, 2021-01-28:
So since this issue is still present and cannot be easily fixed it appears, what is the best method for installing a fresh copy of a f2fs install? Should I install raspbian as f2fs and then do the script to make my own distribution or is there an easier way?

MichaIng, 2021-01-28:
Since there is no F2FS image for Raspbian/Raspberry Pi OS (or is it?), you could install DietPi as F2FS just as you would do with a fresh Raspberry Pi OS image, e.g.:
- Flash a fresh DietPi image onto a new SD card and mount it to another Linux system.
- Copy the content of the 2nd partition somewhere else.
- Delete and re-create the second partition with full size, and format it with F2FS file system.
- Copy the content back in.
- Edit `cmdline.txt` on the 1st partition and `/etc/fstab` on the second partition so that file system type and PARTUUID of the root partition match our new 2nd partition.

That should be it. Since the DietPi fs_partition_resize script does not support F2FS yet, it'll fail, but no issue since partition and file system is extended already.

Cjkeenan, 2021-01-28:
> Since there is no F2FS image for Raspbian/Raspberry Pi OS (or is it?)

Officially? No, but I found this utility (https://github.com/FrangaL/rpi-img-builderwhich) has a bunch of options to pre-setup an image including selecting a filesystem, so my plan was to build a lite Raspberry pi OS image with that then upgrade to dietpi via your script.

MichaIng, 2021-01-31:
Ah yes, that'll work, the script looks good 👍.

> I wonder if that developer would be interested/able to provide dietpi as another option for OS, which would be very cool.

The script installs a very basic Debian or Raspbian system, which moreless matches with the DietPi base system (repository, packages, even most of the applied system configs match to how we ship it). But a DietPi image has configs applied via our scripts and first boot services enabled in a way that at first boot `dietpi.txt` choices are applied, `dietpi-config` (and other tools) detect and apply configs correctly etc. It would be not reasonable to implement and especially maintain this in a separate script when one can simply apply it via our maintained script in a second step: https://github.com/MichaIng/DietPi/blob/dev/PREP_SYSTEM_FOR_DIETPI.sh
Most of the code lines are pre-image specific cleanup and board-specific setup things, so only a minor part would need to be ported when using a debootstraped Raspbian, but already for downloading and installing the `dietpi-*` scripts and enabling the first boot services it wouldn't make sense to double the coding effort, IMO.

DietPi simply is not only a minimal image. Such a debootstraped Raspbian (when choosing "slim" in the script) is the naked minimal image, which in turn excludes the DietPi tools to take users by the hand at first boot and for software installs 😉.

MichaIng, 2021-03-25:
I mark the issue as closed. Feel free to reopen if required..

closed 2021-03-25 by MichaIng

aakerbeere, 2021-07-28:
> At this point I am beginning to think I may just upgrade to a RPi 4 to take advantage of the USB3. Would this actually help anything you think?

My RPi4 is suffering from the same issue ([see here](https://github.com/raspberrypi/linux/issues/3108#issuecomment-887484610)).
Actually i'm back on ext4 for the "root"- Partition, but would prefere f2fs.

cheers

MichaIng, 2021-07-28:
Can you please share in detail all steps you did to migrate that DietPi system onto the F2FS filesystem?

aakerbeere, 2021-07-28:
My equipement is not DietPi but Pi4.
It did boot from USB Samsung Portable SSD T5 having f2fs for "/root"-partition and it did not boot anymore after the nextcloud instance having done syncing some additional data.

Plugged the drive to notebook, rsynced the data, gparted the partition to ext4, rsynced data back, edited fstab and cmdline.txt and on it goes.

Would do exactly the same vice versa (and have done it before) to revert running system to f2fs.

MichaIng, 2021-07-28:
> My equipement is not DietPi but Pi4.

What do you mean by this? DietPi is an operating system, Pi4 (= Raspberry Pi 4) is an SBC. If you do not run DietPi on your Pi, please check for support on the operating system you use 😉.

aakerbeere, 2021-07-28:
Sry
I didn't know DietPi. I only wanted to indicate to Cjkeenan he could get the same issue with a Pi4, because he was asking if it would bee an idea to upgrade the HW to Pi4.
I'm running RPiOS 64bit Buster and get the same issue as he did/does? with DietPi.

MichaIng, 2021-07-28:
Well, we already tested DietPi successfully on F2FS, so it's at least nothing with the kernel, firmware or bootloader parts. I guess it is something with the Raspberry Pi OS specific setup scripts then.