MichaIng/DietPi #3895: DietPi-Software | TOR Relay

ravenclaw900, 2020-11-07:
### Creating a software request

#### Formal software information
- Software title | 
TOR Relay
- Short description | 
Help contribute to the TOR network
- Official URL (if available) | 
https://torproject.org
- GitHub URL (if available) | 
https://github.com/torproject/tor?
- (Official) install documentation | 
https://community.torproject.org/relay/setup/

#### Are there similar/alternative software titles available with DietPi-Software?

- No

#### What makes your requested software better than the above solutions, if available?

- N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- 3 different types of possible relay, could automate setup for them all

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Install `tor` package
2. Configure according to user-provided settings

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

MichaIng, 2020-11-07:
Many thanks for your request.

Note that we have the Tor Hotspot option. But it redirects only requests incoming from the WiFi Hotspot through Tor. Maybe this can be widened to turn the current Tor Hotspot option into a general Tor option that can be optionally configured with the WiFi Hotspot, similarly to how unbound can be optionally configured to work with Pi-hole?

See the code, probably this is already moreless one of the 3 types you have in mind? https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L9994-L10030

ravenclaw900, 2020-11-07:
Not really, I was more thinking of this (https://community.torproject.org/relay/setup/guard/debianubuntu/):
```text
Nickname myNiceRelay
ORPort 443
ExitRelay 0
SocksPort 0
ControlSocket 0
ContactInfo tor-operator@your-emailaddress-domain
```
I think (not sure) that running a relay doesn't actually provide the device with TOR connectivity.  If so, then those two software would actually conflict with each other.

MichaIng, 2020-11-07:
Ahh, now I see what a relay means, it's to contribute another node to the Tor project. Hmm would be interesting to know if it's possible to run a relay aside of an AP. Theoretically, it could be possible since different ports are used and the forwarding rules for the AP only affects requests incoming through the WiFi adapter.

The same config file is used, so it it's either/or or appending to the same config file. Needs testing.

ravenclaw900, 2020-11-09:
It seems possible, the config file does have a `ClientOnly` option, implying that you can run both.  Since you just need different ports, it should be possible.  I'll test it.

ravenclaw900, 2020-12-18:
Strangely, Tor Metrics pegs my relay as being down, even though it is up.  Obviously, this isn't actually the case, as it also says that people are using it.  This only started after I started using IPv6, even though we are using the correct implementation (https://community.torproject.org/relay/setup/post-install/#7.-check-ipv6-availability).

MichaIng, 2020-12-18:
Have you assured that you can ping your server from outside your local network? I'm still wondering how many users have a static public IPv6 address 🤔. Probably we should make it more clear that hence IPv6 can only work with this condition.

Btw, to assure that IPv6 works fine, shouldn't we do the suggested test, pinging the Tor directory authorities via IPv6?
```
ping6 -c2 2001:858:2:2:aabb:0:563b:1526 && ping6 -c2 2620:13:4000:6000::1000:118 && ping6 -c2 2001:67c:289c::9 && ping6 -c2 2001:678:558:1000::244 && ping6 -c2 2607:8500:154::3 && ping6 -c2 2001:638:a000:4140::ffff:189 && echo OK.
```

Also, we could feed the input box with an auto-estimated public IPv6 address. In theory it is possible that a server can be reached via separate IPv6 addresses or it has currently a VPN or other alternative route active, so allowing to change via inputbox is still required:
```
curl -6 https://ifconfig.co/ip
# or
curl -6 https://dietpi.com/myip.php
```

ravenclaw900, 2020-12-18:
Yes, mine is reachable from the outside. `Dec 12 13:43:02 rpi4 Tor[454]: Self-testing indicates your ORPort is reachable from the outside. Excellent. Publishing server descriptor.`  We're actually filling the inputbox with the private IPv6 address (using the public gives errors, and is also not what is recommended), but we probably should ping the directory authorities as well.  That probably also means that you don't even need to have a static IPv6 address.

MichaIng, 2020-12-18:
Ahh, I misunderstood the entry. It is the private IPv6 address that Tor binds to, not the public one (as long as both do not match). Totally makes sense.

Okay, then only the Tor directory authorities pinging could be reasonable to assure IPv6 access works. But a minor thing then, as if it wouldn't, having IPv6 enabled would have caused connection issues before.

closed 2020-12-20 by MichaIng