MichaIng/DietPi #3897: DietPi-Software | Unattended Upgrades

ravenclaw900, 2020-11-07:
### Creating a software request

#### Formal software information
- Software title | 
Unattended Upgrades
- Short description | 
Automatically update packages
- Official URL (if available) | 
N/A
- GitHub URL (if available) | 
https://github.com/mvo5/unattended-upgrades?
- (Official) install documentation | 
N/A, usually configured by packages that recqire it.

#### Are there similar/alternative software titles available with DietPi-Software?

- No

#### What makes your requested software better than the above solutions, if available?

- N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Adding as Linux Software, configure by packages that use it

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Install package
2. Configure according to packages that use it

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

closed 2020-11-07 by ravenclaw900

inspector71, 2021-08-02:
Any chance we could know why this issue was closed?

I ask because I want to harden my DietPi but do not want to mess around with the DietPi efficacy. 

Realise DietPi uses `APT`, and / or `DPKG` wherever possible. 

Does manually installing `unattended-upgrades` interfere with the DietPi scripting?

Possibly helpful:

https://wiki.debian.org/UnattendedUpgrades
https://linuxiac.com/how-to-set-up-automatic-updates-on-debian/
https://packages.debian.org/search?keywords=unattended-upgrades

FWIW, not questioning the validity of any decisions regarding this issue. Just hoping for a reference. If there's an answer applied here, and this issue is indexed, users might get their answer in one spot.

ravenclaw900, 2021-08-02:
See https://github.com/MichaIng/DietPi/pull/3898#issuecomment-723475993 for why this issue was closed.  As a replacement, we implemented automatic APT upgrades as part of DietPi.  You can enable them with `G_CONFIG_INJECT 'CONFIG_CHECK_APT_UPDATES=' 'CONFIG_CHECK_APT_UPDATES=2' /boot/dietpi.txt`.

Joulinar, 2021-08-02:
Just as a side node. This will update apt packages only. DietPi scripts would need to be updated separately 😉

MichaIng, 2021-08-02:
Implemented in: #3899

inspector71, 2021-08-03:
Thanks for the rapid updates everyone. Brilliant. Uninstalled unattended-upgrades now I know it's taken care of by DietPi. 

Having just put a dietpi box online with SSH open, I was trying to think of anything I could do to beef up security.

G_CONFIG_INJECT 'CONFIG_CHECK_APT_UPDATES=' 'CONFIG_CHECK_APT_UPDATES=2' /boot/dietpi.txt

Done :)

Also, yep, I realise apps like Radarr do not have an APT repo, unlike Sonarr for example. They both have self updating though :)

MichaIng, 2021-08-03:
Just for completeness, the `unattended-upgrades` package an be used as well, but the related systemd timers need to be enabled:
```sh
G_CONFIG_INJECT 'CONFIG_CHECK_APT_UPDATES=' 'CONFIG_CHECK_APT_UPDATES=0' /boot/dietpi.txt
systemctl unmask apt-daily
systemctl unmask apt-daily.timer
systemctl unmask apt-daily-upgrade
systemctl unmask apt-daily-upgrade.timer
sytemctl enable --now apt-daily.timer
sytemctl enable --now apt-daily-upgrade.timer
apt install unattended-upgrades
```
But it is a relatively heavy solution, which implies a lot of magic, so outcome needs to be monitored to assure it really does what it is supposed to on non-Debian APT repositories (Raspbian in case of RPi).

inspector71, 2021-08-04:
> 
> 
> Just for completeness, the `unattended-upgrades` package an be used as well, but the related systemd timers need to be enabled:
> 
> ```shell
> G_CONFIG_INJECT 'CONFIG_CHECK_APT_UPDATES=' 'CONFIG_CHECK_APT_UPDATES=0' /boot/dietpi.txt
> systemctl unmask apt-daily
> systemctl unmask apt-daily.timer
> systemctl unmask apt-daily-upgrade
> systemctl unmask apt-daily-upgrade.timer
> sytemctl enable --now apt-daily.timer
> sytemctl enable --now apt-daily-upgrade.timer
> apt install unattended-upgrades
> ```
> 
> But it is a relatively heavy solution, which implies a lot of magic, so outcome needs to be monitored to assure it really does what it is supposed to on non-Debian APT repositories (Raspbian in case of RPi).

Wow, that it is complete! Exhaustive I'd say :)

Joulinar, 2021-08-04:
At least in past it was needed to adjust configuration to be able to work on Raspbian OS 32bit. If I recall correctly it was the repository name. 

The DietPi script is way easier to use compare to `unattended-upgrades`

MichaIng, 2021-08-04:
If I remember correctly it has to do with the dedicated `security.debian.org` repository used by Debian, which does not exist on Raspbian (all is merged directly into the main repo). And `unattended-upgrades` by default only installs upgrades from there, or so. While it sounds reasonable in the first place, actually on a stable Debian release, all package updates are either security-related or fix bugs, as of strict rules Debian has. And it is assured that no breaking chances are involved, so compatibility with existing config files and other software is guaranteed. The only questionable case are 3rd party repositories, where also breaking changes may be shipped with a package upgrade.