MichaIng/DietPi #3901: MineOS not working

CactiChameleon9, 2020-11-08:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | 10.6
- Kernel version | Linux DietPiServer 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux
- SBC model | RPi4
- Power supply used | Idk - some random one. It works under load (stress tested)
- SDcard used | Sandisk ultra, with portable hdd rootfs

#### Additional Information (if applicable)
- Software title | MineOS
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Not sure

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. Install MineOS (may be aarch64 issue - not sure)

#### Expected behaviour
It should all work

#### Actual behaviour
None of the menus seem to work and are blank:
[image]

#### Extra details

- Not sure, but possibly an aarch64 issue. It may be the installer is broken also

Joulinar, 2020-11-08:
same happen on 32bit image.

<details>
  <summary>Logfile attached. Click to expand!</summary>

```
Cloning into 'minecraft'...
remote: Enumerating objects: 24, done.
remote: Counting objects: 100% (24/24), done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 5980 (delta 9), reused 6 (delta 3), pack-reused 5956
Receiving objects: 100% (5980/5980), 5.12 MiB | 1.06 MiB/s, done.
Resolving deltas: 100% (3522/3522), done.
[  OK  ] DietPi-Software | cd minecraft
[  OK  ] DietPi-Software | git config core.filemode false

added 1 package, and audited 1 package in 3s

found 0 vulnerabilities
npm notice
npm notice New patch version of npm available! 7.0.8 -> 7.0.9
npm notice Changelog: https://github.com/npm/cli/releases/tag/v7.0.9
npm notice Run npm install -g npm@7.0.9 to update!
npm notice

  Error: version required

npm ERR! code 1
npm ERR! path /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! make: Entering directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid/build'
npm ERR!   CXX(target) Release/obj.target/userid/src/userid.o
npm ERR! make: Leaving directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@15.1.0 | linux | arm
npm ERR! gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
npm ERR! gyp http GET https://nodejs.org/download/release/v15.1.0/node-v15.1.0-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v15.1.0/node-v15.1.0-headers.tar.gz
npm ERR! gyp http GET https://nodejs.org/download/release/v15.1.0/SHASUMS256.txt
npm ERR! gyp http 200 https://nodejs.org/download/release/v15.1.0/SHASUMS256.txt
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/root/.cache/node-gyp/15.1.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/15.1.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/15.1.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/userid.cc:11:
[... 63 lines cut ...]
npm ERR!        (node::addon_register_func) (regfunc),                          \
npm ERR!                                            ^
npm ERR! /root/.cache/node-gyp/15.1.0/include/node/node.h:869:3: note: in expansion of macro ‘NODE_MODULE_X’
npm ERR!    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
npm ERR!    ^~~~~~~~~~~~~
npm ERR! ../src/userid.cc:174:1: note: in expansion of macro ‘NODE_MODULE’
npm ERR!  NODE_MODULE(userid, Init);
npm ERR!  ^~~~~~~~~~~
npm ERR! make: *** [userid.target.mk:109: Release/obj.target/userid/src/userid.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:327:20)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:277:12)
npm ERR! gyp ERR! System Linux 5.4.72-v7+
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
npm ERR! gyp ERR! node -v v15.1.0
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-08T19_11_06_421Z-debug.log
```
</details>

MichaIng, 2020-11-08:
The first failure happens when trying to install Node 8, AFAIK, and since MinoOS (still) requires Node 8, it's build fails.
Could you try to run the command manually:
```
n -a arm64 8
```
If this fails:
```
n 8
```
Explicitly defining arm64 architecture was required due to an "n" bug: https://github.com/MichaIng/DietPi/issues/1880#issuecomment-464097174
Not sure if it's still required. But it seems that the "8" is not recognised as version as your error messages state that either a version is missing or "-a arm64" is taken as version instead of as option 🤔. Probably the syntax has changed.

Joulinar, 2020-11-08:
according MineOS wiki, it should work with Node10

https://minecraft.codeemo.com/mineoswiki/index.php?title=MineOS-node_(apt-get)_(Debian10)

MichaIng, 2020-11-08:
Dammit, the quotation needs to be removed!
`n "$arch" 8` => `n $arch 8`

Joulinar, 2020-11-08:
just for my understanding. Variable `n` is set after installation of `node`?

```
			# Install Node 8, as MineOS is currently not compatible with Node 10: https://github.com/MichaIng/DietPi/issues/1880
			npm i -g --unsafe-perm n
			# - On ARMv8 explicitly add arch info, since it is not recognized by "n": https://github.com/MichaIng/DietPi/issues/1880#issuecomment-464097174
			local arch=
			(( $G_HW_ARCH == 3 )) && arch='-a arm64'
			n "$arch" 8
```

MichaIng, 2020-11-08:
It's not a variable but a Node version management module, installed above with: `npm i -g --unsafe-perm n` (install globally module "n")

It seems to work even with Node 11 now.
`--unsafe-perm` seems to be not required anymore to allow running npm as root.
And why the hack do we launch it via supervisor instead of a systemd unit? Just tested the one the provided and aside of node and binary paths it works pretty well.

@CactiChameleon9 
Could you test the following:
```
n 11
cd /mnt/dietpi_userdata/mineos/minecraft
npm i
systemctl restart supervisor
```

Joulinar, 2020-11-08:
> It's not a variable but a Node version management module, installed above with: npm i -g --unsafe-perm n (install globally module "n")

thx for explanation

CactiChameleon9, 2020-11-09:
doesn't seem to be working
```
dietpi@DietPiServer:/mnt/dietpi_userdata/mineos/minecraft$ sudo npm i

> posix@4.2.0 install /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/11.15.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/.node-gyp'
gyp ERR! System Linux 5.4.72-v8+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! posix@4.2.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! !     /root/.npm/_logs/2020-11-09T08_02_36_339Z-debug.log
npm ERR! Failed at the posix@4.2.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-09T08_02_36_339Z-debug.log
```
The file is here:
```
dietpi@DietPiServer:/mnt/dietpi_userdata/mineos/minecraft$ sudo cat /root/.npm/_logs/2020-11-09T08_02_36_339Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'i' ]
2 info using npm@6.7.0
3 info using node@v11.15.0
4 verbose npm-session c0500770665c4b74
5 silly install runPreinstallTopLevelLifecycles
6 silly preinstall mineos-node@1.1.1
7 info lifecycle mineos-node@1.1.1~preinstall: mineos-node@1.1.1
8 silly install loadCurrentTree
9 silly install readLocalPackageData
10 timing stage:loadCurrentTree Completed in 3329ms
11 silly install loadIdealTree
12 silly install cloneCurrentTreeToIdealTree
13 timing stage:loadIdealTree:cloneCurrentTree Completed in 22ms
14 silly install loadShrinkwrap
15 timing stage:loadIdealTree:loadShrinkwrap Completed in 1863ms
16 silly install loadAllDepsIntoIdealTree
17 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 1921ms
18 timing stage:loadIdealTree Completed in 4561ms
19 silly currentTree mineos-node@1.1.1
19 silly currentTree ├── @babel/code-frame@7.5.5
19 silly currentTree ├── @babel/generator@7.6.2
19 silly currentTree ├── @babel/helper-function-name@7.1.0
19 silly currentTree ├── @babel/helper-get-function-arity@7.0.0
19 silly currentTree ├── @babel/helper-split-export-declaration@7.4.4
19 silly currentTree ├── @babel/highlight@7.5.0
19 silly currentTree ├── @babel/parser@7.6.2
19 silly currentTree ├── @babel/template@7.6.0
19 silly currentTree ├─┬ @babel/traverse@7.6.2
19 silly currentTree │ ├── debug@4.1.1
19 silly currentTree │ └── ms@2.1.2
19 silly currentTree ├── @babel/types@7.6.1
19 silly currentTree ├── @types/node@8.10.54
19 silly currentTree ├── abbrev@1.1.1
19 silly currentTree ├── accepts@1.3.3
19 silly currentTree ├── address@1.1.2
19 silly currentTree ├── adm-zip@0.4.11
[... 1781 lines cut ...]
79 silly saveTree └─┬ xml2js@0.4.19
79 silly saveTree   ├── sax@1.2.4
79 silly saveTree   └── xmlbuilder@9.0.7
80 verbose stack Error: posix@4.2.0 install: `node-gyp rebuild`
80 verbose stack Exit status 1
80 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
80 verbose stack     at EventEmitter.emit (events.js:193:13)
80 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
80 verbose stack     at ChildProcess.emit (events.js:193:13)
80 verbose stack     at maybeClose (internal/child_process.js:999:16)
80 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:266:5)
81 verbose pkgid posix@4.2.0
82 verbose cwd /mnt/dietpi_userdata/mineos/minecraft
83 verbose Linux 5.4.72-v8+
84 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "i"
85 verbose node v11.15.0
86 verbose npm  v6.7.0
87 error code ELIFECYCLE
88 error errno 1
89 error posix@4.2.0 install: `node-gyp rebuild`
89 error Exit status 1
90 error Failed at the posix@4.2.0 install script.
90 error This is probably not a problem with npm. There is likely additional logging output above.
91 verbose exit [ 1, true ]
```

Joulinar, 2020-11-09:
can you try to have it running as user `root` as you have permission issues.

`gyp ERR! stack Error: EACCES: permission denied, mkdir`

CactiChameleon9, 2020-11-09:
Yeah, I tried as root. I will test again quickly now but I tried as `sudo -s` first

CactiChameleon9, 2020-11-09:
Yeah, those permissions denied are because parent directories are not existing. /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/ They seem to get deleted if it is created by me after running `npm i`

MichaIng, 2020-11-09:
Please try
```
cd /mnt/dietpi_userdata/mineos/minecraft
npm i --unsafe-perm
```
In my case the flag was not required but all guides actually suggest to use it when running npm as root.

CactiChameleon9, 2020-11-09:
That seems to work. Thanks a lot. Is sudo -s different then?  :thinking: Maybe just me - thanks for the help

CactiChameleon9, 2020-11-09:
Not sure if it helped much:
[image]

CactiChameleon9, 2020-11-09:
I am trying with n 10

MichaIng, 2020-11-09:
Wait for a while, then the strings might appear. But indeed parts of the web UI also did not work on my tests first. I then switched to a foreground systemd unit and suddenly it worked, including a few gone 403 errors in browser console, although not sure which change exactly made it.

If you are in mood, you could test what I am currently working on: https://github.com/MichaIng/DietPi/pull/3904
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=mineos' /boot/dietpi.txt
dietpi-update 1
dietpi-software reinstall 53
```
Works well so far, cleans up a few obsolete dependencies and things that came with older implementations.

MichaIng, 2020-11-09:
Wow the more I play with it, the more I like it, easy access to all other Minecraft servers we offer: PaperMC, Nukkit, Spigot, Cuberite and others. Config file edit, live logging file monitoring combined with input console for each server, aside of usual status/start/stop etc. In a few cases it fails to download, e.g. Cuberite and the BuildTools (to build all of'em directly from sources instead) successfully, with 404 or 403, not sure if the links need to be updated, but overall very nice impression. Needed to learn that the service only handles the web interface while each Minecraft server runs in an own screen session and needs to be manually started/stopped either from web UI or via `mineos` command for console access.

The project could get some polish here and there, updates to certain Node modules, docs etc, but hey, it's the same with us 😄.

CactiChameleon9, 2020-11-10:
Yeah, it is really cool. I am exited to have a play with it as well, it will allow for such easy management of servers, I can then also proxy it to add to nginx with my long list of services (dietpi is really cool too)

I will try that out when I get the change, the git mineOS branch only includes mineOS changes correct? Thanks for looking into this.

MichaIng, 2020-11-10:
> the git mineOS branch only includes mineOS changes correct?

Yes, it's the current dev branch + MineOS changes. While it's good to have a backup, it should be fine since v6.34 so far mostly contains fixes and individual software title enhancements, no breaking system changes.

CactiChameleon9, 2020-11-10:
OK thanks. Mine OS now works, it took me a couple of times to guess the password and username though, it turns out to be the root password (dietpi) with username root. A small issue, there are no runtime jar options available? I have JRE and JDK headless installed however?

MichaIng, 2020-11-10:
> it took me a couple of times to guess the password and username though

It's explained in our online docs, yes it basically allows to login with any UNIX user and related UNIX password. For initial setup, as well the official docs state that root permissions are required, but it is advised then to use an unprivileged user to install actual Minecraft servers. What I am not sure about is if a UNIX user that you login into the web UI with, requires write access to `/mnt/dietpi_userdata/mineos/serverdata` or not. I mean the web UI process still runs as root 🤔. We could do some tests with this and in case check create a "mineos" system group with write permissions to add users to.

CactiChameleon9, 2020-11-10:
Any clue about no java not showing as installed issue? Also I am experimenting with puffin panel (v2 alpha) at the moment, which does a similar thing

MichaIng, 2020-11-10:
Headless Java JRE + JDK are installed as dependency with MineOS, but indeed not the GUI parts, which is intended to keep it slim. Where is it not shown as as installed? MineOS web UI? I was able to set memory allocation but didn't test to add further custom jar options.

Generally to install the the GUI parts for JRE: `apt install default-jre`
Check out how the packages are divided: https://packages.debian.org/default-jre

CactiChameleon9, 2020-11-10:
Yeah, it shows as no JRE in the webUI, I will send a screenshot in a bit after I try a reinstall.

CactiChameleon9, 2020-11-11:
[image]
I definitely have java installed, because jar --version

ravenclaw900, 2020-11-11:
It works for me. Did you already create and set the profile?

CactiChameleon9, 2020-11-11:
Yeah I did. Thanks for asking. Could it be 64 bit trouble?

ravenclaw900, 2020-11-11:
I'm using a VM.  I'll test it on my RPi too.

CactiChameleon9, 2020-11-11:
OK thanks

MichaIng, 2020-11-11:
Before being able to select a jar, you need to select a profile first at the left side, ah that is what **ravenclaw900** already said.

But also I recognised that some Minecraft downloads contain a tarball (or other kind of file) only without any jar file and I didn't understand how this is supposed to work. To be true I didn't try to manually extract those.

The files are located in `/mnt/dietpi_userdata/mineos/serverdata/servers/<servername>/` after selecting the profile. If there is no jar file, none can be selected.

Okay, so this has nothing to do with the Java installation, that is good to know so far. Probably we can find out about the above cases an add some information about that in our docs.

ravenclaw900, 2020-11-11:
Spigot distributes Build Tools, as it is illegal to actually distribute the pure Minecraft JAR. Paper downloads the JAR on first run (then patches it).  I would imagine other servers do similar things.

MichaIng, 2020-11-11:
I'd suggest Spigot then not available in Profiles for download but in ehm the other page where all the sources can be downloaded. There of course it is expected that no jar is available but one needs to compile first.
But there are direct (binary) downloads available without jars as well. I'll run some tests tomorrow to get a complete/verified list, so when I did not do something obvious wrong, we can report to MineOS devs.

CactiChameleon9, 2020-11-11:
> Okay, so this has nothing to do with the Java installation, that is good to know so far. Probably we can find out about the above cases an add some information about that in our docs.

Actually, I think it does. I am testing with cuberite, which doesn't need java, but it doesn't even let me start it because a java runtime (installed java) isn't selected - which seems to be a required field anyway for all minecraft servers (odd, I know), and none are available in the selection menu because it seems to see as none as installed - I think I am correct here because I have used MineOS very very briefly in the past on a pi zero (with dietpi), and I had to select the installed java to let me start cuberite.

MichaIng, 2020-11-11:
> I am testing with cuberite, which doesn't need java, but it doesn't even let me start it because a java runtime (installed java) isn't selected.

That is exactly what I was wondering. Selecting a jar seems to be mandatory but not all Minecraft servers are Java-based, Cuberite is C++. Not sure what you mean by selecting a "JRE", because the field clearly is about selecting a "jar" file, not selecting an installed JRE. What I would have expected, is when you're selecting a Cuberite profile, that there is no jar selection field at all but at best a field to select the executable and add optional command line options 🤔.

> and I had to select the installed java to let me start cuberite.

Very strange, because there is no such field as far as I could see and the attempt to execute Cuberite with Java can only fail. However, to rule it out, did you try to install full Java? `apt install default-jre`

MichaIng, 2020-11-12:
It is not under heavy but steady development: https://github.com/hexparrot/mineos-node
Ah I think here is the one bug I also faced one time: https://github.com/hexparrot/mineos-node/issues/362
Which then leads to: https://github.com/hexparrot/mineos-node/issues/329
But cannot find an issue that is related to the fundamental problem of running non-Java Minecraft servers from MineOS UI: https://github.com/hexparrot/mineos-node/issues

CactiChameleon9, 2020-11-12:
Me looking at the wrong one :D

MattElekHarris, 2020-11-23:
Is it possible to apply the fixes myself? I saw the branch was merged, so I can't seem to change to that branch anymore. Thanks!

ravenclaw900, 2020-11-23:
You could use the `dev` branch, which has that and other fixes/features.

MattElekHarris, 2020-11-23:
Thanks, I wasn't sure what branch it was on. I'll continue testing it, but it all seems to be working great now, thanks!

closed 2020-12-20 by MichaIng