MichaIng/DietPi #3905: DietPi-Software | Mycroft: Skills folder created with incorrect permissons

Joulinar, 2020-11-09:
@MichaIng 
before creating a PR, just like to check with you. There was an issue reported in incorrect permissions for `skills` folder inside `/mnt/dietpi_userdata/mycroft-data`. The directory is created by installation script `dev_setup.sh` and usually it should change permission according the script. 
```
    if [[ ! -d /opt/mycroft/skills ]] ; then
        echo 'This script will create that folder for you.  This requires sudo'
        echo 'permission and might ask you for a password...'
        setup_user=$USER
        setup_group=$(id -gn $USER)
        $SUDO mkdir -p /opt/mycroft/skills
        $SUDO chown -R ${setup_user}:${setup_group} /opt/mycroft
        echo 'Created!'
```
However it's not gonna do that and directory belongs to `root` still. Probably `setup_user=$USER` is not detected correctly.

Basically we would have to options 

1. adjust dietpi-software and change owner after execution `dev_setup.sh`
2. adjust `dev_setup.sh` to change user correctly?

I could create RP if you prefer option 1.

Related forum post: https://dietpi.com/phpbb/viewtopic.php?f=11&t=8252

MichaIng, 2020-11-10:
That is strange indeed, let me try to debug. In case it is easiest to re-set `mycroft:mycroft` owner recursively on all dirs: core, data, log

Joulinar, 2020-11-10:
We already create `mycroft-data` ourselves, let's add `skills` and we are done

MichaIng, 2020-11-10:
Jep indeed, that'll do it. However, still good to understand what's going wrong with `dev_setup.sh` as this might produce other issues, if not now then in the future.

Joulinar, 2020-11-10:
I guess it's again the challenge `sudo` vs native `root`. We call `dev_setup.sh` as user `mycroft`

https://github.com/MichaIng/DietPi/blob/a5407d1b5fefe4e1647eca69d8341c7a5ff0bdf4/dietpi/dietpi-software#L6141

However the directory `skills` is created as user `root`. 🤔 
Therefore I guess following returns `root` as well

```
setup_user=$USER
setup_group=$(id -gn $USER)
```

MichaIng, 2020-11-10:
I was thinking the same first, but that works:
```
2020-11-10 21:24:02 root@dietpi:~# sudo -u www-data bash -c 'echo $USER'
www-data
```

MichaIng, 2020-11-12:
Found it:
```
$SUDO chown -R ${setup_user}:${setup_group} /opt/mycroft
```
`chown` does not follow symlinks, so the above sets the permissions for the `/opt/mycroft` symlink only, but not for the contained skills directory 😄.
```
2020-11-12 15:48:18 root@VM-Buster:/mnt/dietpi_userdata/mycroft-core# ls -l /opt
total 0
lrwxrwxrwx 1 mycroft mycroft 33 Nov 12 14:59 mycroft -> /mnt/dietpi_userdata/mycroft-data
```
It is intended to be a regular directory, so this is not an issue with the Mycroft installer. Pre-creating the directory makes that code block being skipped, which is indeed the cleanest solution.

Xerono, 2020-12-05:
On the dev branch:

```
[ INFO ] DietPi-Software | sudo -u mycroft ./dev_setup.sh -sm, please wait...
sudo: ./dev_setup.sh: command not found
[FAILED] DietPi-Software | sudo -u mycroft ./dev_setup.sh -sm
```

Opening a subshell and setting 

```root@DietPi:/mnt/dietpi_userdata/mycroft-core# chmod +rx dev_setup.sh```

and then retrying let me go on with the installation.

Ernstian, 2020-12-05:
On dev branch:

```
[ INFO ] DietPi-Software | sudo -u mycroft ./dev_setup.sh -sm, please wait...
sudo: ./dev_setup.sh: command not found
[FAILED] DietPi-Software | sudo -u mycroft ./dev_setup.sh -sm
```

Setting

```root@DietPi:/mnt/dietpi_userdata/mycroft-core# chmod +rx dev_setup.sh```

let me continue with the installation.

Besides that, I got it to understand my "Hey Jarvis" as a wakeword once (in about 1 hour of testing), so this probably works.

MichaIng, 2020-12-05:
> Besides that, I got it to understand my "Hey Jarvis" as a wakeword once (in about 1 hour of testing), so this probably works.

Lol, I never hot "Hey Mycroft" to work, "Hey Jarvis" not well (probably one of two times, better after some attempts), "Hey Ezra" almost great and "Christopher" by far best. Not sure if the microphone plays a role as well, or if it needs to learn your voice for a while (not sure if it actually learns it or STT is done a static way). I'll fix the permissions issue.

Ernstian, 2020-12-05:
Ezra didn't work at all (or maybe it did, but with a similar success rate as Jarvis), and I didn't try Christopher (my name is too similiar, and I don't want it to react if I get called :D ) I got it to work it with Jarvis a few more times, but so far, it's not really usable for me. It also always falls back to Pocket Sphinx, because no model for Precise gets found.

>Strange, did you face the missing execute permissions on a fresh install?

If uninstalling via Dietpi and reinstalling it (after updating to the dev branch) counts as fresh, it's fresh. I didn't touch any files there myself.

ravenclaw900, 2020-12-05:
I got "Hey Mycroft" to work most of the time, but it never responds to me.  I know the sound works, because I get the beep when it hears the wake word. Looking at the CLI, it hears me and responds with text, but that's it.

MichaIng, 2020-12-05:
Interesting how different the experiences are. Pocket Sphinx fallback btw is expected in all cases but "Hey Mycroft", AFAIK.

> I got "Hey Mycroft" to work most of the time, but it never responds to me.

You did not install Mimic, did you? I needed to learn that the default British Male voice it the local Mimic instance, there is no online version of it. The other two voices, American Male and Google Voice are online voices which should work OOTB. Check out the docs about how to install local Mimic: https://dietpi.com/docs/software/hardware_projects/#mycroft-ai

ravenclaw900, 2020-12-05:
I used American Male, but I'll try Mimic anyway.

MichaIng, 2020-12-05:
Okay, if American Male fails, then probably the `/var/log/mycroft/audio.log` will give a hint, or `voice.log`, I always mix up those two 😄.

ravenclaw900, 2020-12-06:
After reinstalling, the audio is working some of the time, and to be honest, this is probably the configuration of my VM messing it up.  Don't know why it wasn't working before.

MichaIng, 2020-12-06:
Seems to be an external issue:
- https://community.mycroft.ai/t/googletts-verification-failing/9712?u=michaing
- https://github.com/pndurette/gTTS/issues/232

Looks like next Home Assistant update will have it fixed.

closed 2020-12-20 by MichaIng