MichaIng/DietPi #3906: Wireguard don´t work

adntsierradegata, 2020-11-09:
HI, last version clean installation wireguard don´t work correctly install with dietpi optimised software.
stemctl status wg-quick@wg0-client.service
\u25cf wg-quick@wg0-client.service - WireGuard via wg-quick(8) for wg0/client
   Loaded: loaded (/lib/systemd/system/wg-quick@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-11-09 19:54:01 CET; 5min ago
     Docs: man:wg-quick(8)
           man:wg(8)
           https://www.wireguard.com/
           https://www.wireguard.com/quickstart/
           https://git.zx2c4.com/wireguard-tools/about/src/man/wg-quick.8
           https://git.zx2c4.com/wireguard-tools/about/src/man/wg.8
  Process: 13662 ExecStart=/usr/bin/wg-quick up wg0-client (code=exited, status=127)
 Main PID: 13662 (code=exited, status=127)

nov 09 19:54:01 DietPi wg-quick[13662]: [#] ip link add wg0-client type wireguard
nov 09 19:54:01 DietPi wg-quick[13662]: [#] wg setconf wg0-client /dev/fd/63
nov 09 19:54:01 DietPi wg-quick[13662]: [#] ip -4 address add 10.9.0.2/24 dev wg0-client
nov 09 19:54:01 DietPi wg-quick[13662]: [#] ip link set mtu 1420 up dev wg0-client
nov 09 19:54:01 DietPi wg-quick[13662]: [#] resolvconf -a wg0-client -m 0 -x
nov 09 19:54:01 DietPi wg-quick[13662]: /usr/bin/wg-quick: line 32: resolvconf: command not found
nov 09 19:54:01 DietPi wg-quick[13662]: [#] ip link delete dev wg0-client
nov 09 19:54:01 DietPi systemd[1]: wg-quick@wg0-client.service: Main process exited, code=exited, status=127/n/a
nov 09 19:54:01 DietPi systemd[1]: wg-quick@wg0-client.service: Failed with result 'exit-code'.
nov 09 19:54:01 DietPi systemd[1]: Failed to start WireGuard via wg-quick(8) for wg0/client.

Joulinar, 2020-11-09:
Hi,

many thanks for your message. I guess you like to use Wireguad in client mode. correct?

If yes, you are missing `resolvconf` package.

` /usr/bin/wg-quick: line 32: resolvconf: command not found`

pls check our docs for WireGuard client https://dietpi.com/docs/software/vpn/#wireguard-an-extremely-simple-yet-fast-and-modern-vpn

adntsierradegata, 2020-11-09:
Server mode. is configured.

Joulinar, 2020-11-09:
well you are starting WireGuard in client mode

`nov 09 19:54:01 DietPi wg-quick[13662]: [#] ip link add wg0-client type wireguard`

adntsierradegata, 2020-11-09:
Apologize. Im so sorry waste your time. Thank you very much. All it´s ok.

Joulinar, 2020-11-09:
no problem.

closed 2020-11-09 by Joulinar