MichaIng/DietPi #3908: Dietpi-banner suddenly outputs more then WAN IP address

maartenlangeveld, 2020-11-10:
NanoPI NEO2 /  DietPi v6.33.3
Linux M-DNS 5.8.16-sunxi64 #20.08.16 SMP Sat Oct 24 23:52:46 CEST 2020 aarch64 GNU/Linux

Dietpi-banner suddenly outputs bunch of information as WAN IP: 

```
 ─────────────────────────────────────────────────────
 DietPi v6.33.3 : 17:05 - Tue 11/10/20
 ─────────────────────────────────────────────────────
 - Device model : NanoPi NEO2 (aarch64)
 - Uptime : up 5 days, 2 hours, 15 minutes
 - CPU temp : 41'C : 105'F (Optimal temperature)
 - FQDN/hostname : M-DNS
 - LAN IP : 192.---.---.--- (eth0)
 - WAN IP : {
  "ip": "---.---.---.---",
  "ip_decimal": ----------,
  "country": "Netherlands",
  "country_iso": "NL",
  "country_eu": true,
  "region_name": "South Holland",
  "region_code": "ZH",
  "zip_code": "----",
  "city": "---------",
  "latitude": --.----,
  "longitude": -.----,
  "time_zone": "Europe/Amsterdam",
  "asn": "AS1136",
  "asn_org": "KPN B.V.",
  "hostname": "---------------.fixed.kpn.net",
  "user_agent": {
    "product": "curl",
    "version": "7.64.0",
    "raw_value": "curl/7.64.0"
  }
}
 - Freespace (RootFS) : 23G
 - MOTD : Take care yourself and the people around you :)
```

MichaIng, 2020-11-10:
Ah I recognised it as well, but was busy with another topic. Good to know it was not just me.

Seems the JSON output has changed... although the fields we parse are all still there and IP at the expected field. I'll check what's going on.

MichaIng, 2020-11-10:
Ah lol, the issue simply is that it is a multi line output now. Here is the related commit: https://github.com/mpolden/echoip/commit/cd687f6343548a30fb11f47a143b31031bff6de5
That could make parsing it actually simpler 🤔. I'm now watching the repository to be notified about any change.

ignatiusmb, 2020-11-11:
Hello, is there any reason to not just get the ip straight from https://ifconfig.co/ip?

MichaIng, 2020-11-11:
Because we show the country, and, if available, region as well, which is very informative when using a VPN.

MichaIng, 2020-11-11:
Fixed: #3911

maartenlangeveld, 2020-11-12:
@MichaIng Thanks!

closed 2020-12-20 by MichaIng