MichaIng/DietPi #3909: Redirect fails during APT update

mrcamuti, 2020-11-11:
#### Details:
- Date           | Wed Nov 11 08:09:47 GMT 2020
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- follow the redirect, correctly
#### Actual behaviour:

- curl failing at redirect
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u6_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/t/tzdata/tzdata_2020d-0+deb10u1_all.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-atheros_20190114-1+rpt9_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20190114-1+rpt9_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-iwlwifi_20190114-1+rpt9_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-misc-nonfree_20190114-1+rpt9_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-realtek_20190114-1+rpt9_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20201022-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20201022-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20201022-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20201022-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-sys-mods/raspberrypi-sys-mods_20201026_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

mrcamuti, 2020-11-11:
Updating locale worked to get resolution

Joulinar, 2020-11-11:
Hi,

this seems to be a local DNS issue as you are not able to resolve the host name.

`could not resolve 'archive.raspberrypi.org'`

Do you running a local DNS server like PiHole? Or maybe you could check/change your upstream DNS server?

mrcamuti, 2020-11-11:
I was setting up DietPi to run a PiHole, but during setup, DNS was set to
8.8.4.4. When I did the manual lookup through my browser, redirects not
being honored by curl's -L flag

mrcamuti, 2020-11-11:
And for clarity - was able to get this working by changing Locale, and
rebooting a couple times.

Joulinar, 2020-11-11:
means it's solved and can be closed?

MichaIng, 2020-11-11:
Not sure if I understand the correlation between curl and the actual name resolution with APT you faced, and also the locale should neither have an effect on name resolution nor, definitely not, on redirects being performed by curl or not, as long as the initial name resolution and connection succeeded of course 🤔.

However, good that it works now.

mrcamuti, 2020-11-12:
> resolution failures with APT you faced, and also the locale should neither

Only thing I changed between issue (two re-images, same issue both times)
and working was killing IPv6 and changing locale, I agree that curl should
have redirected correctly. So, please feel free to close, more just wanted
to do an updated FYI.

Joulinar, 2020-11-12:
probably IPv6 was the issue

MichaIng, 2020-11-12:
Likely, sad but fact that IPv6 is still responsible for a variety of network issues, some on a level that exceeds my knowledge by far. I just remember the 20-40+ seconds delay that we faced on our VPS, which forced us to disable IPv6 without any idea what the reason might be, neither us, nor VPS provider. And network on the VPS is build up on an extremely simply low-level basis, very transparent down to the kernel level.

Okay I'm closing the issue, however, feel free to reopen if required.

closed 2020-11-12 by MichaIng