MichaIng/DietPi #3910: Unable to install PiVPN Wireguard on Orange Pi Zero

mk13139, 2020-11-11:
**- DietPi version** | 6.33.3
**- Distro version** | Debian Buster (official Orange Pi Zero Debian image converted to DietPi)
**- Kernel version** | 5.3.5
**- SBC model** | Orange Pi Zero
**- Power supply used** | 5v 2.4A
**- SDcard used** | SanDisk ultra class 10

**Additional Information**
- Software title | PiVPN
- Was the software title installed freshly or updated/migrated? Attempt to install PiVPN with Wireguard
- Can this issue be replicated on a fresh installation of DietPi? Yes

**Steps to reproduce**
1. Flash official Orange Pi Zero Debian image
2. Convert to DietPi using PREP_SYSTEM_FOR_DIETPI.sh script
3. Select and install PiVPN from 'Optimized Software' section
4. Choose Wireguard install option

**Expected behaviour**
- Installation of PiVPN Wireguard

**Actual behaviour**
- Installer exits when failed to install 'linux-headers-amd64'

**Extra details**
- PiVPN installer script fails to recognize CPU architecture of Orange Pi Zero (armhf/armv7l)

Joulinar, 2020-11-11:
Hi,

Pls can you share the error message when the installation failed.

mk13139, 2020-11-11:
> Hi,
> 
> Pls can you share the error message when the installation failed.

Updated OP.

Joulinar, 2020-11-11:
hmm can you check your available package architecture 

`dpkg --print-architecture`

mk13139, 2020-11-11:
```
dpkg --print-architecture
armhf
```

Joulinar, 2020-11-11:
strange that PiVPN install script is trying to install `amd64` packages. That would be an issue of PiVPN and outside DietPi as we just download the install script from PiVPN side. Will do some test on my RPi3B+

mk13139, 2020-11-11:
Yes I guess armhf architecture is not supported by PiVPN.
They claim to support the following boards:
`All SBC's running DietPi. This is currently: Odroid C1, Odroid C2 (arm64), Odroid XU3/4, Pine A64, NanoPi NEO, NanoPi NEO Air, NanoPi M1.`
The NanoPi has also a armv7 architecture right?

Joulinar, 2020-11-11:
well my RPi3B+ is detected correctly ;)

```
root@DietPi3:~# uname -a
Linux DietPi3 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux
root@DietPi3:~# dpkg --print-architecture
armhf
root@DietPi3:~#
```

Probably PiVPN is assuming your Orange Pi as `amd64` architecture.

As workaround, you could try to install native WireGuard from `dietpi-software`

mk13139, 2020-11-11:
Yeah, for now I just installed it straight from the CLI...
Is there any benefit from installing it via `dietpi-software`?

Joulinar, 2020-11-11:
`dietpi-software` is taking care on all needed steps automatically and will perform required optimisation to have software adjusted to DietPi needs. Just select the required software from `dietpi-software` catalogue and magic happen 😉

mk13139, 2020-11-11:
Well, unfortunately Wireguard is not listed as a separate package. Only PiVPN is listed as Wireguard server :(

Joulinar, 2020-11-11:
probably it's disabled. Let's check

`dietpi-software list | grep wireguard`

mk13139, 2020-11-11:
Yes it is:

```
dietpi@DietPi:~$ sudo dietpi-software list | grep wireguard
id 117 | =0 | pivpn: openvpn/wireguard server install & management tool | +git | https://d            ietpi.com/phpbb/viewtopic.php?p=3469#p3469
id 172 | =0 | wireguard: an extremely simple yet fast and modern vpn | disabled for orange            pi zero (armv7l) | https://dietpi.com/phpbb/viewtopic.php?p=16308#p16308
```

Joulinar, 2020-11-11:
I guess @MichaIng knows why

MichaIng, 2020-11-11:
Do you indeed use the `linux-image-arm64` kernel package from the Debian repository, or is it Armbian or such? The version `5.3.5+` doesn't seem to match any current kernel package version.
```
dpkg -l | grep 'linux-image'
```

Using a current Linux package from Debian `buster-backports` or Armbian repository would contain a builtin WireGuard module, which enables the install option.

mk13139, 2020-11-11:
`dpkg -l | grep 'linux-image'`

Does not return anything. `uname -a` shows the following:

```
dietpi@DietPi:~$ uname -a
Linux DietPi 5.3.5+ #2 SMP Fri Nov 15 16:24:33 CST 2019 armv7l GNU/Linux
```
Are you suggesting to enable the backports repository?

MichaIng, 2020-11-11:
Hmm, how exactly did you install the initial Debian or where did you get the image from?

In attempt do find out a bit more:
```
ls -l /lib/modules/
dpkg -S /boot /lib/modules
```

mk13139, 2020-11-11:
I just flashed the latest available Debian image from the official Orange Pi website: [http://www.orangepi.org/downloadresources/](url)
Which is pretty old actually..
```
dietpi@DietPi:~$ ls -l /lib/modules/
totaal 4
drwxr-xr-x 3 root root 4096 nov 15  2019 5.3.5+
dietpi@DietPi:~$ sudo dpkg -S /boot /lib/modules
base-files: /boot
dpkg-query: geen pad gevonden dat overeenkomt met /lib/modules
dietpi@DietPi:/lib/modules$ ls
5.3.5+
```
Maybe it is wiser to flash the latest Armbian image, as it includes kernel 5.8 already..
I will try that tomorrow.

MichaIng, 2020-11-11:
> Maybe it is wiser to flash the latest Armbian image, as it includes kernel 5.8 already..

Probably. But the official Debian kernel as well supports it, Linux 5.8 when taken from backports: https://packages.debian.org/buster-backports/linux-image-armmp
```
apt install linux-image-armmp/buster-backports
```
Orange Pi Zero device tree is explicitly contained.

What I am not 100% sure about is if u-boot needs to be configured for this. At least we could have a look into `boot.ini` (or `boot.cmd`) if the device tree and kernel file names still match the new kernel.

MichaIng, 2020-11-11:
I wanted to download the Orange Pi Debian image but fail to. Baidu only works with a dedicated downloader program it seems, that that is a no-go for me. On Google Drive, when hitting the download button, I need to accept download regardless of missing anti-virus check (file too large) but nothing happens. This is true somehow in all cases recently, the same happened when I tried to download Firefly images. I wonder why all (?) Chinese manufacturers host on those two sites only, instead of hosting those few GiB on their own servers.
Just tried it on MS Edge, where even the anti-virus verification does not show up.

mk13139, 2020-11-12:
I've flashed the latest Armbian as base image, however the Wireguard package is still disabled for the Orange Pi SBC:
```
dietpi@DietPi:~$ sudo dietpi-software list | grep wireguard
id 117 | =0 | pivpn: openvpn/wireguard server install & management tool | +git | https://dietpi.com/phpbb/viewtopic.php?p=3469#p3469
id 172 | =0 | wireguard: an extremely simple yet fast and modern vpn | disabled for orangepi zero (armv7l) | https://dietpi.com/phpbb/viewtopic.php?p=16308#p16308
```
```
dietpi@DietPi:~$ sudo dpkg -l | grep 'linux-image'
ii  linux-image-current-sunxi              20.08.14                     armhf        Linux kernel, version 5.8.16-sunxi
```
```
dietpi@DietPi:~$ ls -l /lib/modules/
totaal 4
drwxr-xr-x 3 root root 4096 nov 12 09:35 5.8.16-sunxi
dietpi@DietPi:~$ dpkg -S /boot /lib/modules
linux-dtb-current-sunxi, linux-image-current-sunxi, base-files: /boot
linux-image-current-sunxi: /lib/modules
```

MichaIng, 2020-11-12:
Yes there is one change that needs to be done since the `wireguard.ko` changed it's location recently: https://github.com/MichaIng/DietPi/commit/8ee591baa826860aad0454225418a03497fa2a96
```
sed -i 's|kernel/net|kernel{,/drivers}/net|' /boot/dietpi/dietpi-software
```

mk13139, 2020-11-12:
Yes I can confirm that above path change works!
Thanks, looking forward for the DietPi v6.34 release.

BTW, I did not test the PiVPN install script. The Wireguard package alone is sufficient for me.

MichaIng, 2020-11-12:
AFAIK, PiVPN does not yet support builtin WireGuard modules. I posted our solution a while ago, but so far only RPi and x86_64 are supported and built the WireGuard module via headers, AFAIK. Strange that in your case, even it is clearly armhf/armv7l, it tried to install the x86_64 headers `linux-headers-amd64`. Probably I find time to review/fix their installer 😎.

mk13139, 2020-11-13:
Yeah that would be cool!

Is it correct that the installer script of the Wireguard package does not enable `net.ipv4.ip_forward=1` in `/etc/sysctl.conf`?
I had to add it to be able to access the LAN behind the Wireguard server.

Joulinar, 2020-11-13:
usually `net.ipv4.ip_forward=1` should be set on interface startup. At least if you use `dietpi-software` Wireguard installation. It's part of `wg0.conf` file

```
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.$(sed -n 3p /run/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.$(sed -n 3p /run/dietpi/.network).forwarding=1
```

mk13139, 2020-11-13:
Yes you are right, I overlooked it.
I commented the line in `/etc/sysctl.conf` again and indeed I'm still able to access the LAN.

Joulinar, 2020-11-13:
that's the magic of DietPi I have talking above 😉

MichaIng, 2020-11-28:
I was just thinking that we still don't know why the PiVPN installer detects the architecture incorrectly. Good that a related bug report links this issue. I'll have a look into it when I find time, should be easy to fix.

MichaIng, 2020-12-11:
Indeed a bug in the PiVPN installer: https://github.com/pivpn/pivpn/issues/1180#issuecomment-743445513
But @mk13139 your board will not be supported after the fix, as the 5.3.5+ kernel you use most likely has no builtin WireGuard and the correct kernel headers package for it is not known or does not exist. Where did you pull the initial Debian image from?

Xerono, 2020-12-23:
After getting an error installing Wireguard, installation of all other software fails due to an apt upgrade error with wireguard-dkms. Running

```
apt remove wireguard-dkms
apt remove wireguard-tools
```
removes the problematic wireguard packages and lets you install everything like normal again. I don't think this needs a fix as it looks like the next update will fix the Wireguard installation, but it may be helpful if all installed packages get removed if the installation of software fails.

MichaIng, 2021-02-08:
The fix has finally been uploaded. Marking this as closed.

closed 2021-02-08 by MichaIng