MichaIng/DietPi #3912: No DNS resolve on fresh install

bennydiamond, 2020-11-11:
#### Details:
- Date           | Fri Sep 11 01:35:10 EDT 2020
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi B (armv6l) (ID=0)
- Kernel version | Linux DietPi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ping -c 1 -W 10 dns9.quad9.net
- Exit code      | 2
- Software title | DietPi-Login
#### Steps to reproduce:
1. Flash DietPi image on SDCard
2. Configure unattended setup using dietpi.txt (Mostly headless & static IP)
> # IMPORTANT:
> # - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
> # - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.
> 
> #------------------------------------------------------------------------------------------------------
> # D I E T - P I
> # DietPi-Automation settings, applied on first boot of DietPi only, ONCE!
> #------------------------------------------------------------------------------------------------------
> # By setting this to "1" you accept the DietPi GPLv2 license and skip the related interactive dialog.
> # - Full license text: /boot/dietpi-LICENSE.txt
> AUTO_SETUP_ACCEPT_LICENSE=1
> 
> ##### Language/Regional Options #####
> # Locale: eg: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
> AUTO_SETUP_LOCALE=C.UTF-8
> 
> # Keyboard Layout eg: "gb" / "us" / "de" / "fr"
> AUTO_SETUP_KEYBOARD_LAYOUT=us
> 
> # Timezone eg: "Europe/London" / "America/New_York" | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
> AUTO_SETUP_TIMEZONE=America/New_York
> 
> ##### Networking Options #####
> # Enable Ethernet or WiFi adapter: 1=enable | 0=disable
> # - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled.
> # - If using WiFi, please edit dietpi-wifi.txt to pre-enter credentials.
> AUTO_SETUP_NET_ETHERNET_ENABLED=1
> AUTO_SETUP_NET_WIFI_ENABLED=0
> 
> # Force ethernet speeds: 0=automatic speed | 10 = 10Mbit | 100 = 100Mbit etc
> # - This is mainly aimed at Pine A64 which may have an HW issue that causes unstable 1Gbit link.
> AUTO_SETUP_NET_ETH_FORCE_SPEED=0
> 
> # WiFi country code: 2 uppercase character value (e.g. GB US DE JP): https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
> AUTO_SETUP_NET_WIFI_COUNTRY_CODE=GB
> 
> # Enter your Static Network details below, if applicable.
> AUTO_SETUP_NET_USESTATIC=1
> AUTO_SETUP_NET_STATIC_IP=192.168.0.253
> AUTO_SETUP_NET_STATIC_MASK=255.255.254.0
> AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
> AUTO_SETUP_NET_STATIC_DNS=1.1.1.1
> 
> # Hostname
> AUTO_SETUP_NET_HOSTNAME=DietPi
> 
> ##### Misc Options #####
> # Swap space size to generate: 0 => disable | 1 => auto | 2 and up => size in MiB
> AUTO_SETUP_SWAPFILE_SIZE=1822
> # Swap space location: "zram" => swap space on /dev/zram0 (auto-size = 50% of RAM size) | /path/to/file => swap file at location (auto-size = 2 GiB minus RAM size)
> AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
> 
> # Set to "1" to disable HDMI output (and GPU/VPU where supported) for supported devices: RPi, Odroid C1, Odroid C2
> AUTO_SETUP_HEADLESS=1
> 
> # Unmask (enable) systemd-logind service (including dbus), which is masked by default on DietPi
[... 194 lines cut ...]
> DEV_GITOWNER=MichaIng
> 
> #------------------------------------------------------------------------------------------------------
> # D I E T - P I
> # Settings, automatically added by dietpi-update
> #------------------------------------------------------------------------------------------------------
> 

#### Expected behaviour:
DNS resolving should be working
- ...
#### Actual behaviour:
DNS resolving is not working

> root@DietPi:/boot# ping google.com
> ping: google.com: Temporary failure in name resolution

- ...
#### Extra details:
This is a simple reproductible scenario. Just flash image on SD card, configure dietpi.txt with quoted file above, run, ping, fail.
- ...
#### Additional logs:
```
ping: dns9.quad9.net: Temporary failure in name resolution
```

Joulinar, 2020-11-11:
Hi,

as far as I can see you are using Cloudflare DNS, pls try another upstream DNS server like Quad9 or Google DNS.

bennydiamond, 2020-11-11:
> 
> 
> Hi,
> 
> as far as I can see you are using Cloudflare DNS, pls try another upstream DNS server like Quad9 or Google DNS.

Hi!
Will simply changing the DNS ip in dietpi.txt and reboot do the trick? If not, where to I configure what DNS is being used?

Joulinar, 2020-11-11:
settings in `/boot/dietpi.txt` are applied during initial first boot only. during normal operation you can use `dietpi-config`

Pls can you post following for DHCP as well as STATIC

```
cat /etc/network/interfaces
cat /etc/resolv.conf
```

bennydiamond, 2020-11-11:
```
root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Loopback
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.0.253
netmask 255.255.254.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9 149.112.112.112

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.253
netmask 255.255.254.0
gateway 192.168.0.1
wireless-power off
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
#dns-nameservers 9.9.9.9 149.112.112.112
root@DietPi:~# cat /etc/resolv.conf
nameserver 8.8.8.8
```

Please note that resolv.conf has been manually edited. On reboot, it reverts to using 1.1.1.1

Joulinar, 2020-11-11:
how does it looks like if you switch to DHCP?

bennydiamond, 2020-11-11:
```
root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.253
netmask 255.255.254.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 0.0.0.0
netmask 0.0.0.0
gateway 0.0.0.0
#dns-nameservers 0.0.0.0
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
root@DietPi:~# cat /etc/resolv.conf
domain local
search local
nameserver 192.168.0.252
```

192.168.0.252 is my primary Pi-Hole instance. I do not wish for this troublesome Pi to go through Pi-Hole for DNS resolution.

Joulinar, 2020-11-11:
just for testing, died you tried using `192.168.0.252` as DNS server? just to see how it behaving?

bennydiamond, 2020-11-11:
It works.... :|

Ok, most likely the issue lies some place else; most likely the router...

closed 2020-11-11 by bennydiamond

Joulinar, 2020-11-11:
probably some network device blocking DNS request? Some firewall like pfSense? Not sure what you have on your network 😃

bennydiamond, 2020-11-11:
> 
> 
> probably some network device blocking DNS request? Some firewall like pfSense? Not sure what you have on your network 😃

Running DD-WRT on a Netgear R7000. I must have something that reject outgoing DNS requests if they are not made through the Pi-Hole...

Joulinar, 2020-11-11:
maybe an active security feature to ensure PiHole is used? 🤔

bennydiamond, 2020-11-11:
Most likely a bad/paranoid combination of DNS redirection on DD-WRT. I do have `Forced DNS Redirection` enabled in DD-WRT and DNSMasq is configured to redirect to my Pi-Hole.

Does DietPi enables DNSSEC by default? I'm no DNS guru but that could explain it. DD-WRT force redirection and maybe breaks DNS reply signature. I don't know. Anyway it's clearly not an issue with DietPi.

Thanks a lot for the swift support. I couldn't ask for more.

Joulinar, 2020-11-11:
by default DietPi has no security on DNS

MichaIng, 2020-11-12:
FYI, DNS security is about to come: #3872