MichaIng/DietPi #3913: Samba can't move files

Dr0bac, 2020-11-11:
In last few months i noticed that i can't move some files from raspberry pi to windows pc, but i can copy them just fine..
I get file access denied error.
"You require permission from Unix User/plex to make changes to this file"

That file is plex:dietpi owned. In what group samba belongs? Why it can't move files in dietpi group?

Joulinar, 2020-11-11:
Hi,

user permissions depends on your settings in `/etc/samba/smb.conf`. There you have a section defining your share, the user and access policies.

Dr0bac, 2020-11-11:
There i have
````
[global]

        workgroup = WORKGROUP
        server string = %h server
        dns proxy = no
        log file = /var/log/samba/log.%m
        max log size = 1000
        syslog only = no
        syslog = 0

        panic action = /usr/share/samba/panic-action %d

        security = user
        encrypt passwords = true
        passdb backend = tdbsam
        obey pam restrictions = yes
        unix password sync = yes

        passwd program = /usr/bin/passwd %u
        passwd chat = *Enter\snew\s*\spassword:* %n\n *Retype\snew\s*\spassword:* %n\n *password\supdated\ssuccessfully* .
        pam password change = yes
        map to guest = bad user

        load printers = no
        printcap name = /dev/null
        disable spoolss = yes

[dietpi]
        comment = DietPi Share
        path = /mnt/dietpi_userdata
        browseable = yes
        create mask = 0775
        directory mask = 0775
        valid users = dietpi
        writeable = yes
max connections = 8
````

Joulinar, 2020-11-11:
can you share an example on permissions your file has

Dr0bac, 2020-11-11:
Permissions of the file i try to move to windows are
````
-rw-rw---- 1 plex dietpi 884910444 Jul 20 20:50 'Ordinary Love (2019).mp4'
````

Joulinar, 2020-11-11:
and the directory permission the file is located on?

Dr0bac, 2020-11-11:
drwxr-xr-x  2 plex dietpi 4096 Jul 20 21:22 'Ordinary Love (2019)'

So i indeed don't have permissions to move it.  Permissions are set in radarr settings which are

[image]

Do i need to change it?

Joulinar, 2020-11-11:
just for testing, if you change folder permission to 775. Will this allow you to move it via SAMBA?

Dr0bac, 2020-11-11:
i downloaded movie again with radarr to test it without changing permissions. si it's still 0664
and folder permisions are fine `drwxrwxr-x 2 radarr dietpi 4096 Nov 12 00:38 'The New Mutants (2020)`

I got error in radarr, and i see that error every time i download movie. 

````
Unable to apply permissions to: /mnt/dietpi_userdata/Plex/Boris_Movies/The New Mutants (2020)/The New Mutants (2020).mp4: Error setting permissions: EPERM
NzbDrone.Mono.Disk.LinuxPermissionsException: Error setting permissions: EPERM
   at NzbDrone.Mono.Disk.DiskProvider.SetPermissions(String path, String mask) in D:\a\1\s\src\NzbDrone.Mono\Disk\DiskProvider.cs:line 79
   at NzbDrone.Core.MediaFiles.MediaFileAttributeService.SetMonoPermissions(String path, String permissions) in D:\a\1\s\src\NzbDrone.Core\MediaFiles\MediaFileAttributeService.cs:line 87
````

movie permissions are `-rw-rw---- 1 debian-transmission dietpi 1220758574 Nov 12 00:36 'The New Mutants (2020).mp4'`

MichaIng, 2020-11-12:
What file system type is it?

The permissions look pretty fine, dir is 775 and file is 664, as intended, and group is `dietpi`, so the `dietpi` user has write access. Would be interesting if it worked when mounting the Samba share to the same system or another Linux system, to rule out that Windows' Samba client does not block access (due to wrong checks) in the first place instead of letting the server do and report actual UNIX permission issues in case.

Dr0bac, 2020-11-14:
@MichaIng i donwloaded few movies and permissions are indeed fine, and i can transfer that files to windows. I had few movies with strict permissions. Who know what happened. 

Can you help me with post processing script in radarr?

i want to restart transmission when no torrents are present, and i can't save it because of an error.
first i was getting `Failed to connect to bus: No such file or directory` but after installing dbus with `apt-get install dbus` that error is gone. Now i have error:
````
2020-11-14 01:16:53.2|Error|transmission_reboot.sh|We trust you have received the usual lecture from the local System
2020-11-14 01:16:53.2|Error|transmission_reboot.sh|Administrator. It usually boils down to these three things:
2020-11-14 01:16:53.2|Error|transmission_reboot.sh|    #1) Respect the privacy of others.
2020-11-14 01:16:53.2|Error|transmission_reboot.sh|    #2) Think before you type.
2020-11-14 01:16:53.2|Error|transmission_reboot.sh|    #3) With great power comes great responsibility.
2020-11-14 01:16:53.2|Error|transmission_reboot.sh|sudo: no tty present and no askpass program specified
2020-11-14 01:16:53.2|Warn|RadarrErrorPipeline|Invalid request Validation failed: 
 -- : Script exited with code: 1
```` 
script permissions : `-rwxrwxr-- 1 radarr dietpi  409 Nov 14 00:56 transmission_reboot.sh`

and script:
````
#!/bin/bash

# sleep 3m

# list torrents that are present in transmission

TORRENT=`transmission-remote --auth username:password --list`

# if there are no numbers (1-9) in string then restart transmission service

if [[ ! $TORRENT =~ [1-9] ]]; then
  service transmission-daemon restart
fi
````

I guess that radarr:dietpi can't restart transmission-daemon, but when i change script ownership to root:root and 0777, i get permission denied in radarr webui.

MichaIng, 2020-11-17:
Okay, great that it works so far.

Indeed an unprivileged user (radarr) cannot restart system services OOTB and you don't need dbus for this, but there is indeed a way via dbus with policykit rules or a simpler way by granting systemctl access via sudo.

But first of all, why do you actually want to restart Transmission if there are no torrents? I guess all cleanup and maintenance steps are either done on a regular basis already or can be triggered via `transmission-remote`?
Ah or is it because of still present increasing memory usage by Transmission?

#### "Simple" solution via sudoers to allow Radarr using sudo without interactive password input to become root and execute: "systemctl restart transmission-daemon"
```
cat << '_EOF_' > /etc/sudoers.d/radarr
Cmnd_Alias TRANSMISSION = /bin/systemctl restart transmission-daemon, ! /bin/systemctl restart transmission-daemon?
radarr ALL=(root) NOPASSWD: TRANSMISSION
_EOF_
```
The command alias is required here to max security by allowing only "systemctl restart transmission-daemon" without any further arguments, as "systemctl restart transmission-daemon ssh" could otherwise be used to restart OpenSSH additionally 😉.
Then use `sudo systemctl restart transmission-daemon`.

#### More complex solution, a security expert might prefer, although I cannot see a practical advantage in this particular case
```
apt update
apt install dbus policykit-1
cat << '_EOF_' > /etc/polkit-1/rules.d/99-radarr.rules
// Allow radarr to manage transmission-daemon.service;
// fall back to implicit authorization otherwise.
polkit.addRule(function(action, subject) {
    if (action.id == "org.freedesktop.systemd1.manage-units" &&
        action.lookup("unit") == "transmission-daemon.service" &&
        subject.user == "radarr") {
        return polkit.Result.YES;
    }
});
_EOF_
```
This allows Radarr to manage the Transmission service in other ways, stop, start, status, probably enable/disable, didn't play much with it, for sure if could be more fine grained. The bonus is that you don't need sudo anymore, but dbus and policykit services are additionally running.

Dr0bac, 2020-11-17:
Transmission doesn't release cache memory after torrent is downloaded, after days of inactivity transmission is still using that memory, so i want to restart it if no torrents are present in trasmission.. Thanks for helping me

closed 2020-11-17 by Dr0bac

MichaIng, 2020-11-18:
The package that you want is [transmission-daemon](https://packages.debian.org/bullseye/transmission-daemon) btw, not the GUI frontend. That seems to work without pulling too much other packages from Bullseye. You could try:
```
apt install transmission-daemon/testing transmission-common/testing libevent-2.1-7/testing
```
You could create APT pins to allow only these three packages to be auto upgraded while nothing else is allowed to be installed from Bullseye. There is otherwise always the risk that a new dependency tail upgrades half of the system. But actually Bullseye is not that unstable anymore, feature freeze it about to come, so probably you'll have more benefits then downsides after all 😄.