MichaIng/DietPi #3914: WINE

jakues, 2020-11-11:
### Creating a software request

#### Formal software information
- Software title | 
- Short description | 
- Official URL (if available) | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2020-11-11:
Hi,

many thanks for the request. According install page, WineHQ packages are for x86 architecture only; they cannot be installed on an ARM system. Means it limits the usage. It can't be used on SBC like Raspberry Pi.

jakues, 2020-11-11:
but twister os support wine. hmmm

Joulinar, 2020-11-11:
well that`s how it is stated on WineHQ web site

ravenclaw900, 2020-11-12:
TwisterOS supports Wine by using Box86, an emulator that translates armhf calls to x86 calls.

jakues, 2020-11-12:
> TwisterOS supports Wine by using Box86, an emulator that translates armhf calls to x86 calls.

if remove box86 wine not work right ?

ravenclaw900, 2020-11-12:
Yes.

MichaIng, 2020-11-12:
Hmm, so you need to run an x86 emulator to run a Windows emulator to run a Windows program. Probably you should simply run a program that is available for Linux or can be compiled for Linux on ARM 😄.

Jokes aside, since Windows programs are usually x86 only anyway, there is no point to get this running on ARM, IMO. Windows ARM64 is on the way, but it will take a looooong time until Windows-only software developers start to provide compatible software I'm afraid 😉. If there is really a need to run an x86 Windows program on e.g. Raspberry Pi 4, then qemu static emulation or even a VM should be easier, isn't it?

jakues, 2020-11-12:
> Hmm, so you need to run an x86 emulator to run a Windows emulator to run a Windows program. Probably you should simply run a program that is available for Linux or can be compiled for Linux on ARM [...]

yep

CactiChameleon9, 2020-11-13:
Just wanted to add, that box86 is translation (I think - check their github) instead of emulation. It tries to use native binaries making it faster than qemu.

MichaIng, 2020-12-05:
The combination SBC + Box64 + Wine x86 is shown here: https://www.cnx-software.com/2020/11/19/box86-is-an-x86-emulator-for-raspberry-pi-and-other-32-bit-arm-platforms/
Especially for OpenGL software/games support seems to be not (yet) perfect, but the combination in general seems to work. So looks like we might want to add both, Box64 and Wine x86 and make the second depend on the first on ARM boards.

MichaIng, 2020-12-15:
For Odroids, thanks to @meveric, there is a DEB package available for WINE via Box86: https://dietpi.com/meveric/pool/main/b/
So we can easily make it available there. For other SoCs, we need to either build and host packages or build Box86 on demand on install.

I'll just reference the Box86 request here, which IMO makes sense to implement it first: https://github.com/MichaIng/DietPi/issues/3957

jakues, 2020-12-15:
> For Odroids, thanks to @meveric, there is a DEB package available for WINE via Box86: https://dietpi.com/meveric/pool/main/b/
> So we can easily make it available there. For other SoCs, we need to [...]

gud af

meveric, 2020-12-15:
my package comes with a wrapper box86_wine to run box86 in combination with wine, still a command line tool, but it helps alot.

For example:
```
export MESA=1 
export WINEPREFIX=/home/odroid/.PlayOnLinux/wineprefix/Anno1602 
cd /home/odroid/.PlayOnLinux/wineprefix/Anno1602/drive_c/GOG\ Games/Anno\ 1602/
box86_wine 1602.exe
```
or as a one-liner (assuming you're already in the correct folder)
`MESA=1 WINEPREFIX=/home/odroid/.PlayOnLinux/wineprefix/Anno1602 box86_wine 1602.exe`

once again, this is standard x86/armhf software and not ODROID specific

MichaIng, 2020-12-16:
> my package comes with a wrapper box86_wine to run box86 in combination with wine

Very nice. After upcoming DietPi release I'll play a bid with it