MichaIng/DietPi #3915: update to 6.33.3 fails

andyrtr, 2020-11-12:
current running DietPi version | 6.32.2
SBC model | Device model : RPi B (armv6l)

when running dietpi-update it fails giving this output:

Adding 'diversion of /boot/overlays/wittypi.dtbo to /usr/share/rpikernelhack/overlays/wittypi.dtbo by rpikernelhack'
Unpacking raspberrypi-kernel (1.20201022-1) over (1.20200902-1) ...
dpkg-deb (subprocess): decompressing archive member: lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive /tmp/apt-dpkg-install-ID3sRk/07-raspberrypi-kernel_1.20201022-1_armhf.deb (--unpack):
 cannot copy extracted data for './boot/COPYING.linux' to '/usr/share/rpikernelhack/COPYING.linux.dpkg-new': unexpected end of file or stream
Removing 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'

Then it removes all the stuff and breaks the update. Is the package corrupt or may this be caused by some other reason I can solve here?

Joulinar, 2020-11-12:
Hi,

Many thanks for your message. Looks like RPi kernel update failed. Pls can you try `apt update && apt upgrade`

andyrtr, 2020-11-12:
Fails at random points. Let's see if I can manage to fix existing filesystem errors.

root@dietpi:~# dosfsck /dev/mmcblk0p1
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1
Perform changes ? (y/n) y
/dev/mmcblk0p1: 300 files, 112094/516191 clusters

After reboot the dirtybit is set again.

Joulinar, 2020-11-12:
you could try to clear cache as well.
`/boot/dietpi/func/dietpi-set_software apt-cache clean`

perform file system check on 2nd partition
`fsck -a /dev/mmcblk0p2`

Check for kernel errors
`dmesg -l emerg,alert,crit,err`

andyrtr, 2020-11-12:
I couldn't manage to fix the filesystems. I've done a fresh installation to a new sdcard. Thanks for your help. Closing this report now.

closed 2020-11-12 by andyrtr