MichaIng/DietPi #3918: dietpi-update failure

prov3it, 2020-11-17:
Okay, i haven't been contributing to this issue anymore, but its still present. In my case the issue is originating from having set up a openvpn server and alter the subnet in the /etc/openvpn/server.conf file. The old server address was still being used for dns. doing a ```grep -Ril "10.16.0.1" /etc``` gave me all the files containing this pattern:
/etc/resolvconf/run/interface/tun1.openvpn
/etc/resolvconf/run/resolv.conf
/etc/resolv.conf
/etc/netns/protected/resolv.conf

Removing /etc/resolvconf/run/interface/tun1.openvpn was not an option. Its recreated after reboot. /etc/resolvconf/run/resolv.conf is linked to /etc/resolv.conf. 

Altering /etc/resolv.conf was my temp solution.

To answer your previous question:
cat /etc/network/interfaces{.d/*} doesnt give me any output. Directory {.d} is empty. /etc/network/interfaces:
```
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.2.20
netmask 255.255.255.0
gateway 192.168.2.254
dns-nameservers 192.168.2.20

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 0.0.0.0
netmask 0.0.0.0
gateway 0.0.0.0
#dns-nameservers 0.0.0.0
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

The strangest thing is, 10.16.0.* is not the subnet used by pivpn, its 10.8.0.*. I changed it to 172.16.0.*. Where is this 10.16.0.1 coming from?!

_Originally posted by @prov3it in https://github.com/MichaIng/DietPi/issues/3555#issuecomment-728897444_

Joulinar, 2020-11-17:
Hi,

it doesn't make sense to have the issue reported in 2 different treads. Pls stick to one. 😉 

DNS Server settings can be done inside `dietpi-config` > `network`. Maybe you can have a look to your adapter. As well pls can you have a look to PiHole if used. We had a couple of reports that PiHole is blocking `raw.githubusercontent.com`

prov3it, 2020-11-17:
> Hi,
>
> it doesn't make sense to have the issue reported in 2 different treads. Pls stick to one. 😉
>
> DNS Server settings can be done inside `dietpi-config` > `network`. Maybe you can have a look to [...]

Im not able to reopen closed issues. Commenting and opening a new one with a connection to the older issue was my only option. Its common sense that you work on the open issue and not the closed one. But i needed to refer to the older one somehow.

Pihole is not blocking raw.githubusercontent.com as i can execute the command without issue on the command line.

Dont play down on my report regarding this issue. Im doing extensive research on this issue for the better of DietPi. You dont have to put me next to captain obvious.

MichaIng, 2020-11-17:
Copy&paste from the other thread 😄:

`/etc/resolvconf/run/interface/tun1.openvpn` seems to be what induces the nameserver entry but that should be reset and reboot at latest and otherwise be part of the OpenVPN config somewhere. Or probably it has been placed by PiVPN installer somehow?

Not sure how this `/etc/resolvconf/run/interface` works at all, cannot find something about it in the man pages. Also I wonder why it is tun1 instead of tun0 (the default) 🤔.

And for the VPN server it does not make any sense to mess with the DNS entries at all, while this only makes sense for the clients to prevent DNS leaks... I'm confused here as well. I'd go through all OpenVPN and PiVPN settings and as well remove `/etc/resolvconf/run/interface/tun1.openvpn` to see if/when it reappears, e.g. if restarting OpenVPN does it.

prov3it, 2020-11-17:
Thanks for your reply @MichaIng. I Think that last file is created by update-resolv-conf i believe, which is located in /etc/openvpn:
```
#!/bin/bash
# 
# Parses DHCP options from openvpn to update resolv.conf
# To use set as 'up' and 'down' script in your openvpn *.conf:
# up /etc/openvpn/update-resolv-conf
# down /etc/openvpn/update-resolv-conf
#
# Used snippets of resolvconf script by Thomas Hood and Chris Hanson.
# Licensed under the GNU GPL.  See /usr/share/common-licenses/GPL. 
# 
# Example envs set from openvpn:
#
#     foreign_option_1='dhcp-option DNS 193.43.27.132'
#     foreign_option_2='dhcp-option DNS 193.43.27.133'
#     foreign_option_3='dhcp-option DOMAIN be.bnc.ch'
#

if [ ! -x /sbin/resolvconf ] ; then
    logger "[OpenVPN:update-resolve-conf] missing binary /sbin/resolvconf";
    exit 0;
fi

[ "$script_type" ] || exit 0
[ "$dev" ] || exit 0

split_into_parts()
{
	part1="$1"
	part2="$2"
	part3="$3"
}

case "$script_type" in
  up)
	NMSRVRS=""
	SRCHS=""
	foreign_options=$(printf '%s\n' ${!foreign_option_*} | sort -t _ -k 3 -g)
	for optionvarname in ${foreign_options} ; do
		option="${!optionvarname}"
		echo "$option"
		split_into_parts $option
		if [ "$part1" = "dhcp-option" ] ; then
			if [ "$part2" = "DNS" ] ; then
				NMSRVRS="${NMSRVRS:+$NMSRVRS }$part3"
			elif [ "$part2" = "DOMAIN" ] ; then
				SRCHS="${SRCHS:+$SRCHS }$part3"
			fi
		fi
	done
	R=""
	[ "$SRCHS" ] && R="search $SRCHS
"
	for NS in $NMSRVRS ; do
        	R="${R}nameserver $NS
"
	done
	echo -n "$R" | /sbin/resolvconf -a "${dev}.openvpn"
	;;
  down)
	/sbin/resolvconf -d "${dev}.openvpn"
	;;
esac
```

I know some bash scripting, but how this is all linked together goes beyond my knowledge. Weird thing is, i dont even call update-resolv-conf in my openvpn script:

```
dev tun
proto udp
port 1194
ca /etc/openvpn/easy-rsa/pki/ca.crt
cert /etc/openvpn/easy-rsa/pki/issued/DietPi_405884d6-4479-47d6-b007-81db127340e3.crt
key /etc/openvpn/easy-rsa/pki/private/DietPi_405884d6-4479-47d6-b007-81db127340e3.key
dh /etc/openvpn/easy-rsa/pki/dh2048.pem
topology subnet
server 172.16.0.0 255.255.255.0
# Set your primary domain name server address for clients
push "dhcp-option DNS 192.168.2.20"
push "block-outside-dns"
push "route 192.168.2.0 255.255.255.0"
push "fast-io"
push "tun-mtu 1500"
# Override the Client default gateway by using 0.0.0.0/1 and
# 128.0.0.0/1 rather than 0.0.0.0/0. This has the benefit of
# overriding but not wiping out the original default gateway.
push "redirect-gateway def1"
client-to-client
tun-mtu 1500
keepalive 10 60
remote-cert-tls client
tls-version-min 1.2
tls-auth /etc/openvpn/easy-rsa/pki/ta.key 0
cipher AES-128-CBC
auth SHA256
ncp-disable
user openvpn
group openvpn
persist-key
persist-tun
crl-verify /etc/openvpn/crl.pem
### EXIT STANDAARD ###

status /var/log/openvpn-status.log 20
status-version 3
syslog
verb 4
#DuplicateCNs allow access control on a less-granular, per user basis.
#Remove # if you will manage access by user instead of device. 
duplicate-cn
# Generated for use by PiVPN.io
```

prov3it, 2020-11-17:
Okay so annoyed as i am with this problem i think i have a temp solution:

Not sure if this helped, but i cant recall having the file /etc/resolvconf/interface-order (which i found in: /etc/resolvconf/update.d/libc) I think it showed up after reinstalling resolvconf (incl. update.d folder):
```
sudo apt-get -o Dpkg::Options::="--force-confmiss" install --reinstall resolvconf
```

altering interface-order:

```
sudo vi /etc/resolvconf/interface-order
```

```
# interface-order(5)
lo.inet
lo.@(dnsmasq|pdnsd)
lo.!(pdns|pdns-recursor)
lo
tun*
tap*
hso*
em+([0-9])?(_+([0-9]))*
p+([0-9])p+([0-9])?(_+([0-9]))*
@(br|eth)*([^.]).inet
@(br|eth)*([^.]).@(dhclient|dhcpcd|pump|udhcpc)
@(br|eth)*
ppp*
*
```
i removed tun/tap/hso.

Now running update:
```
sudo dietpi-update
```

And its crashing again. But i know what the problem is now. I run pihole in docker.

```
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : rtorrent
[  OK  ] DietPi-Services | stop : minidlna
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php7.3-fpm
```
Is there a way to dietpi-update without stopping the services? 

I also tried changing the dns (dietpi-config), but then it starts adding wifi-packages (which i dont even have on this system). Last but not least i need to change the option to disable ipv6. What a nightmare.

Joulinar, 2020-11-17:
set the `docker` service to `Systemd controlled` and `excluded` using `dietpi-services`

prov3it, 2020-11-18:
Okay so the 10.16.0.1 entry is gone. I did check all the files on my installation for the ip 10.16.0.1 and all i could get was the files from resolvconf and logfiles from a program which is not to be mentioned by name. That got me thinking, im running ```namespaced-openvpn``` which came back in my search results, it must have something to do with that. I removed namespaced-openvpn and the program running in this namespace since im using docker & co. After a reboot 10.16.0.1 was gone. 

This issue can be closed if @MichaIng or someone else doesnt have any questions related to this issue.

MichaIng, 2020-11-18:
Jep that was it: https://github.com/slingamn/namespaced-openvpn#dns-leaks
> Since by default, OpenVPN does not remove LAN routes ...

`namespaced-openvpn` seems to do a lot of things, each likely reasonable with some value but making it very hard to know what is coming from where and messing up in combination with PiVPN.

Generally, a good advice is to go with a single solution for a single task, so if you need a VPN to access your local network securely from remote, then setup either plain OpenVPN or better WireGuard (e.g. via `dietpi-software`) or only PiVPN or only `namespaced-openvpn`. Each wrapper has/sets up it's own ideas and two will likely break each other 😉. WireGuard btw solves many of the concerns mentioned in `namespaced-openvpn` readme, e.g. a clear `DNS =` that can be set client-side to avoid DNS leaks.

Joulinar, 2020-11-24:
ok I will go to close the topic, feel free to reopen if needed

closed 2020-11-24 by Joulinar