MichaIng/DietPi #3919: Selecting read-only FS should work when mounted

springfielddatarecovery, 2020-11-18:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
When attempting to use dietpi-drive_manager to set the root fs to read-only, I am given the error:
"Mount point is busy"
This is after disabling the swap in the same menu. 
- ...

#### Describe the solution you'd like:
It seems to me the intuitive fix for this is to say "mount point is busy now, editing /etc/fstab, changes will occur on reboot"
- ...

#### Additional context
Related feature request https://github.com/MichaIng/DietPi/issues/229
- ...

Joulinar, 2020-11-18:
Hi,

mounting RootFS r/o will cause serious issues as services will not work anymore. Some needs read access to `/var/log` directory to be able to create log files. You should think on something like `OverlayRoot` if you really like to have RootFS r/o

https://dietpi.com/phpbb/viewtopic.php?f=11&t=7677&p=24920&hilit=Overlayroot#p24920

springfielddatarecovery, 2020-11-18:
I'll give that a try, thank you!

MichaIng, 2020-11-18:
But the idea is good to handle the being busy error (that is quite common) gracefully as you suggested. Let me turn that into an enhancement request.