MichaIng/DietPi #3925: RockPi S don't start with DiwtPi IMG

custodis-intestina, 2020-11-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | IMG FROM 
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
All Information comes from that Image. https://dietpi.com/downloads/images/DietPi_ROCKPiS-ARMv8-Buster.7z
- SBC model | RockPi S V12
- Power supply used | Meanwell RS-25-5
- SDcard used | SanDisk 16GB

#### Additional Information (if applicable)

#### Steps to reproduce
Flasching the Image to SD Card with Win32Diskimager
Insert SD Card in RockPi S
Plugin Power and Wait

#### Extra details
Serial Console
`DDR Version V1.26
REGFB: 0x00000031, 0x00000031
In
589MHz
DDR3
 Col=10 Bank=8 Row=15 Size=512MB
msch:1
Returning to boot ROM...
INFO:    Preloader serial: 0
NOTICE:  BL31: v1.3(release):3a579d3
NOTICE:  BL31: Built : 09:09:17, Jan 29 2019
INFO:    Lastlog: last=0x100000, realtime=0x102000, size=0x2000
INFO:    ARM GICv2 driver initialized
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x600000
INFO:    SPSR = 0x3c9

U-Boot 2017.09-02378-g6de46245e6 (Mar 12 2020 - 07:47:46 +0000), Build: jenkins-linux-build-release-32

Model: Radxa ROCK Pi S
PreSerial: 0
DRAM:  510 MiB
Relocation Offset is: 1f92b000
vdd_core 1044000 uV
vdd-core init 1044000 uV
MMC:   dwmmc@ff480000: 0, dwmmc@ff490000: 1
Using default environment

In:    serial@ff0c0000
Out:   serial@ff0c0000
Err:   serial@ff0c0000
Model: Radxa ROCK Pi S
Bootdev(atags): mmc 0
PartType: EFI
## Error: Can't overwrite "serial#"
## Error inserting "serial#" variable, errno=1
get part misc fail -1
boot mode: None
CLK:
apll 816000 KHz
dpll 1300000 KHz
vpll0 1179647 KHz
vpll1 903167 KHz
aclk_bus 185714 KHz
[... 368 lines cut ...]
[    1.411492] [BT_RFKILL]: ** disable irq
[    1.411717] [BT_RFKILL]: bt_default device registered.
[    1.411871] Key type dns_resolver registered
[    1.412330] flash vendor_init_thread!
[    1.412355] flash vendor storage:20170308 ret = -1
[    1.413467] rk3308-acodec ff560000.acodec: Don't need to change dac_output: 0
[    1.413667] registered taskstats version 1
[    1.415546] Key type encrypted registered
[    1.416279] hctosys: unable to open rtc device (rtc0)
[    1.432759] ALSA device list:
[    1.432816]   #0: Loopback 1
[    1.432828]   #1: rockchip,rk3308-acodec
[    1.433201] RAMDISK: gzip image found at block 0
[    1.542616] RAMDISK: incomplete write (31546 != 32768)
[    1.542672] write error
[    1.542768] Waiting for root device UUID=c3800af9-fa58-4fb3-bb56-e52fd04837ba...
[    4.433554] vendor storage:20160801 ret = -1
[   54.137571] random: nonblocking pool is initialized
`

After "[   54.137571] random: nonblocking pool is initialized" the System does nothing. LAN Leds blinking, Heartbeat is ON.

Sry but i have no Information found. 

I Hope you can Help me . LG Steve

MichaIng, 2020-11-21:
Many thanks for your report. There are a few more errors, not sure which one is relevant here, but in the end, the root device seems to be not found, although the partitions are detected before:
```
[    0.000000] Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
...
[    1.345696]  mmcblk0: p1 p2
...
[    1.433201] RAMDISK: gzip image found at block 0
[    1.542616] RAMDISK: incomplete write (31546 != 32768)
[    1.542672] write error
[    1.542768] Waiting for root device UUID=c3800af9-fa58-4fb3-bb56-e52fd04837ba...
```

... ah a look into the changelog: https://wiki.radxa.com/RockpiS/Debian#Changelogs
> ### 2020.06.16
> **rockpis_debian_buster_minimal_arm64_20200615_1225-gpt.img**
>
> - Fix the issue of failing to find root device. System boot log may show like this.
> ```
> [    1.478059] RAMDISK: gzip image found at block 0
> [    1.581744] RAMDISK: incomplete write (23864 != 32768)
> [    1.581801] write error
> [    1.581895] Waiting for root device UUID=37055840-4ec4-444f-979b-9e47ee4bd848...
> ```

Strange, since it worked before, but probably due to a change in newer hardware revision or firmware version. I'll re-create the image.

MichaIng, 2020-11-21:
Image ready for testing, would be great if you could give it a try: https://dietpi.com/downloads/images/testing/DietPi_ROCKPiS-ARMv8-Buster.7z

custodis-intestina, 2020-11-22:
Hey and sry for the late anwser. That Image Works. What wars the Solution? New Kernel? LG Steve

MichaIng, 2020-11-22:
Great, many thanks for testing.

Since Radxa documented exactly the error message you faced as well, I guess it was an error in their kernel in combination with a new hardware revision or shipped firmware version. So yes, I simply redid the image based on their new bootloader and kernel.

custodis-intestina, 2020-11-22:
There is another flaw. The key to the Radxa repository is no longer valid. Tried to reload it from Radxa, but no success.

MichaIng, 2020-11-22:
Lol, expired today:
```
root@DietPi:/tmp/DietPi-Software# apt-key list
/etc/apt/trusted.gpg
--------------------
pub   rsa3072 2018-11-23 [SC] [expired: 2020-11-22]
      C79E F8D1 49CF C9C0 418A  9A60 5761 288B 2B52 CC90
uid           [ expired] Radxa <dev@radxa.com>
```
And Radxa does not have a newer image yet, and ... there is no current key available... 👎
Just tested it, it's the same: https://apt.radxa.com/buster-stable/public.key
or: https://github.com/radxa/apt/blob/gh-pages/buster-stable/public.key

And there we have the bug report: https://github.com/radxa/apt/issues/9

MichaIng, 2020-11-22:
There is another bug in the repo while I tried to download and install the "new" (renamed) kernel meta package manually from the GitHub repo: https://github.com/radxa/apt/issues/10
Will be a messy image currently, probably we should have switched to Armbian directly, which have a functional (although limited featured) Linux 5.8 (even a testing 5.9) image.

MichaIng, 2020-11-22:
Okay image up with the Radxa repository disabled for now and the newer kernel packages installed manually: https://dietpi.com/downloads/images/testing/DietPi_ROCKPiS-ARMv8-Buster.7z
The repo will be re-enabled once a valid key is available.

MichaIng, 2020-11-24:
The repository key has been updated, but the bootloader package still depends on the old/removed kernel meta package: https://github.com/radxa/apt/issues/10
So currently, enabling the repository would lead to a dependency conflict that can be either resolved by an upgrade or bootloader package removal. The latter is possible since the bootloader has been flashed already, but in case of breaking changes I'd rather like to keep it to allow upgrades, when available. So I'll wait for this issue to be fixed before re-enabling the Radxa repository.

MichaIng, 2020-11-27:
Finally, both issues are solved. I'll apply the new key and re-enable the the repo on the image.

MichaIng, 2020-11-27:
Above image repacked, including some fine tuning and cleanup + really enabled serial console for first boot: https://github.com/MichaIng/DietPi/commit/324b59b89e8b849ef87539a777b2338d01542a36

@custodis-intestina
Should be fine now for production usage, would be great if you could give it a final test.

closed 2020-12-20 by MichaIng