MichaIng/DietPi #3926: apt update extremly slow (Raspberry Pi 3+)

Gill-Bates, 2020-11-22:
Hey there,

Is it only me or did others already notice that an `apt update` on a Raspberry Pi (3+ in my case) takes extremely long?

It does not matter whether it is a fresh installation or an older one. After an `apt update` it takes up to 2-3 minutes. Some people on the Internet say that this could be due to IPv6. Are there any findings on this?

ravenclaw900, 2020-11-22:
On my Raspberry Pi 4, it can take a long time to read the package lists.  It usually finishes within thirty seconds though.  The time stays the same when installing from the internet and installing from a local file, and because of that, using IPV6 makes no difference to me.

Joulinar, 2020-11-22:
Hi,

This is as expected. Because DietPi is keeping apt files in compressed format. Means they need to be decompress first. This is taking time. Reason is to reduce r/w operation on SD card.

MichaIng, 2020-11-22:
We should probably add a switch for this to the APT cache menu and make this somehow more prominent. E.g. when lists stay compressed, they could be moved to RAM without taking too much space, although bottleneck is not read speed but decompression time here. I chose strongest compression xz which is sadly slowest as well. The problem with faster low-ratio algorithms (like lz4) is that it saves nearly no writes then, since APT always downloads them in xz format, then extracts them (50 - 100 MiB writes only for extraction) and repacks them to new compression format. Same with gzip, even if gzip lists are available online (usually they are), APT always takes the xz ones when available.

I'm also not very happy with this, but heaving 100 MiB unnecessary SD writes on every apt update as well contradicts all efforts to increase SD lifespan where some worry already in case of much smaller sizes. What finally made me doing it is that Armbian images are shipped with APT list compression enabled by default as well, although they didn't change it to xz but left default lz4 in place, which means there it doesn't save SD writes but only used drive space, something I again don't care much about (in this case).

Strange is that the APT cache itself (which allows other APT tasks to parse package info much faster) has no effect on APT update speed, even decreases it slightly since the cache is rebuild then.

Joulinar, 2020-11-22:
> We should probably add a switch for this to the APT cache menu and make this somehow more prominent

Thats a good idea to have a switch to get it disabled. There are couple of user who moved to SSD/HDD. For them it might not matter how many r/w operations are done 😃

MichaIng, 2022-04-21:
Implemented for DietPi v8.4: #5447

closed 2022-04-21 by MichaIng