MichaIng/DietPi #3928: Dietpi boots straight to retropie but no input works

rostomzer, 2020-11-23:
I have installed RetroPie and set it to boot automatically however, when it boots, it asks le to setup controller (which is already detected) but when i press the controller nothing happens! The keyboard also is not working 
The only keys that are working on the keyboard is the ctrl+alt+backspace which kills the xserver

Magically when i start the desktop environment (startx) then launch retropie, everything works fine (controller+keyboard)

MichaIng, 2020-11-23:
Many thanks for your report.

How do you try to start RetroPie? With our `dietpi-autostart` option? Since I have no time to investigate this currently and we have removed RetroPie from `dietpi-software` a long time ago, I removed the autostart option as well just yesterday.

However, since the number if users seems to be quite high, I'll probably revert and fix it. If you tried it via `dietpi-autostart`, please instead try to run the following from console:
```
xinit /opt/retropie/supplementary/emulationstation/emulationstation.sh
```
It seems that emulationstation does not invoke the X server anymore, so this needs to be done manually now by invoking the script via `xinit`.

rostomzer, 2020-11-23:
> Many thanks for your report.
>
> How do you try to start RetroPie? With our `dietpi-autostart` option? Since I have no time to investigate this currently and we have removed RetroPie from [...]

That's exactly the command i use to run retropie via custom.sh script and it boot straight to emulationstation successfuly. but if you are familiar with emulationstation, at the initial screen it detects the controllers and asks you to configure the boutons. And its there where it is stuck. When i press boutons nothing works. Even the key board keys are not working. The only key combination that works is the ctrl+alt+backspace that kills the xserver.
I'm not skilled enough with linux but i think its related with driver initializing (maybe xpad) ??
[image]

MichaIng, 2020-11-23:
Ah wait, I think the custom.sh service does not have a STDIN. Could you try to edit `/etc/systemd/system/dietpi-autostart_custom.service` and add `StandardInput=tty` just below `StandardOutput=tty`?

rostomzer, 2020-11-24:
Just added that line and now retropie is not booting
Now it brings the login command

MichaIng, 2020-11-24:
Hmm, the login prompt is actually expected with 50% chance, since it (the getty@tty.service) and custom.sh service both have the same service `Type=idle` which makes them start when all other scheduled services have been, but with no clear order. However, even if the login prompt appears first, RetroPie should start shortly after, as it did before (with broken input).

Can you paste from the console:
```
systemctl status dietpi-autostart_custom
journalctl -u dietpi-autostart_custom
```

Finally it makes sense to create an own systemd unit for RetroPie that defines exactly when it should start (which can be much earlier, AFAIK), in case on a different TTY to not interfere with ongoing boot logs and without the overhead of a wrapping shell. But lets see how to get the basics running first.

rostomzer, 2020-11-24:
> Can you paste from the console:
> ```
> systemctl status dietpi-autostart_custom
> journalctl -u dietpi-autostart_custom
> ```

That gives invalid option -- 'u'

When i remove that -u it gives the results in the screenshot.

Tbh i'm not really skilled on linux but i can try what ever you want i'm available 24h/24 7d/7
If you need any thing testing just tell me.

Ps: i also tried to enable rc.local and input the same settings i have on my raspberry pi running osmc (debian based os) but it looks like it didn't work

MichaIng, 2020-11-24:
> That gives invalid option -- 'u'

That must have been a typo, probably you missed the dash `-`?

When not using it I get:
```
2020-11-24 20:18:14 root@micha:~# journalctl dietpi-autostart_custom
Failed to add match 'dietpi-autostart_custom': Invalid argument
```

Ah, probably you ran the two lines as a single command? Those are two dedicated commands. Ah yes that is it:
```
2020-11-24 20:18:53 root@micha:~# systemctl status dietpi-autostart_custom journalctl -u dietpi-autostart_custom
systemctl: invalid option -- 'u'
```
Please run both lines individually.

> When i remove that -u it gives the results in the screenshot.

Which screenshot? 😄

> i also tried to enable rc.local

Does it start not at all or with the same input issue? Not that the file needs to be executable: `chmod +x /etc/rc.local`
Then it should execute automatically at boot, very similar to the custom.sh service, as well invoked by a systemd unit without `StandardInput=tty`.

rostomzer, 2020-11-25:
Looks like rc.local is loaded successfully
I used it to initiate xboxdrv  driver (installed from retropie setup script) and as it interferes with xpad i used rmmod xpad command also to disable it. A'd the result is still the same. No input is working.
The driver detects the controllers and cant use them.
Xboxdrv always detects 2 controllers if it's loaded. But in my case i see 3 controllers! Which leads me to suspect that xpad is always still running!

The link belw is a screenshot of those 2 commands systemctl and journalctl

You will notice also the 3  last lines in the commands. Thise are when i plug in the controller and unplug it.

MichaIng, 2020-11-26:
Okay, then its definitely a driver issue. Can you paste the following:
```
ls -al /root/.startxrc
cat /root/.startxrc
ls -al /usr/lib/*/sys.startxrc
cat /usr/lib/*/sys.startxrc
ls -al /root/.xinitrc
cat /root/.xinitrc
ls -al /etc/X11/xinit/xinitrc
cat /etc/X11/xinit/xinitrc
```

rostomzer, 2020-11-27:
> Okay, then its definitely a driver issue. Can you paste the following:
> 
> ```
> ls -al /root/.startxrc
> cat /root/.startxrc
> ls -al /usr/lib/*/sys.startxrc
> cat /usr/lib/*/sys.startxrc
> ls -al /root/.xinitrc
> cat /root/.xinitrc
> ls -al /etc/X11/xinit/xinitrc
> cat /etc/X11/xinit/xinitrc
> ```

all commands returns with ''no such file or directory'' except the last two ones

`ls -al /etc/X11/xinit/xinitrc
-rwxr-xr-x 1 root root 164 Mar 19  2018 /etc/X11/xinit/xinitrc

cat /etc/X11/xinit/xinitrc
#!/bin/sh

# /etc/X11/xinit/xinitrc
#
# global xinitrc file, used by all X sessions started by xinit (startx)

# invoke global X session script
. /etc/X11/Xsession
`

MichaIng, 2020-11-27:
Could you paste:
```
cat /etc/X11/Xsession
ls -al /etc/X11/xinit/startxrc
cat /etc/X11/xinit/startxrc
```

rostomzer, 2020-11-27:
```
dietpi@DietPi:/$ cat /etc/X11/Xsession
#!/bin/sh
#
# /etc/X11/Xsession
#
# global Xsession file -- used by display managers and xinit (startx)

# $Id: Xsession 967 2005-12-27 07:20:55Z dnusinow $

set -e

PROGNAME=Xsession

message () {
  # pretty-print messages of arbitrary length; use xmessage if it
  # is available and $DISPLAY is set
  MESSAGE="$PROGNAME: $*"
  echo "$MESSAGE" | fold -s -w ${COLUMNS:-80} >&2
  if [ -n "$DISPLAY" ] && which xmessage > /dev/null 2>&1; then
    echo "$MESSAGE" | fold -s -w ${COLUMNS:-80} | xmessage -center -file -
  fi
}

message_nonl () {
  # pretty-print messages of arbitrary length (no trailing newline); use
  # xmessage if it is available and $DISPLAY is set
  MESSAGE="$PROGNAME: $*"
  echo -n "$MESSAGE" | fold -s -w ${COLUMNS:-80} >&2;
  if [ -n "$DISPLAY" ] && which xmessage > /dev/null 2>&1; then
    echo -n "$MESSAGE" | fold -s -w ${COLUMNS:-80} | xmessage -center -file -
  fi
}

errormsg () {
  # exit script with error
  message "$*"
  exit 1
}

internal_errormsg () {
  # exit script with error; essentially a "THIS SHOULD NEVER HAPPEN" message
  # One big call to message() for the sake of xmessage; if we had two then
  # the user would have dismissed the error we want reported before seeing the
  # request to report it.
  errormsg "$*" \
           "Please report the installed version of the \"x11-common\"" \
           "package and the complete text of this error message to" \
           "<debian-x@lists.debian.org>."
}

# initialize variables for use by all session scripts

OPTIONFILE=/etc/X11/Xsession.options

SYSRESOURCES=/etc/X11/Xresources
USRRESOURCES=$HOME/.Xresources

SYSSESSIONDIR=/etc/X11/Xsession.d
USERXSESSION=$HOME/.xsession
USERXSESSIONRC=$HOME/.xsessionrc
ALTUSERXSESSION=$HOME/.Xsession
ERRFILE=$HOME/.xsession-errors

# attempt to create an error file; abort if we cannot
if (umask 077 && touch "$ERRFILE") 2> /dev/null && [ -w "$ERRFILE" ] &&
  [ ! -L "$ERRFILE" ]; then
  chmod 600 "$ERRFILE"
elif ERRFILE=$(tempfile 2> /dev/null); then
  if ! ln -sf "$ERRFILE" "${TMPDIR:=/tmp}/xsession-$USER"; then
    message "warning: unable to symlink \"$TMPDIR/xsession-$USER\" to" \
             "\"$ERRFILE\"; look for session log/errors in" \
             "\"$TMPDIR/xsession-$USER\"."
  fi
else
  errormsg "unable to create X session log/error file; aborting."
fi

exec >>"$ERRFILE" 2>&1

echo "$PROGNAME: X session started for $LOGNAME at $(date)"

# sanity check; is our session script directory present?
if [ ! -d "$SYSSESSIONDIR" ]; then
  errormsg "no \"$SYSSESSIONDIR\" directory found; aborting."
fi

# Attempt to create a file of non-zero length in /tmp; a full filesystem can
# cause mysterious X session failures.  We do not use touch, :, or test -w
# because they won't actually create a file with contents.  We also let standard
# error from tempfile and echo go to the error file to aid the user in
# determining what went wrong.
WRITE_TEST=$(tempfile)
if ! echo "*" >>"$WRITE_TEST"; then
  message "warning: unable to write to ${WRITE_TEST%/*}; X session may exit" \
          "with an error"
fi
rm -f "$WRITE_TEST"

# use run-parts to source every file in the session directory; we source
# instead of executing so that the variables and functions defined above
# are available to the scripts, and so that they can pass variables to each
# other
SESSIONFILES=$(run-parts --list $SYSSESSIONDIR)
if [ -n "$SESSIONFILES" ]; then
  set +e
  for SESSIONFILE in $SESSIONFILES; do
    . $SESSIONFILE
  done
  set -e
fi

exit 0

# vim:set ai et sts=2 sw=2 tw=80:
dietpi@DietPi:/$ 
```
```
dietpi@DietPi:/$ ls -al /etc/X11/xinit/startxrc
ls: cannot access '/etc/X11/xinit/startxrc': No such file or directory
dietpi@DietPi:/$ 
```
```
dietpi@DietPi:/$ cat /etc/X11/xinit/startxrc
cat: /etc/X11/xinit/startxrc: No such file or directory
dietpi@DietPi:/$ 
```

MichaIng, 2020-11-27:
All defaults only. Last test so far, else the desktop itself seems to load something differently:
```
startx /opt/retropie/supplementary/emulationstation/emulationstation.sh
```

rostomzer, 2020-11-28:
> All defaults only. Last test so far, else the desktop itself seems to load something differently:
> 
> ```
> startx /opt/retropie/supplementary/emulationstation/emulationstation.sh
> ```

same result. ES runs, controllers detected but no input works (except ctrl+alt+backspace)

MichaIng, 2020-11-28:
Then I'm out of ideas. You could check if the desktop loads some custom X features/drivers at some point and otherwise the RetroPie devs will likely have a better idea, since I have no experience with game controllers 😅. We install X server and desktop from Debian/Raspbian repositories and, aside if disabling DPMS, do not touch X server configs, so that should be all pretty much standard setup.

Which board/device are you actually using?
And I wonder if there is no docs by RetroPie about how to autostart it best? Or do they only support their full applicances/images?

rostomzer, 2020-11-29:
> Then I'm out of ideas. You could check if the desktop loads some custom X features/drivers at some point and otherwise the RetroPie devs will likely have a better idea, since I have no experience [...]

actually i'm trying on virtualbox and a x64 laptop. i don't want to mess with my raspberry pi 3 as it's my main box on my TV. it's running debian based OS with retropie

i will download retropie OS for raspebrry pi (that OS that runs only retropie) and try to investigate how emulationstations starts there.

MichaIng, 2020-12-26:
I'm gonna mark this issue as closed. We removed our autostart option until someone finds time to test the RetroPie install and reasonable implementation into DietPi from ground up. That issue is handled here: #3090
Please report your possible finds from RetroPie OS there.

closed 2020-12-26 by MichaIng